noble/noble #201: Support for win32 using the websocket binding

bragma, 2015-06-16:
Hi,
as an alternative to https://github.com/sandeepmistry/noble/issues/84, why not allowing to install on windows and use the websocket client binding? It may be useful for testing and remote sensors.
Changes should be small:

In package.json, add "win32"

```
"os": [
    "darwin",
    "linux",
    "win32"
  ],
```

In lib/noble.js, detect "win32" platform and use the websocket binding

```
if (process.env.NOBLE_WEBSOCKET || process.title === 'browser' || platform === 'win32') {
  bindings = require('./websocket/bindings');
```

I can provide the PR if you agree.

bragma, 2015-06-18:
@loghorn great submission, I hope it will be accepted soon!

sandeepmistry, 2015-07-07:
As per #202, closing this request for now.

closed 2015-07-07 by sandeepmistry