noble/noble #203: Service and Characteristic name not configurable

sonkrishna, 2015-06-18:
Hi Sandeep,

Hope you are doing great!

We followed your node module and wrote code on BLE and its working. However Service and Characteristic name are not configurable. It's coming like UNKNOWN SERVICE , UNKNOWN CHARACTERISTIC.

Your valuable input will be highly appreciable.

Thanks in advance!

Regards,
Krishna

bragma, 2015-06-19:
Hi @sonkrishna, I'm not sure about what you are trying to do, but service and characteristic names are defined in lib/characteristics.json and services.json. Have you checked that the service and characteristic UUID you receive from the device are present in those files?

sandeepmistry, 2015-06-24:
@sonkrishna this is expected behaviour, only Bluetooth SIG approved UUID's have non-unknown name values.

https://developer.bluetooth.org/gatt/Pages/default.aspx

closed 2015-06-24 by sandeepmistry