noble/noble #204: Problem connecting to bluetooth peripheral on Linux

aderhgawen, 2015-06-25:
I have been experiencing difficulties connecting to a bluetooth device (a [Lightblue Bean](https://punchthrough.com/bean/) from Punch Through Design) with a Raspberry Pi. It works fine for a couple of minutes, but then it seems to hang. I wrote a simple javascript file that connects and disconnects to the bluetooth device to see how long it runs without stalling. This code seems to run fine on my Mac, but on the Raspberry it hangs within 2-5 minutes.

Here is the code I'm using to test - https://gist.github.com/aderhgawen/32f1df8a7c495cd509db

```
Raspberry Pi: Pi 2 Model B
Node version: v0.10.38
Bluetooth adapter: IOGear Bluetooth 4.0 USB Micro Adapter (GBU521)
```
```
Laptop: MacBook Pro (Retina, 13-inch, Early 2015) running OS X Yosemite (10.10.3)
Node version: v0.10.36
```
Any idea where the problem could be?

Thanks!

sandeepmistry, 2015-07-05:
@aderhgawen what kernel version are you using?

Could you add some logging to handle the connection error scenario? Then provide a Gist of the updated code, log output, and hcidump capture.

aderhgawen, 2015-07-06:
Kernel version: 3.18.14-v7+

The code works fine in debugging mode if I have breakpoints at certain points. If I have a breakpoint right before the "peripheral.disconnect", it works fine. However, if I have a breakpoint inside the callback for peripheral.disconnect, it stalls. See my comments in this code - https://gist.github.com/aderhgawen/2efcb648d0abc5eb5dca

This is what the hcidump output looks when it crashes:

[image]

Or, sometimes it looks like this:

[image]

Here is the complete log:

@raspberrypi ~ $ sudo hcidump
HCI sniffer - Bluetooth packet analyzer ver 2.4
device: hci0 snap_len: 1028 filter: 0xffffffff
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 27
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 60:03:08:AE:D8:B1 (Public)
      Flags: 0x1a
      Unknown type 0xff with 10 bytes data
      RSSI: -89
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 60:03:08:AE:D8:B1 (Public)
      RSSI: -89
> HCI Event: LE Meta Event (0x3e) plen 27
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 18:EE:69:1F:AC:C5 (Public)
      Flags: 0x1a
      Unknown type 0xff with 10 bytes data
      RSSI: -71
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 18:EE:69:1F:AC:C5 (Public)
      RSSI: -72
> HCI Event: LE Meta Event (0x3e) plen 36
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
[... 1150 lines cut ...]
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 36
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C4:BE:84:E4:D3:CF (Public)
      Flags: 0x06
      Unknown type 0x06 with 16 bytes data
      TX power level: 4
      RSSI: -68
> HCI Event: LE Meta Event (0x3e) plen 18
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr C4:BE:84:E4:D3:CF (Public)
      Complete local name: 'node'
      RSSI: -59
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout

aderhgawen, 2015-07-06:
I seem to have found the place where the problem begins. The problem is in peripheral.js:

```
Peripheral.prototype.disconnect = function(callback) {
  if (callback) {
    this.once('disconnect', function() {
      callback(null);
    });
  }
  this.state = 'disconnecting';
  // ... Setting a breakpoint here makes the program work ...
  this._noble.disconnect(this.uuid);
  // ... Setting a breakpoint here makes no difference and the program stalls ...
};
```

Setting a breakpoint before ```  this._noble.disconnect(this.uuid);``` makes the program work more consistently. I am guessing there is some timing issue - The device doesn't finish disconnecting and Noble starts scanning again.

Your thoughts?

Thanks!

aderhgawen, 2015-07-08:
@sandeepmistry <b>Another update:</b> Digging deeper into the issue revealed that the problem starts in l2cap-ble.js. I modified the original disconnect function:

```
L2capBle.prototype.disconnect = function() {
  this._l2capBle.kill('SIGHUP');
};
```
I modified this and added a 500 ms delay before sending the signal:

```
L2capBle.prototype.disconnect = function() {
var _this = this;
setTimeout(function(){ _this._l2capBle.kill('SIGHUP'); },500);
};
```
This made it considerably more stable, and my test code successfully connected/disconnected more than 100 times. Without this, it rarely runs more than 10 times.

I really don't know why delaying this signal makes it more stable. Signals are asynchronous, and a signal could be delivered to a process while its handling another signal, and this could lead to race conditions. I'm guessing something like this could be happening here. Any thoughts?

rosek86, 2015-07-08:
@aderhgawen Can this be related to #190? I had very similar behaviour due to unhandled signals in the C part of noble. Can you try apply #192 PR which I've prepared a while ago and let us know if this fixes your issue?

you can apply the fix and then run 'npm rebuild noble' to re-compile C part.

aderhgawen, 2015-07-08:
@rosek86 That does fix the issue to a large extent! Great job. It just needed better signal handling. My program still hangs occasionally, and I get "LMP Response Timeout" on the hcidump log. 

```
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout
```
Any idea why this could be happening?

In your hci-ble.c code:
```
hci_le_set_scan_enable(hciSocket, 0x00, 1, 1000);
if (hci_le_set_scan_enable(hciSocket, 0x01, 1, 1000) == 0) {
          printf("scan start,success,withFilter\n");
        } else {
          printf("scan start,failure,withfilter,%s\n", strerror(errno));
        }
```
Why don't you check if the first statement (disable scan) executed successfully or not?  ```hci_le_set_scan_enable(hciSocket, 0x00, 1, 1000);```

Thanks!

rosek86, 2015-07-09:
@aderhgawen I'm glad that this fixes your issue a bit. I don't know what the 'LMP Response TImeout' means, I'm afraid. I'll have a look on the core specification in a spare time. Do you send/receive anything to/from your peripheral or simply establish a connection and get the timeout after a while? 

Regarding your question about scan disable statement, I've noticed that the statement is there to 'make sure that scanning is disabled', in normal operation it returns with error code (scanning already disabled).

@sandeepmistry Can you consider to merge the #192 PR, please? It seems that the common issue.

Thanks

aderhgawen, 2015-07-13:
@rosek86 I don't send/receive anything, and my script basically just connects to the peripheral and disconnects. I wrote this script to test how reliable noble is on Linux. My script runs without ever stalling on Mac, but I get the LMP response timeout error on Linux. Your changes have definitely made it more stable, but I still get that timeout error. Here is the script I am using to test noble - https://gist.github.com/aderhgawen/32f1df8a7c495cd509db

sandeepmistry, 2015-08-12:
noble now uses bluetooth-hci-socket instead of spawned (hci-ble and l2cap-ble) processes.

This issue should be resolved now.

closed 2015-08-12 by sandeepmistry