noble/noble #206: When I install noble on my MAC, the error is like this.

kaiser-ren, 2015-07-03:
Below is debug.log, please view. If you need my further help, please let me know.

kaiser-ren, 2015-07-03:
0 info it worked if it ends with ok
1 verbose cli [ 'node', '/usr/local/bin/npm', 'install', 'noble' ]
2 info using npm@2.11.2
3 info using node@v0.12.5
4 verbose install initial load of /Users/renkai/Desktop/git/SIG/nodejs/package.json
5 warn package.json node-api@ No description
6 warn package.json node-api@ No repository field.
7 warn package.json node-api@ No README data
8 warn package.json node-api@ No license field.
9 verbose installManyTop reading scoped package data from /Users/renkai/Desktop/git/SIG/nodejs/node_modules/body-parser/package.json
10 verbose installManyTop reading scoped package data from /Users/renkai/Desktop/git/SIG/nodejs/node_modules/express/package.json
11 verbose installManyTop reading scoped package data from /Users/renkai/Desktop/git/SIG/nodejs/node_modules/mongoose/package.json
12 info package.json mongoose@3.6.20 No license field.
13 verbose readDependencies loading dependencies from /Users/renkai/Desktop/git/SIG/nodejs/package.json
14 silly cache add args [ 'noble', null ]
15 verbose cache add spec noble
16 silly cache add parsed spec { raw: 'noble',
16 silly cache add   scope: null,
16 silly cache add   name: null,
16 silly cache add   rawSpec: 'noble',
16 silly cache add   spec: '/Users/renkai/Desktop/git/SIG/nodejs/noble',
16 silly cache add   type: 'directory' }
17 verbose addLocalDirectory /Users/renkai/.npm/noble/1.0.1/package.tgz not in flight; packing
18 verbose tar pack [ '/Users/renkai/.npm/noble/1.0.1/package.tgz',
18 verbose tar pack   '/Users/renkai/Desktop/git/SIG/nodejs/noble' ]
19 verbose tarball /Users/renkai/.npm/noble/1.0.1/package.tgz
20 verbose folder /Users/renkai/Desktop/git/SIG/nodejs/noble
21 info prepublish noble@1.0.1
22 verbose addLocalTarball adding from inside cache /Users/renkai/.npm/noble/1.0.1/package.tgz
23 silly cache afterAdd noble@1.0.1
24 verbose afterAdd /Users/renkai/.npm/noble/1.0.1/package/package.json not in flight; writing
25 verbose afterAdd /Users/renkai/.npm/noble/1.0.1/package/package.json written
26 silly install resolved [ { author: { name: 'Sandeep Mistry', email: 'sandeep.mistry@gmail.com' },
26 silly install resolved     name: 'noble',
26 silly install resolved     description: 'A node.js BLE (Bluetooth low energy) central library.',
26 silly install resolved     version: '1.0.1',
26 silly install resolved     repository:
26 silly install resolved      { type: 'git',
26 silly install resolved        url: 'git+https://github.com/sandeepmistry/noble.git' },
26 silly install resolved     bugs: { url: 'https://github.com/sandeepmistry/noble/issues' },
26 silly install resolved     keywords:
26 silly install resolved      [ 'bluetooth',
26 silly install resolved        'BLE',
26 silly install resolved        'bluetooth low energy',
26 silly install resolved        'bluetooth smart',
26 silly install resolved        'central' ],
26 silly install resolved     main: './index.js',
26 silly install resolved     engines: { node: '>=0.8' },
26 silly install resolved     os: [ 'darwin', 'linux' ],
26 silly install resolved     dependencies:
26 silly install resolved      { debug: '~2.2.0',
26 silly install resolved        'bplist-parser': '0.0.6',
26 silly install resolved        'xpc-connection': '~0.1.3' },
26 silly install resolved     optionalDependencies: { 'bplist-parser': '0.0.6', 'xpc-connection': '~0.1.3' },
26 silly install resolved     devDependencies:
26 silly install resolved      { jshint: 'latest',
26 silly install resolved        mocha: '~1.8.2',
26 silly install resolved        should: '~1.2.2',
26 silly install resolved        sinon: '~1.6.0',
26 silly install resolved        async: '~0.2.9',
26 silly install resolved        ws: '~0.4.31' },
26 silly install resolved     scripts:
26 silly install resolved      { pretest: 'jshint *.js lib/. test/.',
26 silly install resolved        test: 'mocha -R spec test/*.js',
26 silly install resolved        install: 'node-gyp rebuild' },
26 silly install resolved     browser:
26 silly install resolved      { './distributed/bindings': false,
26 silly install resolved        './mac/bindings': false,
26 silly install resolved        './linux/bindings': false },
26 silly install resolved     gypfile: true,
[... 736 lines cut ...]
444 error Darwin 14.4.0
445 error argv "node" "/usr/local/bin/npm" "install" "noble"
446 error node v0.12.5
447 error npm  v2.11.2
448 error code ELIFECYCLE
449 error noble@1.0.1 install: `node-gyp rebuild`
449 error Exit status 1
450 error Failed at the noble@1.0.1 install script 'node-gyp rebuild'.
450 error This is most likely a problem with the noble package,
450 error not with npm itself.
450 error Tell the author that this fails on your system:
450 error     node-gyp rebuild
450 error You can get their info via:
450 error     npm owner ls noble
450 error There is likely additional logging output above.
451 verbose exit [ 1, true ]
452 verbose unbuild node_modules/noble
453 info preuninstall noble@1.0.1
454 info uninstall noble@1.0.1
455 verbose unbuild rmStuff noble@1.0.1 from /Users/renkai/Desktop/git/SIG/nodejs/node_modules
456 info postuninstall noble@1.0.1
457 silly gentlyRm /Users/renkai/Desktop/git/SIG/nodejs/node_modules/noble is being purged from base /Users/renkai/Desktop/git/SIG/nodejs
458 verbose gentlyRm don't care about contents; nuking /Users/renkai/Desktop/git/SIG/nodejs/node_modules/noble
459 silly vacuum-fs purging /Users/renkai/Desktop/git/SIG/nodejs/node_modules/noble
460 silly vacuum-fs quitting because other entries in /Users/renkai/Desktop/git/SIG/nodejs/node_modules

sandeepmistry, 2015-07-05:
@kaiser-ren can you please provide the output from the console as well?

I'm guessing it's related to #200  ...

sandeepmistry, 2015-07-11:
@kaiser-ren any updates

kaiser-ren, 2015-07-14:
Sorry, now I can use noble in my Mac, I should close this issue.

closed 2015-07-14 by kaiser-ren