noble/noble #207: Error code from an attribute write-with-response is not propagated out of noble

mvines, 2015-07-08:
When a peripheral returns an error for an attribute write request, the resulting error code is not provided to the noble user.  Taking the Yosemite bindings as an example, `args.kCBMsgArgResult` [1] is not *emitted* in any way.

[1] https://github.com/sandeepmistry/noble/blob/180bf2f5291b008f2d3c23b9fdebe736b9d57d46/lib/mac/yosemite.js#L562

TorstenRobitzki, 2015-10-21:
+1 I would love to see this fixed too :-) Anything I can do to help with this?

sandeepmistry, 2015-11-01:
@TorstenRobitzki I'm thinking of moving to using [node-core-bluetooth](https://github.com/sandeepmistry/node-core-bluetooth) before working on this.

However, I'd be up for a solution with the current code base. Easy solution would be to add an error param to the binding events and check for it in the listener.

TorstenRobitzki, 2016-02-22:
Any news on this. Address to send pizza to? ;-)

tybro0103, 2017-03-10:
An idea for fixing this: https://github.com/sandeepmistry/noble/pull/564

zkiiito, 2017-03-10:
#479 fixes this, but has a bug and nobody says anything, what to do about it :)

tybro0103, 2017-03-10:
@zkiiito ah yeah, looks like https://github.com/sandeepmistry/noble/pull/479 addresses a specific instance of the lack of error handling. https://github.com/sandeepmistry/noble/pull/564 aims to address error handling as a whole. Both use the exact same approach.