noble/noble #208: Notify only triggered once (on setup)

OJFord, 2015-07-10:
At first I thought I was experiencing https://github.com/sandeepmistry/noble/issues/136, but where that issue describes execution stalling, with no callback, instead I'm not getting the events at all (i.e. program continues, without notification).

- Peripheral notifications known working, as tested with 'nRF Master Control Panel' for Android
- Simple debug handler: `characteristic.notify(true, function(err){ console.log('Notification. Error:', err) });`
- One-time output: "Notification. Error: null"
- Triggering notifications on the peripheral has no effect on the program - which continues, but is never notified

1. Is there some kind of ack that noble expects of peripherals, but nRF MCP does not?
2. Is my code, or expectation, totally wrong?

Running on OS X 10.10.3

closed 2015-07-10 by OJFord

OJFord, 2015-07-10:
I realise after reading [the 'getting started' section](https://github.com/sandeepmistry/noble/wiki/Getting-started#user-content-getting-notified-on-characteristic-value-change) again that `.notify(enable, callback(error))` is indeed only a setup.

In my opinion this is not clear from [the pizza example](https://github.com/sandeepmistry/noble/blob/master/examples/pizza/central.js#L139), and somewhat curiously `.notify` isn't in the readme at all.

I suggest it's added to [the characteristic > notify section](https://github.com/sandeepmistry/noble#notify-1), as without it presumably `.on('data', callback(data, isNotification))` will only ever be triggered by reads.

sandeepmistry, 2015-07-11:
@OJFord please submit a Pull Request for your suggested change to the read me.