noble/noble #210: Can't connect to a LightBlue Bean from OS X

kofno, 2015-07-12:
I'm running master because it wouldn't compile without the fix from #205. When I try to connect, the code appears to work, but the peripheral never gets out of the 'connecting' state. LightBlue app on my iPhone can talk to the Bean just fine.

Here is my example code and the full out put with debugging on:
https://gist.github.com/kofno/94d9c1b90350845f923f

sandeepmistry, 2015-07-12:
@kofno can you please try connecting with LightBlue for OS X? https://itunes.apple.com/ca/app/lightblue/id639944780?mt=12

If that doesn't work, kill blued and try again: ```sudo killall blued```

kofno, 2015-07-12:
So, I couldn't get all that to work, so in frustration, I REBOOTED ALL THE THINGS! Now everything seems to be working. Sorry to have bothered you. Thank you for your help.

closed 2015-07-12 by kofno