noble/noble #212: Documentation: a proposal

gbrault, 2015-07-14:
I start documenting your code: are you intersted to get this documentation. I use JSDoc comments.
For example:

/**
 * discover services 
 * @param uuids call it with [] to discover all primary services
 * 
 * 
 * when all services discovered
 * set this._services for each services [{startHandle, endHandle, uuid}]
 * emit servicesDiscover event
 */
L2capBle.prototype.discoverServices = function(uuids) {...

Do you beleive this is a good idea?
How can we manage to get it in your code if you want?

sandeepmistry, 2015-07-19:
@gbrault I'm be open to using JSDoc comments.

I'm working on changing the Linux side to use [bluetooth-hci-socket](https://github.com/sandeepmistry/node-bluetooth-hci-socket), see this branch: https://github.com/sandeepmistry/noble/tree/bluetooth-hci-socket

Maybe we should wait until that is done? Then you can submit a pull request with your changes.

gbrault, 2015-07-19:
Just let me know when (and how) to test the use of this new branch with the cc2650stk and I can start documenting.

sandeepmistry, 2015-08-12:
@gbrault I've merged the branch into master.

How about you start document a small piece, then get feedback on it before continuing the rest?

gbrault, 2015-08-12:
Where should I start to make this first trial? Any preference?

sandeepmistry, 2015-08-13:
@gbrault how about the public facing API's? The ones in the read me, I think those would have most value. What do you think?

gbrault, 2015-08-14:
@sandeepmistry I have started, please see noble fork in my github place. You can see
* how to setup jsdoc in eclipse
* you open the doc in out directory (github will show the source...): index.html
* you can see noble as being a class and stratscanning being a method
* please see noble.js and read jsdoc decoration there

gbrault, 2015-08-14:
you can see how to setup eclipse [there](https://github.com/gbrault/noble/blob/master/jsdocsetup.md)

sandeepmistry, 2015-08-30:
@gbrault looks good!

Is there a way for jsdoc to create markdown file? I'm thinking it would be better to remove the API docs from the read me and have it generated by jsdoc. Do you have a few examples of other projects that use jsdoc?

Also, instead of using Eclipse, can jsdoc be run from an npm run script? So we can add it as a prepublish  step.

sandeepmistry, 2015-09-20:
@gbrault any updates on the above?

gbrault, 2015-09-20:
pls see answers to your above points
#### Is there a way for jsdoc to create markdown file?
I made some research and found some jsdoc like which would generate .md, but they look not very robust compared to jsdoc3. I rather think we need to stick to jsdoc...
#### Can jsdoc be run from an npm run script?
Yes, it's actually what I am doing (even if i use eclipse as an IDE). 
##### Install jsdoc3
npm install jsdoc: jsdoc is a nodejs extension
##### from the command line, run it against the code to document
jsdoc -r --verbose {directory of the code} -d {directory of the documentation}

#### Do you have a few examples of other projects that use jsdoc?
Not really, but it looks like the de-facto standard for nodejs projects nowadays. Jsdoc is linked to the http://usejsdoc.org/ site which document quite well jsdoc usage. The project is on Github https://github.com/jsdoc3/jsdoc

#### How to link the html documentation pages without having it embedded into the master branch of a git project (not one of your question)?
I uderstand it's quite puzzling to have the html pages inside the git project. But there is a possibility to set a specific branch 'gh-pages' into a github project, where you put html files (index.html is mandatory as an entry point). You need to follow those steps
##### Create a 'gh-pages' branch to your project
##### Put the documentation pages in this branch
Of course it then is not necessary to have that documentation in the master branch...
##### In the readme.md file of your master branch 
you can now link to http://&lt;account&gt;.github.io/&lt;project&gt;

You can see this setting live in the 'node' clone I have done: see https://github.com/gbrault/noble and have a look to the Readme where I added a link to the jsdoc I started http://gbrault.github.io/noble/

rzr, 2021-03-02:
Please can you update status of this issue on this latest release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-12