noble/noble #213: Connect and write to a peripheral without scanning (address and handle is known)

yasdfgr, 2015-07-14:
Is it possible to connect to a peripheral without scanning?
I have all addresses xx:xx:xx:xx:xx:xx.
How can i do that?

Is it possible to create a peripheral object by my own and simply use the methods connect() and writeHandle()?

littlemonkey, 2015-07-16:
I also think this function is very valuable!

sandeepmistry, 2015-07-19:
This is not be possible on OS X, so I haven't implemented it.

Work around for now is to start scanning and check ```peripheral.address``` on discover event.

yasdfgr, 2015-07-19:
I had the idea to compose a peripheral object on my own - but it seems not so easy. Would it work theoretically?

sandeepmistry, 2015-08-18:
This enhancement will not work with OS X, closing for now ....

closed 2015-08-18 by sandeepmistry