noble/noble #214: Is there a maximum of parallel BLE Connections? Only 5 are working!

yasdfgr, 2015-07-15:
I have the following Blutooth-Dongle on Raspberry Pi. I scan and connect one after one of my devices (i want to use 30, i tested it with 9). After some time my program says, ok - 9 devices connected.
By the way, i figuered out that i should do only one connection attempt at a time - so i block new connections until the one started returns (callback of connect). Otherwise there will be some strange errors.
But i see that only a maximum of 5 are finally working. That depends on the detection queue.
I have also implemented a disconnect function and i can see that 4 times noble says "unknown peripheral".
I think something is wrong with the connection management. Noble thinks 9 are connected but the layer below holds only a maximum of 5.

Any ideas?

Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 00:1A:7D:DA:71:13  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING
        RX bytes:200270 acl:20 sco:0 events:6605 errors:0
        TX bytes:4049 acl:155 sco:0 commands:167 errors:0

sandeepmistry, 2015-07-19:
@yasdfgr can you please provide an ```hcidump``` capture: ```sudo hcidump -t```

Also, have you checked what the behaviour of ```gatttool``` with 9 peripherals connected is?

yasdfgr, 2015-07-19:
Ok, here first is my own app-logging.
I'm scanning for 9 devices, where the device names are known.
I have a list with the names. If the scanned device's name matches and it's not yet connect it will be.

You can see, one after one ist connecting.
Then i send several values to all devices.
The first time, seems good, but no reaction on devices.
Then you can see, only 5 send commands (0x52) are transmitted. You can see 4 times: "noble warning: unknown peripheral ..."

And the same on disconnecting: 4 times  "noble warning: unknown peripheral ... "

```
pi@raspberrypi ~ $ sudo node playbulbart.js
start scanning...
found peripheral: playbulb 03 (ac:e6:4b:06:7b:f4)
wait 2s before connect
found peripheral: playbulb 01 (ac:e6:4b:06:7b:e1)
found peripheral: playbulb 02 (ac:e6:4b:06:b3:b1)

connected to peripheral: playbulb 03 (ac:e6:4b:06:7b:f4)

wait 2s after connect
restart scanning after connect...
found peripheral: playbulb 02 (ac:e6:4b:06:b3:b1)
wait 2s before connect

connected to peripheral: playbulb 02 (ac:e6:4b:06:b3:b1)

wait 2s after connect
restart scanning after connect...
found peripheral: playbulb 01 (ac:e6:4b:06:7b:e1)
wait 2s before connect

connected to peripheral: playbulb 01 (ac:e6:4b:06:7b:e1)

wait 2s after connect
restart scanning after connect...
found peripheral: playbulb 06 (ac:e6:4b:06:b3:a1)
wait 2s before connect

connected to peripheral: playbulb 06 (ac:e6:4b:06:b3:a1)

wait 2s after connect
restart scanning after connect...
found peripheral: playbulb 05 (ac:e6:4b:06:7b:fd)
wait 2s before connect

connected to peripheral: playbulb 05 (ac:e6:4b:06:7b:fd)

wait 2s after connect
restart scanning after connect...
found peripheral: undefined (ac:e6:4b:06:7c:f6)
wait 2s, no match
restart scanning, because no match...
found peripheral: playbulb 07 (ac:e6:4b:06:7c:f6)
wait 2s before connect

connected to peripheral: playbulb 07 (ac:e6:4b:06:7c:f6)

wait 2s after connect
restart scanning after connect...
found peripheral: playbulb 01 (ac:e6:4b:06:7b:e1)
wait 2s, no match
restart scanning, because no match...
found peripheral: playbulb 03 (ac:e6:4b:06:7b:f4)
wait 2s, no match
restart scanning, because no match...
found peripheral: undefined (ac:e6:4b:06:7c:81)
wait 2s, no match
restart scanning, because no match...
[... 1678 lines cut ...]
    handle 57 reason 0x13
    Reason: Remote User Terminated Connection
2015-07-19 16:40:13.238853 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 62 reason 0x16
    Reason: Connection Terminated by Local Host
2015-07-19 16:40:13.239857 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2015-07-19 16:40:13.275859 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 67 reason 0x16
    Reason: Connection Terminated by Local Host
2015-07-19 16:40:13.279822 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 57 reason 0x16
    Reason: Connection Terminated by Local Host
2015-07-19 16:40:13.314863 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 72 reason 0x16
    Reason: Connection Terminated by Local Host
2015-07-19 16:40:13.451530 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 52 reason 0x13
    Reason: Remote User Terminated Connection
2015-07-19 16:40:13.453883 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2015-07-19 16:40:13.519876 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 52 reason 0x16
    Reason: Connection Terminated by Local Host
```

yasdfgr, 2015-07-19:
Ok i've found something here:
http://stackoverflow.com/questions/25193975/gatttool-limited-to-5-connections

>It is a hardware limitation of the CSR dongle. I did a similar test using bluetoothctl not long ago: Using >a CSR dongle, only 5 bluetooth LE devices could connect. With a Broadcom dongle (BCM20702) I >could establish 14 concurrent connections. Then I ran out of bluetooth LE devices....
>
>Regards, Martin Rösch

This limitation should be handled by noble.

yasdfgr, 2015-07-21:
I have new's. The [Broadcom BCM20702 Bluetooth Dongle ] (http://www.amazon.de/gp/product/B007MKMJGO/?ie=UTF8&camp=1638&creative=6742&linkCode=ur2&psc=1&redirect=true&site-redirect=de&tag=brandtlightin-21) **can handle up to 14 BLE device connections simulateous**.
And overall it's a lot faster than this one from [Cambridge Silicon Radio](http://www.amazon.de/gp/product/B00LSGUBLI/?ie=UTF8&camp=1638&colid=30M4T2T3VGULZ&coliid=I2D81GCY2PIR5Y&creative=6742&linkCode=ur2&site-redirect=de&tag=brandtlightin-21) (**can handle a max. of 5 BLE device connections simulateous**).

PimButton, 2015-07-24:
@yasdfgr Did you test with more then 14 BLE devices or did you also ran out of devices? I might need to go all the way up to 20 and looking for a chip that will support it...

yasdfgr, 2015-07-25:
Yes, tested it. 14 connections were the maximum with this chip. Than it has the same behaviour like my first adapter with 5.

sandeepmistry, 2015-08-12:
@yasdfgr can you please re-test this with what's on noble master?

yasdfgr, 2015-08-12:
@sandeepmistry I don't quite understand your question. What should i test exactly? Do you want a hcidump until i can't connect anymore?

sandeepmistry, 2015-08-13:
@yasdfgr yes please.

The Linux side has been reworked to use [node-bluetooth-hci-socket](https://github.com/sandeepmistry/node-bluetooth-hci-socket). I'm wondering if the behaviour has changed. I expect the adapter to fail the create connection when over the limit ...

sandeepmistry, 2015-08-18:
@yasdfgr any updates?

sandeepmistry, 2015-08-30:
Closing do to lack of activity ...

closed 2015-08-30 by sandeepmistry

jrobeson, 2015-10-28:
@PimButton : did you ever find a chip that supported 20?

PimButton, 2017-03-24:
@yasdfgr you wrote in July 2015 that you tested the BCM20702 Bluetooth Dongle with 14 connections simulateous. Did you test this with Noble? Because I am currently running in to a problem when connecting to the 12th device. The first 11 connect without problems, but when connecting to the 12th device the status immediately jumps back to 'disconnecting'. 

Do you maybe have the test script that you used for me? I would very much like to test this myself so I can find out the problem.