noble/noble #218: Feature request: handle multiple bluetooth dongles from noble

yasdfgr, 2015-07-21:
I request the feature that noble can handle more than one bluetooth dongles in a system. This would be very helpfull for BLE mass applications, where the count of devices exceeds the maximum possible connections per dongle.

sandeepmistry, 2015-07-26:
This can sort of be done now with multiple processes on Linux by setting an environment variable for each node process:

```sh
sudo NOBLE_HCI_DEVICE_ID=1 node <your file>.js
```

sandeepmistry, 2015-08-18:
Closing this enhancement request until there is more demand.

closed 2015-08-18 by sandeepmistry

NAllred91, 2017-10-06:
I'm interested in this as well.  Multiple node process in a linux environment may be a decent solution though

marukqs, 2018-03-05:
Would be great feature. In my case there are 20 kids in a class using microbits, however CSR chipset on teachers BLE->internet gateway has 5 connections limit. Thank you Sandeep for your great job!

anant3110, 2019-04-18:
Can we have multiple node processes with different environment variables on Windows as well?

shwetankshrey, 2019-04-18:
Can we have multiple processes on Windows with different environment variables as well?