noble/noble #219: noble: Could not start scanning, state is unknown (not poweredOn)

hansamann, 2015-07-23:
Hi Sandeep, 

I've updraded node v0.12.7, mainly due to Promise support, now I am trying to run a demo code that once worked nicely. I'm on a Macbook, Bluetooth is turned on, but I get this:

noble: Could not start scanning, state is unknown (not poweredOn)

What can I do? 

Thx!
Sven

hansamann, 2015-07-23:
OK, very weird. One of your examples that scans for advertisments works. I've added the code, with a tiny change to react just for a special service:

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning(['a495ff10c5b14b44b5121370f02d74de']);
  } else {
    noble.stopScanning();
  }
});

The stateChange event is never fired. Do I need to kick it off somehow?

Cheers
Sven

sandeepmistry, 2015-07-26:
@hansamann how early on are you registering the event listener? Maybe it has changed to ```poweredOn``` before you've registered. You can check ```noble.state```.

Also, take a look at the debug logs: ```sudo DEBUG=* node```

sandeepmistry, 2015-08-02:
@hansamann any updates can we close?

versoul, 2015-08-04:
@sandeepmistry I have the same mistake on svezheustanovlenny system raspbian
noble status unknow allways
and
noble.state is unknow

versoul, 2015-08-04:
@sandeepmistry 
dont work on my laptop also (ubuntu 14) and (raspberry)

but bluetooth is works fine
```
versoul@versoul-laptop:~/workspace/racestory/tst$ hcitool scan
Scanning ...
	44:74:6C:8D:DC:24	Xperia Z3
versoul@versoul-laptop:~/workspace/racestory/tst$ sudo l2ping -c 1 44:74:6c:8d:dc:24
Ping: 44:74:6c:8d:dc:24 from 00:11:67:BA:03:2D (data size 44) ...
0 bytes from 44:74:6c:8d:dc:24 id 0 time 12.75ms
1 sent, 1 received, 0% loss
```

lsusb
```
Bus 003 Device 008: ID 1131:1001 Integrated System Solution Corp. KY-BT100 Bluetooth Adapter
```

sandeepmistry, 2015-08-10:
@versoul looks like a different issue to @hansamann's.

What do the following commands output for you? The commands you are running are for normal Bluetooth, not BLE/Bluetooth 4.0 ...

```sh
sudo hcitool lescan
```

```sh
sudo hciconfig hci0 version
```

versoul, 2015-08-10:
```
pi@raspberrypi ~ $ sudo hcitool lescan
Set scan parameters failed: Connection timed out
pi@raspberrypi ~ $ sudo hciconfig hci0 version
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:11:67:BA:03:2D  ACL MTU: 1021:8  SCO MTU: 48:10
	HCI Version: 2.0 (0x3)  Revision: 0x2c6
	LMP Version: 2.0 (0x3)  Subversion: 0x2c6
	Manufacturer: Integrated System Solution Corp. (57)

```

volschin, 2015-08-10:
Your BT Adapter doesn't seem to fullfill the BLE standard.

sandeepmistry, 2015-08-12:
@versoul I agree with @volschin assessment. The HCI version needs to be version ``4.0 (0x06)``:

```
pi@raspberrypi ~/git/noble $ sudo hciconfig hci0 version
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:19:0E:11:28:17  ACL MTU: 1021:8  SCO MTU: 64:1
	HCI Version: 4.0 (0x6)  Revision: 0x1000
	LMP Version: 4.0 (0x6)  Subversion: 0x220e
	Manufacturer: Broadcom Corporation (15)
```

sandeepmistry, 2015-08-12:
Closing initial issue due to inactivity ....

closed 2015-08-12 by sandeepmistry

dharmeshpurohit, 2016-09-02:
Hi Sandeep,

I am running example  as following:

var noble = require('noble')

noble.on('stateChange', function(state)
{

});

sandeepmistry, 2016-09-05:
Hi @hansamann,

In the state change event what is the value of `state`? Also, noble will only detect BLE devices not Bluetooth Classic.