noble/noble #220: The writeWithoutResponse on AlertLevel (1802) does not trigger anything on my pebblebee

npenin, 2015-07-26:
The writeWithoutResponse on AlertLevel (1802) does not trigger anything on my pebblebee whereas it claims to support writeWithoutResponse.

This does not work
```js
matchingCharac[0].write(data, false, function(error){
                                                debug('callback');
                                                debug(error);
                                            });
```

But this does.
```js
matchingCharac[0].write(data, true, function(error){
                                                debug('callback');
                                                debug(error);
                                            });
```

Is this an issue in the documentation or somewhere else ? https://github.com/sandeepmistry/noble/wiki/Getting-started#writing-a-characteristic

sandeepmistry, 2015-07-26:
@npenin it is a documentation issue, I've updated the getting started guide.

npenin, 2015-07-27:
Thank you very much for the udpate

closed 2015-07-27 by npenin