noble/noble #221: Ble-Bean installation

lgaleana, 2015-07-30:
Hi, I'm trying to install Ble-Bean, but I get the following error:

> > noble@0.3.14 install /Users/lgaleana/hackathons/quadcopter/nodejs/node_modules/ble-bean/node_modules/noble
> > node-gyp rebuild
> 
>  LINK(target) Release/hci-ble
> Undefined symbols for architecture x86_64:
>   "_main", referenced from:
>       start in crt1.10.5.o
> ld: symbol(s) not found for architecture x86_64
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make: *** [Release/hci-ble] Error 1
> gyp ERR! build error 
> gyp ERR! stack Error: `make` failed with exit code: 2
> gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
> gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
> gyp ERR! System Darwin 14.4.0
> gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /Users/lgaleana/hackathons/quadcopter/nodejs/node_modules/ble-bean/node_modules/noble
> gyp ERR! node -v v0.12.7
> gyp ERR! node-gyp -v v2.0.1
> gyp ERR! not ok 
> npm ERR! Darwin 14.4.0
> npm ERR! argv "node" "/usr/local/bin/npm" "install" "ble-bean"
> npm ERR! node v0.12.7
> npm ERR! npm  v2.12.1
> npm ERR! code ELIFECYCLE
> 
> npm ERR! noble@0.3.14 install: `node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR! 
> npm ERR! Failed at the noble@0.3.14 install script 'node-gyp rebuild'.
> npm ERR! This is most likely a problem with the noble package,
> npm ERR! not with npm itself.
> npm ERR! Tell the author that this fails on your system:
> npm ERR!     node-gyp rebuild
> npm ERR! You can get their info via:
> npm ERR!     npm owner ls noble
> npm ERR! There is likely additional logging output above.

OSX 10.10.4
Xcode 6.4
Node 0.12.7

gcc --version
> Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
> Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
> Target: x86_64-apple-darwin14.4.0
> Thread model: posix

sandeepmistry, 2015-08-02:
@lgaleana this has been fixed in a later version of noble, see https://github.com/sandeepmistry/noble/pull/205 However, ble-bean has some dependencies, that use an older version.

I suggest you try upgrading ble-bean to use [noble-device](https://github.com/sandeepmistry/noble-device) version 1.0.

cc/ @jacobrosenthal

closed 2015-08-02 by sandeepmistry

lgaleana, 2015-08-03:
What I did, before that, was downgrading my version of Node to .10 and npm to 2.10.1. I'll try what you said.