noble/noble #222: sensor tag fails to install - same fail to install noble

stumpygreg, 2015-07-31:
After reading all of the posting I thought this would have been easy.  Installed command line tools after completely updating   Xcode.  Can build noble separately and test.js sees sensor tag.  

Yosemite 10.10.4
Xcode 6.4

greglakomski@Gregs-MBP:~/Sensortag/iot-sensor-tag/publish$ g++ -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.4.0
Thread model: posix

0 info it worked if it ends with ok
1 verbose cli [ 'node', '/usr/local/bin/npm', 'install' ]
2 info using npm@1.0.0
3 info using node@v0.12.7
4 verbose readDependencies loading dependencies from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/package.json
5 warn package.json @ No repository field.
6 warn package.json @ No license field.
7 verbose install where, deps [ '/Users/greglakomski/Sensortag/iot-sensor-tag/publish',
7 verbose install   [ 'sensortag', 'mqtt', 'getmac', 'properties' ] ]
8 verbose install where, peers [ '/Users/greglakomski/Sensortag/iot-sensor-tag/publish', [] ]
9 verbose installManyTop reading for lifecycle /Users/greglakomski/Sensortag/iot-sensor-tag/publish/package.json
10 info preinstall @
11 verbose installManyTop reading scoped package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/getmac/package.json
12 verbose installManyTop reading scoped package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/mqtt/package.json
13 verbose installManyTop reading scoped package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/noble/package.json
14 verbose installManyTop reading scoped package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/properties/package.json
15 info package.json noble@1.0.2 No license field.
16 info package.json getmac@1.0.6 license should be a valid SPDX license expression
17 info package.json mqtt@1.0.5 No license field.
18 verbose readDependencies loading dependencies from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/package.json
19 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/getmac/package.json
20 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/mqtt/package.json
21 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/noble/package.json
22 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/properties/package.json
23 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/.bin/getmac-node/package.json
24 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/.bin/mqtt/package.json
25 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/.bin/mqtt_pub/package.json
26 verbose targetResolver reading package data from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/.bin/mqtt_sub/package.json
27 verbose already installed skipping mqtt@1.0.5 /Users/greglakomski/Sensortag/iot-sensor-tag/publish
28 verbose already installed skipping getmac@1.0.6 /Users/greglakomski/Sensortag/iot-sensor-tag/publish
29 verbose already installed skipping properties@1.2.1 /Users/greglakomski/Sensortag/iot-sensor-tag/publish
30 silly cache add args [ 'sensortag@0.1.9', null ]
31 verbose cache add spec sensortag@0.1.9
32 silly cache add parsed spec { raw: 'sensortag@0.1.9',
32 silly cache add   scope: null,
32 silly cache add   name: 'sensortag',
32 silly cache add   rawSpec: '0.1.9',
32 silly cache add   spec: '0.1.9',
32 silly cache add   type: 'version' }
33 silly addNamed sensortag@0.1.9
34 verbose addNamed "0.1.9" is a plain semver version for sensortag
35 silly mapToRegistry name sensortag
36 silly mapToRegistry using default registry
37 silly mapToRegistry registry https://registry.npmjs.org/
38 silly mapToRegistry uri https://registry.npmjs.org/sensortag
39 verbose addNameVersion registry:https://registry.npmjs.org/sensortag not in flight; fetching
40 verbose request uri https://registry.npmjs.org/sensortag
41 verbose request no auth needed
42 info attempt registry request try #1 at 12:28:27 PM
43 verbose request id a9e12dbdf6b98f34
44 verbose etag "1JCQDSFW008CDCU0B0DA0H2BT"
45 http request GET https://registry.npmjs.org/sensortag
46 http 304 https://registry.npmjs.org/sensortag
47 silly get cb [ 304,
47 silly get   { date: 'Fri, 31 Jul 2015 17:28:28 GMT',
47 silly get     via: '1.1 varnish',
47 silly get     'cache-control': 'max-age=60',
47 silly get     etag: '"1JCQDSFW008CDCU0B0DA0H2BT"',
47 silly get     age: '0',
[... 1142 lines cut ...]
631 error     node-gyp rebuild
631 error You can get their info via:
631 error     npm owner ls noble
631 error There is likely additional logging output above.
632 verbose exit [ 1, true ]
633 verbose unbuild node_modules/sensortag/node_modules/noble
634 info preuninstall noble@0.3.14
635 info uninstall noble@0.3.14
636 verbose unbuild rmStuff noble@0.3.14 from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules
637 verbose unbuild rmStuff in /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag/node_modules
638 info postuninstall noble@0.3.14
639 silly gentlyRm /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag/node_modules/noble is being purged from base /Users/greglakomski/Sensortag/iot-sensor-tag/publish
640 verbose gentlyRm don't care about contents; nuking /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag/node_modules/noble
641 silly vacuum-fs purging /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag/node_modules/noble
642 silly vacuum-fs removing /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag/node_modules
643 silly vacuum-fs quitting because other entries in /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag
644 verbose unbuild node_modules/sensortag
645 info preuninstall sensortag@0.1.9
646 info uninstall sensortag@0.1.9
647 verbose unbuild rmStuff sensortag@0.1.9 from /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules
648 info postuninstall sensortag@0.1.9
649 silly gentlyRm /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag is being purged from base /Users/greglakomski/Sensortag/iot-sensor-tag/publish
650 verbose gentlyRm don't care about contents; nuking /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag
651 silly vacuum-fs purging /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules/sensortag
652 silly vacuum-fs quitting because other entries in /Users/greglakomski/Sensortag/iot-sensor-tag/publish/node_modules

sandeepmistry, 2015-08-10:
@stumpygreg 2 oddities from the log snippet:

1) You are using a fairly old version of npm 1.0.0. According to the [change log](https://raw.githubusercontent.com/joyent/node/v0.12.7/ChangeLog) Node.js 0.12.7 ships with npm version ```2.11.3```

2) A very fairly old version of node-sensortag and noble are attempted to be installed, 0.1.9 and 0.3.14, 1.0.2 is the latest version for both of those.

I suggest you upgrade your npm and try again. Making sure version 1.0.2 of node-sensortag is the one selected for install.

sandeepmistry, 2015-08-18:
Closing due to lack of activity ...

closed 2015-08-18 by sandeepmistry