noble/noble #223: scanning seems stopped after connecting to 1 peripheral

npenin, 2015-08-01:
Hi,

I start a discover with duplicated, but as soon as I connect to a device to read its characteristics, it does not trigger the discover event anymore. I have also tried to disconnect from it without success.

Any idea ?

Thanks.

sandeepmistry, 2015-08-02:
@npenin some others had reported this issue, see: https://github.com/sandeepmistry/node-sensortag/issues/37

I think it has something to do with the Bluetooth 4.0 adapter hardware or firmware and/or kernel version.

What are you using?

I have no problems with the following:

```sh
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.0.7+ #802 PREEMPT Wed Jul 8 17:35:23 BST 2015 armv6l GNU/Linux

pi@raspberrypi ~ $ lsusb 
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. 
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 005: ID 0a5c:21e8 Broadcom Corp. 
```

npenin, 2015-08-02:
Hi,

Here is what I have :
```
pi@home ~ $ uname -a
Linux home 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux
pi@home ~ $ lsusb
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 040b:6543 Weltrend Semiconductor
Bus 001 Device 007: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 008: ID 0a5c:21e8 Broadcom Corp.
```

cristian-t, 2015-08-06:
Hello, I've dealing with this issue as well, and what I found out is that most hardware does not support scanning and maintaining a connection at the same time. This means you have to discover all your devices first and then connect to them. If you want to start the discovery process again, you would have to terminate all connections, discover any new devices, and then reconnect.

sandeepmistry, 2015-08-10:
@npenin thanks, looks like we are using similar USB adapters, but different kernel version.

@cristian-t thanks for the info.

After a device connects, can you try manually innovating scanning to see if works.

```sh
sudo hcitool lescan
```

If this works, I'm happy to put this logic in noble.

In addition, could you try the [bluetooth-hci-socket branch](https://github.com/sandeepmistry/noble/tree/bluetooth-hci-socket).

sandeepmistry, 2015-08-12:
The ```bluetooth-hci-socket``` branch has been merged into master, so you can try it from there now.

sandeepmistry, 2015-08-18:
@cristian-t @npenin any updates?

cristian-t, 2015-08-18:
I would love to keep trying stuff, but I am busy at the moment.
I tried a few things with the tools included in the linux build, `hci-ble` and `l2cap-ble`.
I started `hci-ble` and sent it a `USR1` signal to start scanning, no duplicates. At this point, I could see the advertisement data being received. The moment I launched `l2cap-ble`, `hci-ble` stopped outputting. Same thing happens if I send `USR2` to `hci-ble`. Running `sudo hcitool lescan` while `l2cap-ble` is running will work, and advertisement data will be received again.
I haven't tried the new branch, `bluetooth-hci-socket`, yet.

At the moment I successfully implemented the solution I mentioned above. I discover all first, then connect to all. When I want to discover again, I have to disconnect from all devices (stop all `l2cap-ble` instances) and only then, start discovering again. I might be worth mentioning that I am using your `node-sensortag` library to connect to multiple sensor tags.

sandeepmistry, 2015-08-19:
@cristian-t thanks for the update. Let me know when you have more time to look at this again. ```bluetooth-hci-socket``` is in master now, and released with in version ```1.1.0```.

If restarting scanning after a connection works, then we can update noble to do so.

npenin, 2015-08-21:
Sorry for the late reply. I'll try to take some time this week-end and let you know it solved the issue.

sandeepmistry, 2015-08-30:
@npenin did you get a chance to try it out?

sandeepmistry, 2015-09-20:
@npenin any updates?

sandeepmistry, 2015-09-28:
I'm going to close this due to lack of activity, I can't reproduce locally, and no one has been available to test my suggestions unfortunately ...

Please let me know if anyone has been able to try thing out ...

npenin, 2015-10-18:
Hi,

I was quite busy, and did not had a chance to test it. Sorry about that.

@cristian-t 's solution is working. Anyway, scanning still needs to be restarted manually after connection.

sandeepmistry, 2015-10-27:
@npenin got time to make pull request now?

npenin, 2015-10-27:
I acutally made a wrapper around noble, I have not changed noble by myself.

[https://avatars3.githubusercontent.com/u/1405160?v=3&s=400]<https://github.com/npenin>

npenin (Nicolas Penin) · GitHub
npenin has 66 repositories written in JavaScript, HTML, and C#. Follow their code on GitHub.
Lire la suite...<https://github.com/npenin>

-

mhle, 2016-01-19:
Hi Sandeep, I'm experiencing the same issue using Raspbian jessie on Noble 1.3.0. Some system info:
```
Bus 001 Device 005: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
Linux samcloud 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux
```
Can connect to 1 device then advertisement discovery stops. I tried using
```
sudo hcitool lescan
```
and discovery starts again. Will this be included in future version of Noble?

sandeepmistry, 2016-01-23:
@mhle 

>  Will this be included in future version of Noble? Currently getting around this by calling Noble.startScanning() after connect.

Yes if someone has a system that can be used for testing and can run tests :)

Bonus points for submitting a PR.

Essentially ```lib/hci-socket/bindings.js``` needs to track the current scanning state (enabled, uuid filter, duplicates), and then restart scanning if enabled with the same parameters.

Let me know if you have time for a PR.

Also, I'm curious to what the output of ``` sudo hcidump -t -x``` is while the ```sudo hciconfig hci0 reset``` command is on run your system, for reference here is the output from my Raspberry Pi:

```
2016-01-23 15:31:08.996559 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1000
    LMP Version: 4.0 (0x6) LMP Subversion: 0x220e
    Manufacturer: Broadcom Corporation (15)
```

ostrichjones, 2016-04-13:
I see the same issue described above when using an RPi3. As a reproducer this world have the advantage of embedded BT - fewer variables in terms of dongles.

mbifulco, 2016-08-17:
Hi @sandeepmistry - I've run into this issue, and have been able to test it out.  It looks like running `sudo hcitool lescan` will find the devices that haven't been discovered every time.  I'd love to work through fixing this with you, and happy to open a PR if I can figure out how to bake in the changes you were talking about.  

What's the best way to work together on this?

sandeepmistry, 2016-08-21:
@mbifulco thanks!

> What's the best way to work together on this?

We need to better understand why certain adapters / firmware behave differently and see if we can find a way to programmatically detect how the adapter/firmware will behave (maybe reading adapter features?).

The change to call start scanning again after a connection occurs is trivial but I prefer not to do this all the time. Since it could interfere with filtering out duplicates etc.

mbifulco, 2016-08-22:
Hm, okay.  I'm afraid I still don't really know how I can help.  Do you have a suggestion for what I might do as a workaround on a system where I know that the `lescan` call finds missing sensors?

mbifulco, 2016-08-22:
And, can we re-open this issue?

sandeepmistry, 2016-08-24:
@npenin work around is to call `noble.startScanning()` in the connect callback.

mbifulco, 2016-08-24:
I'll give it a shot, but I think that caused some trouble

Steckdoserich, 2016-08-30:
@sandeepmistry after I read your workaround I tried it at different places to fix this problem. But due to the amount of connect callbacks I came up with this workaround:

```
 function stateChanged(state) {
     if (state === 'poweredOn') {
         noble.on('scanStop', function () {
             setTimeout(function () {
                 noble.startScanning([], true, function (err) {
                     if (err) {
                         console.log("startScanning error: " + err);
                     }
                 });
             },2500);
         });
         noble.stopScanning();
         noble.startScanning([], true, function (err) {
             if (err) {
                 console.log("startScanning error: " + err);
             }
         });
     }
     else {
         noble.stopScanning();
     }
 }
```

For the moment this works fine. I am listening for the scanning to stop with your event `scanStop`. Then I just restart the scanning. Therefor, I need to add a `setTimeout` for the next `startScanning`, because otherwise it sometimes doesn't start scanning again.

I am pretty sure this isn't the best way to handle this problem, but I don't know another one.

zeroepoch, 2016-12-11:
This problem still exists on master.  I'm using a `0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth` adapter on my Fedora 25 desktop and as soon as I connect scanning stops.  As @Steckdoserich mentioned it does trigger the 'scanStop' event so it seems to actually be stopping the scan itself cleanly.  In that event callback I run `noble.startScanning` again and it continues to issue discover events.  I even get discover events for the connected device.  I have to keep a flag so it will stop scanning again when I really want it to stop.