noble/noble #224: Characteristics read not working from within a setTimeout callback

jainanshul, 2015-08-03:
```
        let readRssi = function() {
            console.log('reading rssi: ' + rssiChar.uuid);
            rssiChar.read();
            rssiChar.once('data', data => {
              console.log(`read rssi char successfully ${-data.readUInt8(0)}`);
            });
          };

       // This doesn't work
       let updateRssiInterval = setTimeout(() => readRssi(), 500);

       // This works
       readRssi();
```

jainanshul, 2015-08-06:
FYI this is happening on OSX Yosemite.

sandeepmistry, 2015-08-10:
@jainanshul please provide a full code snippet, with debug output:

```sh
DEBUG=* node <file>.js
```

andySigler, 2015-08-17:
Same problem here. Here's my code:

```node
var noble = require("noble");

noble.on('stateChange', function(state) {
	if(state==='poweredOn') {
		noble.startScanning();
	}
	else{
		console.log(state);
	}
});

var myUUID = '6bc4051e36ca44978e2d95b4ee149e46';

var myPeripheral = undefined;
var myChar = undefined;

noble.on('discover',function(peripheral){

	if(peripheral.uuid === myUUID) {

		myPeripheral = peripheral;

		noble.stopScanning();

		myPeripheral.connect(function(error){

			if(error) {
				console.log(error);
				myPeripheral.disconnect();
			}

			else {

				myPeripheral.discoverAllServicesAndCharacteristics(function(error,services,chars){

					if(error) console.log(error);

					else {

						for(var c=0;c<services.length;c++){

							var theseChars = services[c].characteristics

							for(var i=0;i<theseChars.length;i++){

								if(theseChars[i].uuid==='1111') {

									myChar = theseChars[i];

									setTimeout(function(){
										myPeripheral.connect(function(){
											myChar.read(function(error,data){
												console.log(arguments); // returns [null , undefined]
												myPeripheral.disconnect();
											});
										});
									},1000);
								}
							}
						}
					}
					myPeripheral.disconnect();;
				});
			}
		});
	}
});
[... 1183 lines cut ...]
      "type": "Buffer",
      "data": [
        107,
        196,
        5,
        30,
        54,
        202,
        68,
        151,
        142,
        45,
        149,
        180,
        238,
        20,
        158,
        70
      ]
    },
    "kCBMsgArgResult": 0
  }
} +41ms
  yosemite-bindings peripheral 6bc4051e36ca44978e2d95b4ee149e46 disconnected +0ms
```

sandeepmistry, 2015-08-17:
@andySigler 

```
yosemite-bindings xpcEvent: {
  "kCBMsgId": 71,
  "kCBMsgArgs": {
    "kCBMsgArgCharacteristicHandle": 47,
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        107,
        196,
        5,
        30,
        54,
        202,
        68,
        151,
        142,
        45,
        149,
        180,
        238,
        20,
        158,
        70
      ]
    },
    "kCBMsgArgResult": 21
  }
```

Result is non-sucess/non-zero. Similar to https://github.com/sandeepmistry/noble/issues/207 error events are currently not propagated.

Can you please try calling ``discoverAllServicesAndCharacteristics`` before your call to ``read`` I believe OS X's blued requires this ...

andySigler, 2015-08-17:
Now `characteristic.read()` stalls after call `peripheral.discoverAllServicesAndCharacteristics()` before:

```node
var noble = require("noble");

noble.on('stateChange', function(state) {
	if(state==='poweredOn') {
		noble.startScanning();
	}
	else{
		console.log(state);
	}
});

var myUUID = 'e732749b0fd1451e957f26e9aa9ddd6e';

var myPeripheral = undefined;
var myChar = undefined;

noble.on('discover',function(peripheral){

	console.log(peripheral.uuid);
	console.log('\t'+peripheral.advertisement.localName);

	if(peripheral.uuid === myUUID) {

		myPeripheral = peripheral;

		noble.stopScanning();

		myPeripheral.connect(function(error){

			if(error) {
				console.log(error);
				myPeripheral.disconnect();
			}

			else {

				myPeripheral.discoverAllServicesAndCharacteristics(function(error,services,chars){

					if(error) console.log(error);

					else {

						for(var c=0;c<services.length;c++){

							var theseChars = services[c].characteristics

							for(var i=0;i<theseChars.length;i++){

								if(theseChars[i].uuid==='1111') {

									myChar = theseChars[i];

									setTimeout(function(){
										myPeripheral.connect(function(){
											myPeripheral.discoverAllServicesAndCharacteristics(function(error,services,chars){

												console.log(error); // null

												console.log('.read() is called without error, but it stalls here');

												myChar.read(function(error,data){

													console.log('and this never prints');

													console.log(arguments);
													myPeripheral.disconnect();
												});
[... 1906 lines cut ...]
    "kCBMsgArgSupervisionTimeout": 400,
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        231,
        50,
        116,
        155,
        15,
        209,
        69,
        30,
        149,
        127,
        38,
        233,
        170,
        157,
        221,
        110
      ]
    }
  }
} +5s
```

andySigler, 2015-08-17:
Maybe I should add that reading a characteristic within `setTimeout` worked until this morning, when I updated to Yosemite.

sandeepmistry, 2015-08-20:
@andySigler you are actually disconnecting the peripheral before reading the characteristic. 

Try something like this:

_Note_: you will have to change the peripheral UUID and characteristic back to the values you had before.

```javascript
var noble = require("./index");

noble.on('stateChange', function(state) {
    if(state==='poweredOn') {
        noble.startScanning();
    }
    else{
        console.log(state);
    }
});

var myUUID = '3d7bd14925ad4f69b73a98cbc998c4db';

var myPeripheral = undefined;
var myChar = undefined;

noble.on('discover',function(peripheral){
console.log(peripheral.uuid);
console.log(peripheral.advertisement);

    if(peripheral.uuid === myUUID) {

        myPeripheral = peripheral;

        noble.stopScanning();

        myPeripheral.connect(function(error){

            if(error) {
                console.log(error);
                myPeripheral.disconnect();
            }

            else {

                myPeripheral.discoverAllServicesAndCharacteristics(function(error,services,chars){

                    if(error) console.log(error);

                    else {
                        var charFound = false;

                        for(var c=0;c<services.length;c++){

                            var theseChars = services[c].characteristics

                            for(var i=0;i<theseChars.length;i++){

                                if(theseChars[i].uuid==='2a26') {

                                    charFound = true;

                                    myChar = theseChars[i];

                                    setTimeout(function(){
                                        myPeripheral.connect(function(){
                                            myChar.read(function(error,data){
                                                console.log(arguments); // returns [null , undefined]
                                                myPeripheral.disconnect();
                                            });
                                        });
                                    },1000);
                                }
                            }
                        }

                        if (!charFound) {
                            // NEW: only disconnect if peripheral does not have the characteristic we want to read from
                            peripheral.disconnect();
                        }
                    }

                });
            }
        });
    }
});
```

I was getting the following result for the read:

```
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 71,
  "kCBMsgArgs": {
    "kCBMsgArgCharacteristicHandle": 23,
    "kCBMsgArgDeviceUUID": [
      61,
      123,
      209,
      73,
      37,
      173,
      79,
      105,
      183,
      58,
      152,
      203,
      201,
      152,
      196,
      219
    ],
    "kCBMsgArgResult": 21
  }
} +1ms
```

kCBMsgArgResult 21 must mean the peripheral is not connected. noble currently does not propagate non-zero result codes, this work will be tracked in https://github.com/sandeepmistry/noble/issues/207

andySigler, 2015-08-20:
Thank you, but I am purposefully disconnecting. My real application requires short connection times, along with user-initiated characteristic reads/writes on previously discovered peripherals.

Regardless of the test script I posted, reading characteristics within a `setTimeout` stopped working on osx 10.10 while they did work on 10.9, and I don't know if #207 address this.

sandeepmistry, 2015-08-20:
@andySigler but how can you read from a disconnected peripheral?

Reading from a setTimeout works for me with the modified code I posted. Do you have another example I can try locally?

andySigler, 2015-08-20:
I'm sorry for the confusion. You will see that my characteristic interactions have been inside of the `.connect()` callback, so I have assumed I've been connected. Here is a snippet from the test script I posted above:

```node
if(theseChars[i].uuid==='1111') {

    myChar = theseChars[i];

    setTimeout(function(){
        myPeripheral.connect(function(){
            myChar.read(function(error,data){
                console.log(arguments); // returns [null , undefined]
                myPeripheral.disconnect();
            });
        });
    },1000);
}
```

I will try another test shortly and post my results. Thank you again.

sandeepmistry, 2015-08-20:
@andySigler my bad, I was using a SensorTag, I needed to increase the timeout value.

I'm now seeing the event being propagated from:
1) https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js#L518
2) https://github.com/sandeepmistry/noble/blob/master/lib/noble.js#L281

But it never reaches, this:

https://github.com/sandeepmistry/noble/blob/master/lib/characteristic.js#L39

Needs more investigation ...

sandeepmistry, 2015-08-20:
Ok, so everytime you call discover services and characteristics, the reference to your previous referenced characteristic becomes stale. It can still perform actions, but will never listen to new events (because a new characteristic is created).

This works for me now ...
```javascript
var noble = require("./index");

noble.on('stateChange', function(state) {
    if(state==='poweredOn') {
        noble.startScanning();
    }
    else{
        console.log(state);
    }
});

var myUUID = '3d7bd14925ad4f69b73a98cbc998c4db';

var myPeripheral = undefined;
var myChar = undefined;

noble.on('discover',function(peripheral){

    console.log(peripheral.uuid);
    console.log('\t'+peripheral.advertisement.localName);

    if(peripheral.uuid === myUUID) {

        myPeripheral = peripheral;

        noble.stopScanning();

        myPeripheral.connect(function(error){

            if(error) {
                console.log(error);
                myPeripheral.disconnect();
            }

            else {

                myPeripheral.discoverAllServicesAndCharacteristics(function(error,services,chars){

                    if(error) console.log(error);

                    else {

                        for(var c=0;c<services.length;c++){

                            var theseChars = services[c].characteristics

                            for(var i=0;i<theseChars.length;i++){

                                if(theseChars[i].uuid==='2a26') {

                                    myChar = theseChars[i];

                                    setTimeout(function(){
                                        myPeripheral.connect(function(){
                                            myPeripheral.discoverAllServicesAndCharacteristics(function(error,services,chars){

                                                    for(var c=0;c<services.length;c++){

                                                        var theseChars = services[c].characteristics

                                                        for(var i=0;i<theseChars.length;i++){

                                                            if(theseChars[i].uuid==='2a26') {

                                                                myChar = theseChars[i];

                                                                console.log(error); // null

                                                                console.log('.read() is called without error, but it stalls here');

                                                                myChar.read(function(error,data){

                                                                    console.log('and this never prints');

                                                                    console.log(arguments);
                                                                    myPeripheral.disconnect();
                                                                });

                                                            }
                                                        }
                                                    }
                                                });
                                            });
                                    },10000);
                                }
                            }
                        }
                    }
                    myPeripheral.disconnect();;
                });
            }
        });
    }
});

```

Basically re-discovering the characteristic you wanted to read from.

noble needs to be updated to make sure the old characteristic reference is stale, and report an error.

andySigler, 2015-08-20:
Ok great, just ran this and works. Funny how the characteristics didn't go stale before.

Anyways thanks for the fast help and great library!!

sandeepmistry, 2015-08-20:
@andySigler I think the stale issue would still existing on non-Yosemite systems.

OS X 10.10 now forces you to re-discover services and characteristics after disconnecting and re-connecting.

sandeepmistry, 2015-08-30:
Closing due to lack of activity from initial reporter ...

closed 2015-08-30 by sandeepmistry