noble/noble #225: Declaring the `disconnect` callback has different functionality on different macs

taktran, 2015-08-08:
When we run `noblePeripheral.disconnect(callback)`, the callback is called on both a osx 10.8.5 and osx 10.10.3 machine, but on the newer osx 10.10.3 machine, our bluetooth hardware turns off after the `disconnect`, but on the osx 10.8.5 machine, it disconnects but doesn't turn the hardware off.

If we remove the `callback`, on both machines, the `disconnect` trigger doesn't turn the hardware off (which is what we want). Do you have any idea why?  It seems really weird that this discrepancy is affected by a declaration of a callback.

I'm looking inside [legacy.js](https://github.com/sandeepmistry/noble/blob/master/lib/mac/legacy.js#L199) for osx 10.8.5 and [yosemite.js](https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js#L224), but can't seem to find where the callbacks are declared.

sandeepmistry, 2015-08-10:
@taktran I haven't experienced this myself.

Can you please provide a short code snippet I can test on my Mac running 10.10.4?

sandeepmistry, 2015-08-18:
Closing due to lack of activity ...

closed 2015-08-18 by sandeepmistry