noble/noble #226: Build Failure on Mac

theideasmith, 2015-08-12:
## Problem

I'm really excited about the potential for this library, especially for using my brother's bluetooth controlled car, which has a bluetooth "API". 

I've ran `npm install noble` and am hit with the error messages found below. I have read some issues on similar issues, and tried to do what they suggest, but to no avail. Here is my info. 
**Running `which gcc` returns** `/usr/bin/gcc`
**Running `gcc -v` returns**
```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

**and `xcode-select -v` returns** `xcode-select version 2339`

## Error Message

```
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.3.2
-
> xpc-connection@0.1.3 install /Users/akivalipshitz/Desktop/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
In file included from ../src/XpcConnection.cpp:7:
In file included from ../src/XpcConnection.h:10:
../node_modules/nan/nan.h:324:27: error: redefinition of 'NanEnsureHandleOrPersistent'
  NAN_INLINE v8::Local<T> NanEnsureHandleOrPersistent(const v8::Local<T> &val) {
                          ^
../node_modules/nan/nan.h:319:17: note: previous definition is here
  v8::Handle<T> NanEnsureHandleOrPersistent(const v8::Handle<T> &val) {
                ^
../node_modules/nan/nan.h:344:27: error: redefinition of 'NanEnsureLocal'
  NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Handle<T> &val) {
                          ^
../node_modules/nan/nan.h:334:27: note: previous definition is here
  NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Local<T> &val) {
                          ^
../node_modules/nan/nan.h:757:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:768:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/akivalipshitz/.node-gyp/3.0.0/include/node/node_buffer.h:35:40: note: candidate function not
      viable: no known conversion from 'uint32_t' (aka 'unsigned int') to 'enum encoding' for 3rd
      argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/akivalipshitz/.node-gyp/3.0.0/include/node/node_buffer.h:47:40: note: candidate function not
      viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/akivalipshitz/.node-gyp/3.0.0/include/node/node_buffer.h:32:40: note: candidate function not
      viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/akivalipshitz/.node-gyp/3.0.0/include/node/node_buffer.h:40:40: note: candidate function not
      viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/XpcConnection.cpp:7:
In file included from ../src/XpcConnection.h:10:
../node_modules/nan/nan.h:772:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to
      'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/akivalipshitz/.node-gyp/3.0.0/include/node/v8.h:210:7: note: candidate constructor
      (the implicit copy constructor) not viable: no known conversion from
[... 87 lines cut ...]
      'v8::Local<v8::Object>::Local<v8::Value>' requested here
    return NanEscapeScope(
           ^
../node_modules/nan/nan.h:483:30: note: expanded from macro 'NanEscapeScope'
# define NanEscapeScope(val) scope.Escape(Nan::imp::NanEnsureLocal(val))
                             ^
10 errors generated.
make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/akivalipshitz/.nvm/versions/io.js/v3.0.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "/Users/akivalipshitz/.nvm/versions/io.js/v3.0.0/bin/iojs" "/Users/akivalipshitz/.nvm/versions/io.js/v3.0.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/akivalipshitz/Desktop/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v3.0.0
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok 
npm WARN optional dep failed, continuing xpc-connection@0.1.3
noble@1.1.0 node_modules/noble
├── bplist-parser@0.0.6
└── debug@2.2.0 (ms@0.7.1)
```

theideasmith, 2015-08-12:
I've solved the problem. I was using `iojs`, not `node`. Don't know why that makes a difference, but it did for me.

sandeepmistry, 2015-08-13:
@theideasmith which version of ```iojs``` did you try with?

Please open an issue here: https://github.com/sandeepmistry/node-xpc-connection/issues and close this one.

```node-xpc-connection``` probably needs to be updated to use a new version of [nan](https://github.com/nodejs/nan) 2.0 or later based on the [changelog](https://github.com/nodejs/nan/blob/master/CHANGELOG.md#200-jul-31-2015). It is currently using 1.7.0.

sandeepmistry, 2015-08-30:
This should be resolved by https://github.com/sandeepmistry/node-xpc-connection/commit/9009af97ecbdb5130f42fe2a55ad10a436606ce3 which is included with v0.1.4 of node-xpc-connection

closed 2015-08-30 by sandeepmistry