noble/noble #227: Advertising Event Type

carlst, 2015-08-12:
Could the LE Advertising Report `Event_Type` be passed up to the `discover` handler?

This is useful for deciding whether or not to attempt a connection (e.g. a Beacon is non-connectable).

sandeepmistry, 2015-08-13:
@carlst would something like a ```peripheral.connectable``` property work? It would be set to ``true`` or ``false`` ...

carlst, 2015-08-13:
Yes, but for future expansion, it may also be helpful to retain other `Event_Type` information (undirected/directed, scannable).

I modified the pre-1.1.0 version for testing; `node-bluetooth-hci-socket` is a really nice addition, and will presumably make this change much cleaner.

sandeepmistry, 2015-08-14:
Let's stick to ```.connectable``` for now, as OS X (10.10) provides this flag on discovery as well.

sandeepmistry, 2015-09-10:
FYI, WIP for this can be found here: https://github.com/sandeepmistry/noble/tree/peripheral.connectable

Only Linux, Windows, and OS X 10.10 or above will have the property set to true or false ...

sandeepmistry, 2015-09-12:
```peripheral.connectable``` branch has been merged into master, so this feature will be in the next release ...

closed 2015-09-12 by sandeepmistry