noble/noble #228: Could not start scanning, state is unknown (not poweredOn)

TorstenRobitzki, 2015-08-13:
Hello,
I've tried to scan for BLE devices:

    noble = require('noble')

    noble.startScanning();
    console.log "Scanning"

this results in a error message (?):

    noble: Could not start scanning, state is unknown (not poweredOn)
    Scanning

Is the code wrong or is the message wrong? Is intended, that a message is printed on the screen? Shouldn't an exception being thrown if a precondition is not met?

I've tried 1.1.0 on OS/X 10.10

Cheers,
Torsten

carlst, 2015-08-13:
@TorstenRobitzki: you might try waiting for the state to become `poweredOn`:

`noble.on('stateChange', function(state) {
  if (state === 'poweredOn')
    noble.startScanning();
  else
    noble.stopScanning();
});`

TorstenRobitzki, 2015-08-13:
@carlst that works for me. Thank you! Is it intended to work this way? I mean I didn't plunged in or removed an BLE adapter.

carlst, 2015-08-13:
@TorstenRobitzki: Yes, that is per `noble` spec. Perhaps needed to wait until any BLE device initialization or setup done by `noble` has finished, and the BLE device is ready for scanning to be initiated.

You might take a look at some of the examples, such as `examples/peripheral-explorer.js`

They are useful to better understand the control flow (scanning, connecting, discovering Services, discovering Characteristics, and so on).

TorstenRobitzki, 2015-08-13:
@carlst Ok, thank you for the details.

closed 2015-08-13 by TorstenRobitzki

sandeepmistry, 2015-08-14:
@TorstenRobitzki do you still think an exception should be thrown? Happy to put that change in.

@carlst thanks for helping out!

TorstenRobitzki, 2015-08-14:
Hi @sandeepmistry,
if I've misused the API and the API is able to detect this misuse, I think an exception would be appropriate. It's harder to ignore but it would change the API and thus could break existing code.

sandeepmistry, 2015-08-14:
Done in https://github.com/sandeepmistry/noble/commit/dd152fd3901470140ed854b4d96add5e303d58c7

TorstenRobitzki, 2015-08-14:
@sandeepmistry wow, that was fast ;-)

quique123, 2018-07-29:
Im trying to read a ble char from a ble relay board from my rpi3.  startScanning() simply hangs or returns error that ble device isnt powered on.  Why is that?  

I know the ble on the rpi3 works because I use it to write to the relay board every so often and it works fine.