noble/noble #229: Cancel a connection

carlst, 2015-08-13:
Would it make sense to add `peripheral.cancel` (presumably invoking a lower-level `cancelLeConn()`)?

I am calling `peripheral.connect` but the device does not respond, and the call does not timeout.

Calling `peripheral.disconnect` (e.g. after a connection timer expires) does not work, presumably because there is no connection to disconnect.

FWIW, I did a quick mod, calling `cancelLeConn` after a connection timer expires, and it seems to work fine. After calling it, an `onconnect` event occurs and `peripheral.state` is `error` (no subsequent `ondisconnect` event occurs, as expected).

**Note:** above tested only on Windows (WinUSB), which is great work(!) by the way, for those of us using that platform.

sandeepmistry, 2015-08-14:
@carlst good suggestion. I'll have to make sure this lines up with the current OS X behaviour.

It would help with #183 as well.

KWolfe81, 2015-10-26:
carlst,

Can you share your mod?  I'm in the exact same situation and can't find out how to cancel a connect that never succeeds.

carlst, 2015-10-27:
I only applied the mod on Windows. Setting a timer and using <code>peripheral.disconnect</code> seemed to work on Linux although I am still using Noble 1.0.2 there (I have not tried Mac).

3 Noble files are affected (sorry for the ugliness):

<code>lib/noble.js</code>
Add:
```
Noble.prototype.cancelLeConn = function() {
  this._bindings.cancelLeConn();
};
```

<code>lib/hci-socket/bindings.js</code>
Add:
```
NobleBindings.prototype.cancelLeConn = function() {
  this._hci.cancelLeConn();
};
```

<code>lib/hci-socket/hci.js</code>
Add:
```
var OCF_LE_CANCEL_CONN = 0x000e;
var LE_CANCEL_CONN_CMD = OCF_LE_CANCEL_CONN | OGF_LE_CTL << 10;

Hci.prototype.cancelLeConn = function() {
  var cmd = new Buffer(4);

  // header
  cmd.writeUInt8(HCI_COMMAND_PKT, 0);
  cmd.writeUInt16LE(LE_CANCEL_CONN_CMD, 1);

  // length
  cmd.writeUInt8(0x0, 3);

  debug('cancel le conn - writing: ' + cmd.toString('hex'));
  this._socket.write(cmd);
};
```

chumager, 2016-04-26:
Hi, would this be implemented in some near future?

I need it badly cause I work with LOTS of BLEs and some time they ain't connect properly and the connection process seems to be synchronous (from the central device view) so when others peripherals get into the queue, my program fails, cause no one else can connect.

I've already patched the nombre library, but I'd like to use the official release.

Regards.

sandeepmistry, 2016-05-01:
@chumager care to share your patch? Also, do you have a debug trace and HCI dump capture of the failure case?

chumager, 2016-05-02:
I used the patch from @carlst, I need to work with dozens of BLEs, so, to avoid problems I used a Async Queue, with one process at the time, with this I avoid the problem of trying to connect a BLE while other is doing the same, my HCI dump shows correctness in the process, I don't know if some of my code will help you to do it better.

In the mean time I can tell you I use the patch before I start scanning to avoid the HCI state of trying to connect after mi code stops and stay doing it, I can stop the HCI to trying.

esinyor, 2016-07-13:
@chumager I'd like to do something similar. When you mention Async Queue, are you referring to the queue included in this https://github.com/caolan/async ?

If so, care to give some details on what you're doing? Are you queueing calls to peripheral.connect()?

chumager, 2016-07-14:
Hi @esinyor, tell me what do you want to do, maybe a pv message, so I can help u with. 

Regards

relevante, 2016-08-05:
@carlst,

Thanks for the cancel connection code. It's working well for me on Linux (Raspbian). I forked the repo and added the changes here in case it's helpful for anyone:

https://github.com/relevante/noble

And thanks @sandeepmistry for the awesome library!

mattgol, 2017-02-22:
Thanks for this code snippet! This really made the difference for me! I added the code a while a go and my application now runs smoothly over many days. 
I didn't put the cancelation code into a separate function but instead threw it into 'disconnect'. This way the Linux version behaves the same as on OSX where disconnect and cancel use the same call.
@sandeepmistry Do you think this would be a worthy addition? We would get exactly the same behavior on Linux and OSX this way.

sandeepmistry, 2017-03-13:
@mattgol 

> Do you think this would be a worthy addition? We would get exactly the same behavior on Linux and OSX this way (I tested on a Raspberry and Mac Mini).

For sure! However, the linked suggestions don't clear up noble's internal connection queue on Linux as far as I can tell. Pull requests welcome.

mattgol, 2017-03-13:
Thanks for the hint! Removing a device from the connection queue shouldn't be a big issue. I'll prepare a PR.

Jaftem, 2018-02-10:
@sandeepmistry Can we get this cancel LE connection function on an official release?

hothsolo, 2023-01-25:
Any further help on this?  I can't get peripheral.disconnect to cancel (even tho the docs say it should).  It never throws the "disconnect" to peripheral.once and from that point on I can't connect to anything, even other devices..
I couldn't get Carl's code to work either.  Same thing (tho I didn't expect the disconnect to throw to the peripheral).
I'm on a Raspberry Pi4.  

Also one very weird thing.  I set up a setTimeout right before I connect for like 4 seconds - that's what should make it bail.  I'm not sure what's happening, but it's *not* running right away.  It runs for like 30-40 seconds before finally triggering.

Cj

rzr, 2024-02-06:
May you forward the issue to @abandonware noble until upstream gives feedback at:

https://github.com/noble/noble/issues/656