noble/noble #230: Advertising report does not include scan response

carlst, 2015-08-13:
In 1.0.2, Advertising reports included scan response data (if `Event_Type 0x04`).

In 1.1.0, Advertising reports seem to include only initial Advertising packet information (i.e. no `peripheral.advertisement.localName`).

Perhaps a specific BLE scan request command needs to be executed(?).

**Note:** tested only on Windows 7 (WinUSB).

sandeepmistry, 2015-08-17:
@carlst can you please provide DEBUG output

```sh
set DEBUG=*
node examples/advertisement-discovery.js
```

Which USB adapter are you using?

Active scanning is used, see: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L197

carlst, 2015-08-17:
After more checking, I believe I found my basic mistake. Sorry for the false alarm.

I am using a Cirago BTA8000 (CSR).

The Advertising packet has `Event_Type 3 (Non connectable undirected advertising (ADV_NONCONN_IND)` and so there is no scan request/response. 

Several different types of sensors were being tested (some connectable, some not). Connectable ones do send `localName` (in the scan response) as expected.

Just for reference, here is the output for a non-connectable sensor (iBeacon):

    c:\npm\node_modules\noble\examples>node advertisement-discovery.js
    hci-usb reset +0ms
    hci-usb write: 01030c00 +0ms
    hci-usb HCI event: 0e04010c2000 +10ms
    hci onSocketData: 040e04010c2000 +0ms
    hci   event type = 4 +10ms
    hci   sub event type = 14 +0ms
    hci           cmd = 8204 +10ms
    hci           status = 0 +0ms
    hci           result =  +10ms
    hci-usb HCI event: 0e0401030c00 +80ms
    hci-usb reset complete +0ms
    hci onSocketData: 040e0401030c00 +0ms
    hci   event type = 4 +10ms
    hci   sub event type = 14 +10ms
    hci           cmd = 3075 +10ms
    hci           status = 0 +0ms
    hci           result =  +10ms
    hci setting filter to: 1400000020c10000000000400000 +850ms
    hci set event mask - writing: 01010c08fffffbff07f8bf3d +0ms
    hci-usb write: 01010c08fffffbff07f8bf3d +0ms
    hci set le event mask - writing: 010120081f00000000000000 +10ms
    hci-usb write: 010120081f00000000000000 +0ms
    hci read local version - writing: 01011000 +10ms
    hci-usb write: 01011000 +0ms
    hci read bd addr - writing: 01091000 +20ms
    hci-usb write: 01091000 +0ms
    hci-usb HCI event: 0e0401010c00 +10ms
    hci onSocketData: 040e0401010c00 +0ms
    hci   event type = 4 +10ms
    hci   sub event type = 14 +0ms
    hci           cmd = 3073 +10ms
    hci           status = 0 +10ms
    hci           result =  +0ms
    hci-usb HCI event: 0e0401012000 +10ms
    hci onSocketData: 040e0401012000 +0ms
    hci   event type = 4 +20ms
    hci   sub event type = 14 +0ms
    hci           cmd = 8193 +0ms
    hci           status = 0 +10ms
    hci           result =  +10ms
    hci-usb HCI event: 0e0c0101100006bb22060a00bb22 +0ms
    hci onSocketData: 040e0c0101100006bb22060a00bb22 +10ms
    hci   event type = 4 +0ms
    hci   sub event type = 14 +20ms
    hci           cmd = 4097 +0ms
    hci           status = 0 +0ms
    hci           result = 06bb22060a00bb22 +10ms
    hci set scan enabled - writing: 010c20020001 +10ms
    hci-usb write: 010c20020001 +0ms
    hci set scan parameters - writing: 010b200701100010000000 +10ms
    hci-usb write: 010b200701100010000000 +0ms
    hci-usb HCI event: 0e0a01091000895b06dc1b00 +20ms
    hci onSocketData: 040e0a01091000895b06dc1b00 +0ms
    hci   event type = 4 +0ms
    hci   sub event type = 14 +10ms
    hci           cmd = 4105 +10ms
    hci           status = 0 +0ms
    hci           result = 895b06dc1b00 +10ms
    hci address = 00:1b:dc:06:5b:89 +10ms
[... 41 lines cut ...]
    hci-usb HCI event: 3e2a02010300644e00eef30c1e020106 +120ms
    hci-usb HCI event: 1aff4c0002152f234454cf6d4a0fadf2 +0ms
    hci-usb HCI event: f4911ba9ffa600010001b4b6 +10ms
    hci onSocketData: 043e2a02010300644e00eef30c1e0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa600010001b4b6 +0ms
    hci   event type = 4 +10ms
    hci   sub event type = 62 +0ms
    hci           LE meta event type = 2 +10ms
    hci           LE meta event status = 1 +10ms
    hci           LE meta event data = 0300644e00eef30c1e0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa600010001b4b6 +0ms
    hci                   type = 3 +10ms
    hci                   address = 0c:f3:ee:00:4e:64 +0ms
    hci                   address type = public +20ms
    hci                   eir = 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa600010001b4 +0ms
    hci                   rssi = -74 +0ms
    gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,2,21,47,35,68,84,207,109,74,15,173,242,244,145,27,169,255,166,0,1,0,1,180]},"serviceData":[],"serviceUu
  ids":[]} +10ms
  peripheral discovered (0cf3ee004e64 with address <0c:f3:ee:00:4e:64, public>, RSSI -74:
	  hello my local name is:
		  undefined
	  can I interest you in any of the following advertised services:
		  []
	  here is my manufacturer data:
		  "4c0002152f234454cf6d4a0fadf2f4911ba9ffa600010001b4"

    hci-usb HCI event: 3e2a02010300644e00eef30c1e020106 +120ms

sandeepmistry, 2015-08-17:
@carlst no worries, I'll close then.

closed 2015-08-17 by sandeepmistry