noble/noble #231: Unable to discover services/characteristics with peripheral-explorer.js in Linux using 1.1.0

deadprogram, 2015-08-14:
I was unable to run the peripheral-explorer.js example using Noble 1.1.0 when running in Linux for a peripherals that work under 1.0.2

I was able to run peripheral-explorer.js under 1.0.2 by changing line 15 from 

```
if (peripheral.id === peripheralId) {
```

to

```
if (peripheral.uuid === peripheralId) {
```

Here is the output I received under each version:

Running using noble 1.0.2

```
$ sudo node demos/noble-explorer.js d03972c99b39
peripheral with ID d03972c99b39 found
  Local Name        = ronbean
  TX Power Level    = 4
  Service Data      = 
  Service UUIDs     = a495ff10c5b14b44b5121370f02d74de

services and characteristics:
f000ffc004514000b000000000000000
  f000ffc104514000b000000000000000 (Img Identify)
    properties  writeWithoutResponse, write, notify
  f000ffc204514000b000000000000000 (Img Block)
    properties  writeWithoutResponse, write, notify
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       726f6e6265616e | 'ronbean'
  2a01 (Appearance)
    properties  read
    value       0000 | ''
  2a02 (Peripheral Privacy Flag)
    properties  read, write
    value       00 | ''
  2a03 (Reconnection Address)
    properties  write
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       5000a0000000e803 | 'P h'
...
```

Running under Noble 1.1.0

```
$ sudo node demos/noble-explorer.js d03972c99b39
peripheral with ID d03972c99b39 found
  Local Name        = ronbean
  TX Power Level    = 4
  Service Data      = 
  Service UUIDs     = a495ff10c5b14b44b5121370f02d74de

services and characteristics:
```

That was all the output for 1.1.0, the program terminated immediately without listing any services/characteristics.

sandeepmistry, 2015-08-17:
@deadprogram I'm going to need some more info to help you out.

 * Can you please provide ```hcidump``` capture? Output preferably in a Gist: ```sudo hcidump -t -x```
 * Are you using a Punch Through Bean? I just tried it with one of mine, no issues connecting.

Btw, ```peripheral.id``` was introduced in v1.1.0, so the example is not backwards compatible with older releases. See [change log](https://github.com/sandeepmistry/noble/blob/master/CHANGELOG.md#version-110)

deadprogram, 2015-08-17:
I mentioned that about `id` so you'd know I was aware of it. I'll let you know when I have a log for you.

sandeepmistry, 2015-08-30:
@deadprogram any updates?

sandeepmistry, 2015-08-30:
Also, what kernel version are you using, I think I was seeing similar issues with 4.1.

deadprogram, 2015-08-30:
Hi, @sandeepmistry 

I was able to do a complete uninstall and re-install, and my problem "disappeared". One difference was that my previous install was done using `sudo`, and my current one is running `setcap`. I was also able to do a clean install successfully on a couple other Linux machines, both of them running as `root`.

None of the machines I was testing on were running a kernel based on 4.1, all were 3.13+

Anyhow, if I can come up with some more substantive report, I will add to this issue, but sounds like this one can be closed. Thanks!

sandeepmistry, 2015-08-31:
@deadprogram thanks for the follow up. I'll close for now.

closed 2015-08-31 by sandeepmistry