noble/noble #232: Possible bug with multiple connections to the same peripheral

maxogden, 2015-08-14:
given this code running on two separate macbooks:

```js
var noble = require('noble')

var serviceUuid = '13333333333333333333333333333337'
var charUuid = '2a19'
var descUuid = '2901'

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    console.log('scanning...')
    noble.startScanning([serviceUuid], false)
  }
  else {
    noble.stopScanning()
  }
})

noble.on('discover', function(peripheral) {
  noble.stopScanning()
  console.log('found peripheral:', peripheral.advertisement)
  peripheral.connect(function(err) {
    if (err) throw err
    console.log('connected to peripheral', peripheral.advertisement)
    peripheral.discoverServices([serviceUuid], function(err, services) {
      if (err) throw err
      console.log('discovered services', services)
      
      services.forEach(function(service) {
        console.log('found service:', service.uuid)
        service.discoverCharacteristics([], function(err, characteristics) {
          if (err) throw err
          characteristics.forEach(function(characteristic) {
            
            function read () {
              characteristic.read(function (err, buf) {
                if (err) throw err
                console.log('characteristic read', [buf.toString('hex')])
                setTimeout(read, 1000)
              })
            }
            
            read()
            console.log('found characteristic:', characteristic.uuid)
          })
        })
      })
    })
  })
})
```

and this code running on a third macbook:

```js
var bleno = require('bleno')
var util = require('util')
var crypto = require('crypto')

var serviceUuid = '13333333333333333333333333333337'
var charUuid = '2a19'
var descUuid = '2901'

var Service = function () {
  bleno.PrimaryService.call(this, {
    uuid: serviceUuid,
    characteristics: [
      new StreamDataCharacteristic()
    ]
  })
}

util.inherits(Service, bleno.PrimaryService)

var StreamDataCharacteristic = function () {
  bleno.Characteristic.call(this, {
    uuid: charUuid,
    properties: ['read'],
    descriptors: [
      new bleno.Descriptor({
        uuid: descUuid,
        value: 'receive data'
      })
    ]
  })
}

util.inherits(StreamDataCharacteristic, bleno.Characteristic)

StreamDataCharacteristic.prototype.onReadRequest = function(offset, cb) {
  cb(this.RESULT_SUCCESS, crypto.randomBytes(10))
}

var service = new Service()

bleno.on('stateChange', function(state) {
  console.log('on -> stateChange: ' + state)

  if (state === 'poweredOn') {
    bleno.startAdvertising('hello', [serviceUuid])
  } else {
    bleno.stopAdvertising()
  }
})

bleno.on('advertisingStart', function(error) {
  console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'))

  if (!error) {
    bleno.setServices([service])
  } else console.error(error)
})
```

as soon as the 2nd macbook runs the first code snippet it causes the 1st person to stop receiving data, and the 2nd person never receives any data

my use case is trying to create duplex streams over BLE between a swarm of peers, e.g. given 3 peers, all 3 would implement peripherals with readable + writable services using bleno, and all peers would connect to all other peers using noble. there is an experimental library implementing this at https://github.com/mafintosh/ble-swarm, but the test case in this issue is blocking it from working with more then 1 central and 1 peripheral at a time

sandeepmistry, 2015-08-17:
@maxogden I think this is an issue with bleno and not noble.

I need to verify how it handles multiple centrals connected. I only have access to one Mac so need to verify with Linux or smartphone centrals ...

sandeepmistry, 2015-08-18:
It's definitely an issue with CoreBluetooth/blued on OS X. When multiple centrals are connected, read responses for the 2nd one never go through. I've verified this using a native CoreBluetooth app:  https://github.com/sandeepmistry/osx-ble-peripheral with a iOS and Android central connected at the same time.

sandeepmistry, 2015-08-20:
Moved to https://github.com/sandeepmistry/bleno/issues/121, closing this issue.

closed 2015-08-20 by sandeepmistry