noble/noble #233: State unkown when trying to scan and not powering on

pierreduf, 2015-08-14:
Hi Sandeep,

Thanks for the impressive work concerning BT that you made !

I've got a pbm already encountered (I've read some similar issues) : my adapter never turns on !

I'm on a Windows box (Dell E540 with an intel BT 4 BLE compatible). I can see my BLE devices (rolling spiders !) with the legacy Intel drivers but of course with these drivers noble won't work (it was just to check the good working of my device).

When I launch a noble script with WinUSB drivers installed nothing happened. If I check the state of device I see that it's always "poweredOff" (so OK it's normal nothing happens). I tried to get information with your HCI socket scripts and I see that the device is never up (always set to "false") but the device is well detected (good VID and PID and other information).

Everything installed and built perfectly (I've set environment variable for VID/PID of my device and it's working), I've got no error just.....nothing. I've seen previous answer and tooke care to not try to scan before having checked the state (but useless because the state always stays off).

So my questions are:

1. Is it a normal behaviour because my device is not compatible ? So would you know how to make it "compatible" or at least the exact list of devices that should be compatible (I'll buy an external USB adapter). I see on the README that only Broadcom device is stated as compatible. I find this strange.

2. Do you know a way to power up a BT device in Windows (equivalent do the linux hcitool) ? I guess my device is USB connected but the BT stuff is not powered up for some reason.

3. Did I do something wrong and is it a behaviour you've already saw (wrong WinUSB driver or somethg like that) ? I used the latest WinUSB (from Zadig) drivers.

In the meantime I'm gonna try to fiddle with USB tools to see what information I can get from my device and if it's really up or not.

sandeepmistry, 2015-08-14:
@pierreduf This issue really belongs here, do you mind moving it please? https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues

> Is it a normal behaviour because my device is not compatible ? So would you know how to make it "compatible" or at least the exact list of devices that should be compatible (I'll buy an external USB [...]

If you test it out manually and it works, happy to add it to the "compatible" list. I only have access to the Broadcom and CSR adapters.

You can change the USB VID and PID here to test out: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L37

Keep in mind if you enable WinUSB for the built-in adapter you won't be able to use it within Windows, for example with a Bluetooth mouse or keyboard.

> Do you know a way to power up a BT device in Windows (equivalent do the linux hcitool) ? I guess my device is USB connected but the BT stuff is not powered up for some reason.

The device is reset here: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L55

If waits for a reset command complete before marking state as powered on. See: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L104

If you run with debug on, we can see how your adapter is responding:

```sh
set DEBUG=*
node <file>.js
```

Maybe your adapter doesn't support the reset command? It's not entirely needed, but I put it in just so the adapter is in a known state at the start.

> Did I do something wrong and is it a behaviour you've already saw (wrong WinUSB driver or somethg like that) ? I used the latest WinUSB (from Zadig) drivers.

Probably not. However, Windows support is very fresh. Only myself and a friend have tested it so far to my knowledge. We were both using a CSR or Broadcom adapter ...

carlst, 2015-08-14:
I am using a CSR8510 A10 on Windows 7 [Dell Inspiron 580] with WinUSB with general success.

At first, I tried using the USB-BLE adapter through a passive hub, but the adapter was not found:
https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L40

Plugging it directly into a PC USB port works.

pierreduf, 2015-08-16:
Hi Sandeep,

Thx for your answer. No pbm to move this issue. After your explanation it seems clear that the pbm occurs up in the chain (so in the HCI socket).

I close this one and re-open another : I don't konw how to move issue from one project to another :-/

closed 2015-08-16 by pierreduf

sandeepmistry, 2015-08-17:
@pierreduf thanks!

Sorry, when I said move, I was thinking close and open a new one :)