noble/noble #234: Error when two peripherals enter connect function.

foldedtoad, 2015-08-21:
I am seeing a problem when attempt connect to two devices which picked up by a scan 
at the same time.  Most if the time I see the failure below, but occasionally it will all work, but not very often.

There appears to be a race condition, at bind and connect time as seen in the trace.

```
pi@raspberrypi ~/src/example1 $ sudo DEBUG=* node example.js 
  hci-ble hciBle = /home/pi/src/node_modules/noble/lib/linux/../../build/Release/hci-ble +0ms
  noble scanStart +0ms
  hci-ble buffer = "adapterState poweredOn\n" +620ms
  hci-ble line = adapterState poweredOn +3ms
  hci-ble adapterState = poweredOn +11ms
  noble stateChange poweredOn +141ms
  hci-ble buffer = "event D4:84:95:91:4A:43,random,130950696c73794465766963655f39313441343303190000020106,-59\n" +144ms
  hci-ble line = event D4:84:95:91:4A:43,random,130950696c73794465766963655f39313441343303190000020106,-59 +5ms
  hci-ble address = D4:84:95:91:4A:43 +21ms
  hci-ble addressType = random +2ms
  hci-ble eir = 130950696c73794465766963655f39313441343303190000020106 +2ms
  hci-ble rssi = -59 +5ms
  hci-ble advertisement = {"localName":"PilsyDevice_914A43","serviceData":[],"serviceUuids":[]} +73ms
  hci-ble buffer = "event D4:84:95:91:4A:43,random,05030f180a181107d6ce6fe33cbb4aa4d145ea0ed3fa0000,-59\nevent C8:F3:A6:60:F3:C5,random,130950696c73794465766963655f36304633433503190000020106,-76\nevent C8:F3:A6:60:F3:C5,random,05030f180a181107d6ce6fe33cbb4aa4d145ea0ed3fa0000,-75\n" +19ms
  hci-ble line = event D4:84:95:91:4A:43,random,05030f180a181107d6ce6fe33cbb4aa4d145ea0ed3fa0000,-59 +4ms
  hci-ble address = D4:84:95:91:4A:43 +6ms
  hci-ble addressType = random +2ms
  hci-ble eir = 05030f180a181107d6ce6fe33cbb4aa4d145ea0ed3fa0000 +2ms
  hci-ble rssi = -59 +2ms
  hci-ble advertisement = {"localName":"PilsyDevice_914A43","serviceData":[],"serviceUuids":["180f","180a","0000fad30eea45d1a44abb3ce36fced6"]} +62ms
(scan)found:PilsyDevice_914A43 - UUID: d48495914a43
  l2cap-ble D4:84:95:91:4A:43: l2capBle = /home/pi/src/node_modules/noble/lib/linux/../../build/Release/l2cap-ble +0ms
  hci-ble line = event C8:F3:A6:60:F3:C5,random,130950696c73794465766963655f36304633433503190000020106,-76 +191ms
  hci-ble address = C8:F3:A6:60:F3:C5 +3ms
  hci-ble addressType = random +2ms
  hci-ble eir = 130950696c73794465766963655f36304633433503190000020106 +2ms
  hci-ble rssi = -76 +2ms
  hci-ble advertisement = {"localName":"PilsyDevice_60F3C5","serviceData":[],"serviceUuids":[]} +7ms
  hci-ble line = event C8:F3:A6:60:F3:C5,random,05030f180a181107d6ce6fe33cbb4aa4d145ea0ed3fa0000,-75 +2ms
  hci-ble address = C8:F3:A6:60:F3:C5 +2ms
  hci-ble addressType = random +2ms
  hci-ble eir = 05030f180a181107d6ce6fe33cbb4aa4d145ea0ed3fa0000 +2ms
  hci-ble rssi = -75 +2ms
  hci-ble advertisement = {"localName":"PilsyDevice_60F3C5","serviceData":[],"serviceUuids":["180f","180a","0000fad30eea45d1a44abb3ce36fced6"]} +15ms
(scan)found:PilsyDevice_60F3C5 - UUID: c8f3a660f3c5
  l2cap-ble C8:F3:A6:60:F3:C5: l2capBle = /home/pi/src/node_modules/noble/lib/linux/../../build/Release/l2cap-ble +279ms
  l2cap-ble D4:84:95:91:4A:43: buffer = "bind success\n" +114ms
  l2cap-ble D4:84:95:91:4A:43: line = bind success +15ms
  l2cap-ble C8:F3:A6:60:F3:C5: buffer = "bind success\nconnect Transport endpoint is not connected\ndisconnect\n" +27ms
  l2cap-ble C8:F3:A6:60:F3:C5: line = bind success +24ms
  l2cap-ble C8:F3:A6:60:F3:C5: line = connect Transport endpoint is not connected +3ms
/home/pi/src/example1/example.js:23
  if (err) throw err;
                 ^
Error: Transport endpoint is not connected
    at L2capBle.onStdoutData (/home/pi/src/node_modules/noble/lib/linux/l2cap-ble.js:96:55)
    at Socket.emit (events.js:107:17)
    at readableAddChunk (_stream_readable.js:163:16)
    at Socket.Readable.push (_stream_readable.js:126:10)
    at Pipe.onread (net.js:529:20)
```

sandeepmistry, 2015-08-22:
@foldedtoad can you please try again with the 1.1.0 release. L2CAP kernel sockets are no longer used, a HCI socket to the kernel is used instead.

foldedtoad, 2015-08-22:
Hi Sandeep,

I have search for instructions on how to install latest NoBLE on Raspberry Pi, but have not found any.
Could you point me to suitable instructions?

Thanks.

foldedtoad, 2015-08-24:
I have update to release 1.1.0, but I am now seeing the following error.
I have checked that the BLE adapter is up and available.
For example, I can do a "hcitool lesson" without problems.

The error message indicates that BLE adapter is not poweredOn.
Any idea what the problem is?

```
pi@raspberrypi ~/src/example1 $ sudo DEBUG=* node example.js 
  hci setting filter to: 1400000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +99ms
  hci set le event mask - writing: 010120081f00000000000000 +74ms
  hci read local version - writing: 01011000 +7ms
  hci read bd addr - writing: 01091000 +16ms
noble: Could not start scanning, state is unknown (not poweredOn)
  hci onSocketData: 040e0401010c00 +282ms
  hci   event type = 4 +17ms
  hci   sub event type = 14 +3ms
  hci     cmd = 3073 +8ms
  hci     status = 0 +2ms
  hci     result =  +2ms
  hci onSocketData: 040e0401012000 +20ms
  hci   event type = 4 +8ms
  hci   sub event type = 14 +2ms
  hci     cmd = 8193 +5ms
  hci     status = 0 +2ms
  hci     result =  +2ms
  hci onSocketData: 040e0c01011000060010060f000e22 +2ms
  hci   event type = 4 +2ms
  hci   sub event type = 14 +2ms
  hci     cmd = 4097 +2ms
  hci     status = 0 +3ms
  hci     result = 060010060f000e22 +2ms
  hci set scan enabled - writing: 010c20020001 +16ms
  hci set scan parameters - writing: 010b200701100010000000 +17ms
  hci onSocketData: 040e0a0109100034b66870f35c +11ms
  hci   event type = 4 +2ms
  hci   sub event type = 14 +1ms
  hci     cmd = 4105 +2ms
  hci     status = 0 +21ms
  hci     result = 34b66870f35c +2ms
  hci address = 5c:f3:70:68:b6:34 +20ms
  hci onSocketData: 040e04010c200c +11ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +2ms
  hci     cmd = 8204 +1ms
  hci     status = 12 +5ms
  hci     result =  +1ms
  hci onSocketData: 040e04010b2000 +21ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +2ms
  hci     cmd = 8203 +1ms
  hci     status = 0 +4ms
  hci     result =  +1ms
  noble stateChange poweredOn +3ms
```

foldedtoad, 2015-08-24:
My mistake.   All appears to be functioning correctly now.

foldedtoad, 2015-08-24:
Moving to Release 1.1.0 plus user app modification allows multiple device support.

closed 2015-08-24 by foldedtoad