noble/noble #235: Tips for getting to work in electron?

tdicola, 2015-08-21:
Apologies as this isn't really an issue with noble, but I wondered if folks using the library might have more input.  Has anyone gotten noble to successfully build and work with electron? (http://electron.atom.io/)  I've been fighting with node-pre-gyp issues and am not having a lot of luck getting it to work on Windows and Linux (Mac OSX apparently does work though).

For some context electron uses a different node version and has to have some magic done for native modules: https://github.com/atom/electron/blob/master/docs/tutorial/using-native-node-modules.md

Unfortunately when I try to run electron-rebuild for Windows and Linux I keep getting node-pre-gyp errors like `Error: Unsupported target version 0.30.4`.  Digging in further a bit it looks like the usb module that bluetooth-hci-socket depends on requires node-pre-gyp to build, but node-pre-gyp seems to have issues with electron.  From what I can tell it all comes back to this unresolved issue on node-pre-gyp: https://github.com/mapbox/node-pre-gyp/issues/166

So ultimately I'm curious has anyone had luck figuring out a workaround or getting noble to work in electron?  Would love to do some cross-platform BLE apps with electron but getting it built to work with electron seems to be tricky.

sandeepmistry, 2015-08-22:
@tdicola no experience with electron myself, however this comment might by @latentflip might be useful: https://github.com/sandeepmistry/bleno/issues/114#issuecomment-129410837

node-usb is an optional dependency of node-hci-socket, it is only needed on Windows, not Linux.

latentflip, 2015-08-22:
I haven't actually tried building my electron/noble thing on windows/linux yet, so not sure how much that comment will help you. All I've tried/got working thus far is osx.

tdicola, 2015-08-24:
Ah gotcha, yeah I was trying Windows and running into the most trouble with electron.  I'll see if I can get node-usb to build for electron on Windows and share anything that works.

tdicola, 2015-08-26:
Cool, after a bit of fussing with the node-usb module I was able to get it to build for electron and make a simple electron app that uses noble and works on Windows.  For anyone who stumbles on this issue, here's a quick summary of what worked for me:

- First get a Windows toolchain setup for building native node extensions.  Node-gyp has a great summary of what you need (VS 2013 express, python, windows SDK): https://github.com/nodejs/node-gyp#installation  You'll also want to intall node-gyp using npm (`npm install -g node-gyp`).

- Now create a new electron project and use `npm install noble` to install noble into the project.  Everything should build, but you won't be able to use it in electron yet because node-usb is built for the wrong V8 version.  You might be tempted to [follow the steps from electron](https://github.com/atom/electron/blob/master/docs/tutorial/using-native-node-modules.md) to use electron-rebuild, however that will fail because node-usb requires node-pre-gyp to build and [node pre gyp doesn't support electron right now](https://github.com/mapbox/node-pre-gyp/issues/166).

- To fix the node-usb build issue you'll need to modify node-usb not to use node pre gyp and manually run node gyp.  Like [this issue on node-usb mentions](https://discuss.atom.io/t/use-node-usb-inside-electron/16748), dig in to node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/ and edit its binding.gyp file.  Make the variables value at the top to look like:
````
'variables': {
    'use_udev%': 1,
    'use_system_libusb%': 'false',
    'module_name': 'usb_bindings',
    'module_path': './src/binding'
  },
````
The change is to add the module_name and module_path values which node pre gyp apparently sets when it runs.  

- Now you should be able to run node gyp yourself on the node-usb library and point it at a specific version of node to make electron happy.  First figure out what version of electron your'e using (run `electron -v` for example) and then run node gyp on the node-usb library [like the electron native module docs mention](https://github.com/atom/electron/blob/master/docs/tutorial/using-native-node-modules.md#the-node-gyp-way) (see the node gyp section).  For example if you're using electron version 0.30.4 run node gyp like: `node-gyp rebuild --target=0.30.4 --arch=x64 --dist-url=https://atom.io/download/atom-shell`.  This should build without any issues, assuming you have a node gyp toolchain that's working.

- Finally jump up a level to the bluetooth-hci-socket module and run the same node gyp rebuild command to make it compatible with electron.  There shouldn't be any need to change the binding.gyp, etc. file here.

- Run your electron app and have it import noble--success!

Also one other little gotcha I just ran into today.  The latest version of electron, 0.31.0, apparently has a later version of io.js which is now incompatible with libpq.  If you get compile errors when building node-usb that [look like these errors](https://github.com/brianc/node-pg-native/issues/23), then you're hitting the problem.  The easiest fix for now is to fall back to electron version 0.30.4.  In general it seems like you want to pin your electron app to use a very specific version of electron and stick to it as the native modules need to be built against exactly that version.

Phew hopefully that info helps someone else!  Also be warned electron and these libraries are rapidly changing so these instructions will probably fall out of date or new issues might pop up.

Thanks again for the awesome library!

closed 2015-08-26 by tdicola

sandeepmistry, 2015-08-29:
@tdicola great, thanks for sharing!

ozomer, 2016-02-25:
I am struggling with the same issue. Why does it have to be so complicated?
I chose Electron over NW.js because of the separation between main node process and the renderer process. I thought that if I can load (require) noble from node, the main process should be able to load it too... but Electron doesn't run the local node installation - it runs "electron.exe".

Did anyone try using noble on other platforms like NW.js or others?

ozomer, 2016-02-25:
Arrgh. Nothing works. I will just wrap all my noble operations with a [Messenger](https://github.com/weixiyen/messenger.js) server...

ManuelOverdijk, 2016-02-25:
@ozomer I've used NW.js with noble successfully in the past, so you could check that out

ozomer, 2016-02-25:
On windows?

Janneman84, 2016-03-14:
I am trying to get Noble to work in Electron in OSX but when I try require('noble') or remote.require('noble') it just won't find it.

I installed it with npm and it is present in the package.json under dependencies (and node_modules folder). I tried to do the required rebuilding several ways, I'm not sure if this is done successfully (anyway to check?)

I am using gulp-electron by the way. Any advice?

rpastel, 2016-04-03:
This comment contributes tdicola instructions on how to build noble for windows.

My Development Environment:

- Windows 7
- Node v4.1.1
- Node-GYP v3.3.1
- Electron v0.35.4

1) My project uses the Electron v0.35.4 in order to match the Node version that can run Noble. Noble does not run with Node v5 and the current version of Electron runs on Node v5. You can find older version of Electron at

https://github.com/atom/electron/releases

2) I had to make my development environment clean. I uninstall Visual Studio C++ version back through VC++ 2010 and then reinstalled them.  The links below were helpful making my development environment clean.

https://www.robertkehoe.com/2015/03/fix-node-gyp-rebuild-error-on-windows/#comment-9757
https://github.com/nodejs/node-gyp/wiki/Visual-Studio-2010-Setup

3) The current version of the usb module does not appear to compile with Visual Studio 2010 Express. I had to install Visual Studio 2013 Express.

https://www.visualstudio.com/en-us/downloads/download-visual-studio-vs.aspx

4) I followed  the procedure for building by hand that tdicola outlines above.

I did not run into electron-libpq incompatibility issues that tdicola alludes to. 

Thanks, tdicola for the build instructions. Thanks, sadeepmistry for a great api!

Xan-Kun, 2017-06-03:
@ManuelOverdijk I am trying to get it to work for while now with both nw and electron without success. Could you give any pointers on how you did it?

ozomer, 2017-06-04:
@Xan-Kun It was long time ago. I solved it with json-rpc2 - two processes communicating: one for electron and one for noble.

Xan-Kun, 2017-06-04:
@ozomer  I feel you :-) I think I will go with a REST API for now and see what the next iterations of nw, electron, noble and node-usb will bring.