noble/noble #236: Latency issues

xa4, 2015-09-03:
I noticed that the simplest noble app that logs a value sent by a peripheral (BLE anemometer) takes ~1.4 seconds. When using the mobile application that comes with the peripheral, the delay is only ~0.2 seconds, so the delay doesn't come from the peripheral. Is there a way to decrease this latency ?
I am using Bluez 5.23 debian on an intel edison.

Thanks!

sandeepmistry, 2015-09-05:
@xa4 can you please provide an ```hcidump``` capture: ```sudo hcidump -t -x```

I want to see what the connection interval is ...

Is the mobile app running on iOS or Android?

sandeepmistry, 2015-09-20:
Closing due to lack of activity. Please re-op when you have the requested info ...

closed 2015-09-20 by sandeepmistry