noble/noble #239: install seems fail

wangsen1312, 2015-09-15:
when I install the essential dependencies and 
npm install noble 
It shows:

npm http GET https://registry.npmjs.org/noble
npm http 304 https://registry.npmjs.org/noble
npm http GET https://registry.npmjs.org/bluetooth-hci-socket
npm http GET https://registry.npmjs.org/bplist-parser/0.0.6
npm http GET https://registry.npmjs.org/debug
npm http GET https://registry.npmjs.org/xpc-connection
npm http 304 https://registry.npmjs.org/debug
npm http 304 https://registry.npmjs.org/bluetooth-hci-socket
npm http 304 https://registry.npmjs.org/xpc-connection
npm http 304 https://registry.npmjs.org/bplist-parser/0.0.6
npm WARN optional dep failed, continuing xpc-connection@0.1.4
npm http GET https://registry.npmjs.org/ms/0.7.1
npm http GET https://registry.npmjs.org/nan
npm http GET https://registry.npmjs.org/usb
npm http 304 https://registry.npmjs.org/ms/0.7.1
npm http 304 https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/usb

> usb@1.0.6 install /home/xinxinzuo/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

/usr/bin/env: node: No such file or directory
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing usb@1.0.6

> bluetooth-hci-socket@0.3.3 install /home/xinxinzuo/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

/bin/sh: 1: node: not found
gyp: Call to 'node -e "require('nan')"' returned exit status 127. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:431:16)
gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:797:12)
gyp ERR! System Linux 4.1.5-040105-generic
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/xinxinzuo/node_modules/noble/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.25
gyp ERR! node-gyp -v v0.10.10
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.3.3
noble@1.1.0 node_modules/noble
├── bplist-parser@0.0.6
└── debug@2.2.0 (ms@0.7.1)

Is something wrong?

tdicola, 2015-09-16:
`/bin/sh: 1: node: not found`

That looks like node.js isn't found under the node name.  Try running `node --version` at a terminal to see if node exists under that name.  Likely you'll get a command not found error, and if so you'll want to make sure you have node.js installed.  If you're on Debian or Ubuntu Linux you want the nodejs-legacy package installed (`sudo apt-get install nodejs-legacy`) as it adds node.js under its normal node name.  Alternatively install n (https://github.com/tj/n) and use that to install specific versions of node.  In general you probably want to install your own versions of node with n since Debian & Ubuntu's node packages can be a little old.

wangsen1312, 2015-09-16:
@tdicola  Thank you very much~ it works well now ~

sandeepmistry, 2015-09-20:
@tdicola thanks for helping out. Which version of node does nodejs-legacy install?

@wangsen1312 what did you end up doing to resolve the issue? Is this something that I can add to the read me?

tdicola, 2015-09-22:
On Ubuntu 15.04 nodejs-legacy installs 0.10.x IIRC--it's pretty old. :/  I've found using nodejs-legacy and npm packages to get the basic old version installed, then installing n (https://github.com/tj/n, `sudo npm install -g n`, and yeah the sudo on npm -g is ugly but sort of required on Ubuntu), and using `n stable` or `n 0.12.7`, etc. to get specific versions works great.

wangsen1312, 2015-09-23:
Hi @sandeepmistry   I use the npm version node v0.10.25. I follow the @tdicola 's instruction and it works well. 
Maybe you can add it to the Readme

sandeepmistry, 2015-09-28:
@wangsen1312 thanks for the info, v0.10.25 is pretty old, i'm wondering if there's a better way.

Maybe installing a Node Source package? https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions

wangsen1312, 2015-09-28:
@sandeepmistry  Thank you for your information, but I use some old node part only compatible with v0.10, hope it compatible with newer version sooner.

sandeepmistry, 2015-09-28:
@wangsen1312 @tdicola I've updated the read me with some notes (1b4bff1)

closed 2015-09-28 by sandeepmistry