noble/noble #241: RangeError on Discover

PimButton, 2015-09-24:
I'm getting the RangeError when running the peripheral-explorer example. I'm on windows, using a fresh install of both Noble and Async. 
```
Your environment has been set up for using Node.js 0.12.7 (ia32) and npm.

C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble>node test.js
buffer.js:506
    throw new RangeError('index out of range');
          ^
RangeError: index out of range
    at checkOffset (buffer.js:506:11)
    at Buffer.readUInt8 (buffer.js:544:5)
    at BluetoothHciSocket.onHciEventEndpointData (C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble\node_modules\noble\node_modules\bluetooth-hci
-socket\lib\usb.js:115:36)
    at InEndpoint.emit (events.js:107:17)
    at Transfer.transferDone (C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble\node_modules\noble\node_modules\bluetooth-hci-socket\node_modules
\usb\usb.js:295:9)
```

Any ideas? Would like to provide more debug data, how do I enable this? The error looks like #25, but that one is closed and I cannot reopen it.

sandeepmistry, 2015-09-28:
@PimButton thanks for reporting, debug logs would be useful, along with you sharing your code and steps to reproduce.

```sh
set DEBUG=*
node <file>.js
```

PimButton, 2015-09-28:
Ok, I have some more info. The device I am connecting to seems to be the problem. It's a custom designed PCB with a NRF51 as bluetooth device. Noble on the Mac and the Bluetooth Explorer on the Mac have no trouble connecting and writing / reading, so it seems to be a windows only problem. 

## The log
The debug log shows the following: 

```
C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble>node test.js
  hci-usb reset +0ms
  hci-usb write: 01030c00 +15ms
  hci-usb HCI event: 0e04010c200c +16ms
  hci onSocketData: 040e04010c200c +15ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +16ms
  hci           cmd = 8204 +16ms
  hci           status = 12 +15ms
  hci           result =  +16ms
  hci-usb HCI event: 0e0401030c00 +31ms
  hci-usb reset complete +16ms
  hci onSocketData: 040e0401030c00 +15ms
  hci   event type = 4 +16ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3075 +15ms
  hci           status = 0 +16ms
  hci           result =  +0ms
  hci setting filter to: 1400000020c10000000000400000 +811ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +16ms
  hci-usb write: 01010c08fffffbff07f8bf3d +0ms
  hci set le event mask - writing: 010120081f00000000000000 +15ms
  hci-usb write: 010120081f00000000000000 +16ms
  hci read local version - writing: 01011000 +16ms
  hci-usb write: 01011000 +15ms
  hci read bd addr - writing: 01091000 +0ms
  hci-usb write: 01091000 +16ms
  hci-usb HCI event: 0e0401010c00 +15ms
  hci onSocketData: 040e0401010c00 +16ms
  hci   event type = 4 +16ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3073 +15ms
  hci           status = 0 +16ms
  hci           result =  +15ms
  hci-usb HCI event: 0e0401012000 +0ms
  hci onSocketData: 040e0401012000 +16ms
  hci   event type = 4 +16ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8193 +15ms
  hci           status = 0 +16ms
  hci           result =  +15ms
  hci-usb HCI event: 0e0c0101100006bb22060a00bb22 +0ms
  hci onSocketData: 040e0c0101100006bb22060a00bb22 +16ms
  hci   event type = 4 +16ms
  hci   sub event type = 14 +15ms
  hci           cmd = 4097 +0ms
  hci           status = 0 +16ms
  hci           result = 06bb22060a00bb22 +15ms
  hci set scan enabled - writing: 010c20020001 +16ms
  hci-usb write: 010c20020001 +16ms
  hci set scan parameters - writing: 010b200701100010000000 +0ms
  hci-usb write: 010b200701100010000000 +15ms
  hci-usb HCI event: 0e0a010910001371da7d1a00 +16ms
  hci onSocketData: 040e0a010910001371da7d1a00 +15ms
  hci   event type = 4 +16ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4105 +16ms
  hci           status = 0 +15ms
  hci           result = 1371da7d1a00 +16ms
  hci address = 00:1a:7d:da:71:13 +0ms
  hci-usb HCI event: 0e04010c200c +15ms
  hci onSocketData: 040e04010c200c +16ms
  hci   event type = 4 +16ms
  hci   sub event type = 14 +0ms
[... 57 lines cut ...]

services and characteristics:
  hci create le conn - writing: 010d2019600030000001ad5a4b49b1dc0006000c000000c80004000600 +94ms
  hci-usb write: 010d2019600030000001ad5a4b49b1dc0006000c000000c80004000600 +0ms
  hci-usb HCI event:  +15ms
  hci set scan enabled - writing: 010c20020001 +16ms
  hci-usb write: 010c20020001 +16ms
buffer.js:506
    throw new RangeError('index out of range');
          ^
RangeError: index out of range
    at checkOffset (buffer.js:506:11)
    at Buffer.readUInt8 (buffer.js:544:5)
    at BluetoothHciSocket.onHciEventEndpointData (C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble\node_modules\noble\node_modules\bluetooth-hci
-socket\lib\usb.js:115:36)
    at InEndpoint.emit (events.js:107:17)
    at Transfer.transferDone (C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble\node_modules\noble\node_modules\bluetooth-hci-socket\node_modules
\usb\usb.js:295:9)
```

## The code
The code I'm using is a slightly adjusted version of the peripheral-explorer.js that I found in example. I have uploaded my version here: http://pastebin.com/3HC5pFGZ

## Steps to reproduce:
I'm using Windows 7 (64bit), Node 0.12.7 and a USB bluetooth dongle with the CSR8510 A10 chipset. Just run ```node testNoble.js``` in order to reproduce.

sandeepmistry, 2015-09-29:
@PimButton thanks.

Woah, that's really strange, an empty HCI event is received here: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L101

```
hci-usb HCI event:  +15ms
```

Could you please try changing this function, https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L100 to:

```javascript
BluetoothHciSocket.prototype.onHciEventEndpointData = function(data) {
  debug('HCI event: ' + data.toString('hex'));

  if (data.length === 0) {
    return;
  }  

  // ...
};
```

and then run again to see if things go further.

PimButton, 2015-10-01:
@sandeepmistry  Yes, that fixed it, thanks! Now running into something else, will open a new issue.

closed 2015-10-01 by PimButton

sandeepmistry, 2015-10-01:
@PimButton great! I've pushed the suggested change: https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/f2b05b724fa80adca22ed4d4c4817742016b1358 and also published v0.3.4 to npm.