noble/noble #242: Linux Kernel 4.1.x disconnects

sandeepmistry, 2015-09-28:
There appears to be issues maintaining connection with kernel 4.1+. The connection disconnects after a while, ``hcidump`` shows a disconnect command after a while:

```
2015-09-22 17:03:37.964308 < HCI Command: Disconnect (0x01|0x0006) plen 3
handle 71 reason 0x13
```

I think this is due to: https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/net/bluetooth?id=25ba265390c09b0a2b2f3fd9ba82e37248b7a371 - to confirm check ``dmesg`` for ``SMP security requested but not available``

This was originally reported by: https://github.com/sandeepmistry/node-sensortag/issues/55

Raspberry Pi user's can temporarily work around by downgrading their kernels by running:

``sh
sudo rpi-update 46d179597370c5145c7452796acbee0f1ff93392
``

oroce, 2015-10-01:
seems like a kernel issue for me, because arch community is seeing something similar in 4.1.x: https://bugzilla.kernel.org/show_bug.cgi?id=104011

sandeepmistry, 2015-10-02:
@oroce thanks!

I've updated my Raspberry Pi to kernel ``4.1.8+ `` via rpi-update. Here are the results of my investigation so far:

### hcitool

```sh
pi@raspberrypi ~ $ sudo hcitool lescan
LE Scan ...
90:03:B7:E7:8B:98 (unknown)
90:03:B7:E7:8B:98 Flower power 8B98
84:DD:20:EA:F4:52 (unknown)
84:DD:20:EA:F4:52 df1
DC:0B:86:95:E8:A5 Climate_95e8a5
DC:0B:86:95:E8:A5 (unknown)
^Cpi@raspberrypi ~ $ sudo hcitool lecc 90:03:B7:E7:8B:98
Connection handle 64
```

hcidump:
```
2015-10-02 00:08:34.366567 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 90:03:B7:E7:8B:98 type 0
2015-10-02 00:08:34.368802 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2015-10-02 00:08:35.933937 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 90:03:B7:E7:8B:98 (Public)
2015-10-02 00:08:35.934587 < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
  40 00 
2015-10-02 00:08:35.937104 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2015-10-02 00:08:35.967108 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 64
      Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2015-10-02 00:08:37.966972 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2015-10-02 00:08:37.968288 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2015-10-02 00:08:37.973175 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

Disconnect is triggered by the kernel:
```
2015-10-02 00:08:37.966972 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
```

and I also see the following in ``dmesg`` output:
```
[ 3566.316868] Bluetooth: SMP security requested but not available
```

### gatttool
```sh
pi@raspberrypi ~ $ gatttool -I -b 90:03:B7:E7:8B:98
[   ][90:03:B7:E7:8B:98][LE]> connect 
Connecting... connect error: Host is down (112)
[   ][90:03:B7:E7:8B:98][LE]>
```

hcidump:
```
2015-10-02 00:15:06.902075 < HCI Command: Create Connection (0x01|0x0005) plen 13
    bdaddr 90:03:B7:E7:8B:98 ptype 0xcc18 rswitch 0x01 clkoffset 0x0000
    Packet type: DM1 DM3 DM5 DH1 DH3 DH5 
2015-10-02 00:15:06.903892 > HCI Event: Command Status (0x0f) plen 4
    Create Connection (0x01|0x0005) status 0x00 ncmd 1
2015-10-02 00:15:12.027436 > HCI Event: Connect Complete (0x03) plen 11
    status 0x04 handle 12 bdaddr 90:03:B7:E7:8B:98 type ACL encrypt 0x00
    Error: Page Timeout
```

### comparison and summary

Between the two: "Create Connection" is used by gatttool instead of "LE Create Connection" which is really weird!

Need to see if the kernel is sending the "LE Read Remote Used Features" with hcitool, although it might be unrelated.

cc/ @don

sandeepmistry, 2015-10-02:
Hmm, this commit might be the culprit: https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/net/bluetooth?id=0fe29fd1cd77ffbdb8e36ec1715868d9d8011c9b

oroce, 2015-10-02:
pretty similar what I was seeing.

hcitool:
```
pi@raspberrypi ~ $ sudo hcitool lescan
LE Scan ...
1C:BA:8C:20:E2:C8 (unknown)
1C:BA:8C:20:E2:C8 SensorTag
A0:14:3D:08:DC:92 (unknown)
A0:14:3D:08:DC:92 Flower power DC92
B4:99:4C:64:BE:83 (unknown)
B4:99:4C:64:BE:83 SensorTag
pi@raspberrypi ~ $ sudo hcitool lecc B4:99:4C:64:BE:83
Connection handle 64
```

hcidump:
```
pi@raspberrypi ~ $ hcidump -t -X
HCI sniffer - Bluetooth packet analyzer ver 5.34
device: hci0 snap_len: 1500 filter: 0xffffffff
2015-10-01 11:44:11.167291 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-10-01 11:44:12.589422 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2015-10-01 11:44:12.593401 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-10-01 11:47:33.056875 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-10-01 11:47:48.503136 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2015-10-01 11:47:54.076624 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2015-10-01 11:47:56.111768 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2015-10-01 11:47:56.121777 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

gatttol:
```
pi@raspberrypi ~ $ sudo gatttool -b 1C:BA:8C:20:E2:C8 --interactive -t random
[1C:BA:8C:20:E2:C8][LE]> connect
Attempting to connect to 1C:BA:8C:20:E2:C8
Error: connect: Connection refused (111)
[1C:BA:8C:20:E2:C8][LE]>
```

bluez:
```
pi@raspberrypi ~ $ bluetoothctl -v
5.34
```

note: this is what I shared with you yesterday @sandeepmistry, I just wanted to have it here too.

Niksko, 2015-10-27:
@oroce I'm guessing you use Arch based on your comment. Any workaround for this, other than just downgrading the kernel? Is this fixed in the 4.2 kernels? Really wishing Arch had a better way to support multiple kernels...

sandeepmistry, 2015-10-28:
I've merged some workarounds for this into master, the key changes were https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/04e13b3f451ddf106de90af35a7399a337d64893 and https://github.com/sandeepmistry/noble/commit/16a24561e975c8612f438b0e9f3a4d3fa1ad40a3

Things look much better on my Raspberry Pi running kernel 4.1.11+ and @oroce's Linux setup.

Note: to try out make sure you pull the latest node-bluetooth-hci-socket version, or install via: ```npm install sandeepmistry/noble``` before it is published to npm.

closed 2015-10-28 by sandeepmistry

liamkennedy, 2015-10-28:
For anyone who finds this post (like I did) and are having problems with a Raspberry Pi running Wheezy.   

I was running some code that worked perfectly on an older Raspbian image that communicates with an RFDuino BLE device.   I tried to port my code to the most recent wheezy image - and received the connection error mentioned here.  

My fix to this is to switch to using the new Jessie image.   That has built-in support for 5.23 of bluez and  fixed my problem

masinde70, 2016-03-11:
4.1.18-v7+ I have problem with this version and I have installed 5.37 bluez i get the same error Error: connect: Connection refused (111)
but it was working with version 4.0.X only after the upgrade. am using debian jessie..any help..thanks

jeberhardt, 2016-04-19:
So which version of Bluez is recommended for use with this Kernel?  I have downgraded, and am now installing Bluez 5.39, hoping that it will work.

sandeepmistry, 2016-04-24:
@jeberhardt I'm not sure, Bluez 5.39 should work.

onosAdmin, 2017-03-16:
Try to do: 
sudo ./btmgmt le on

you need to find where btmgmt is ,
On my system locate btmgmt tell me that is on:
 /usr/bin/btmgmt

source:
https://eclipse.github.io/kura/doc/bluetooth-le-example.html