noble/noble #243: Unable to pair with iOS for ANCS

robotastic, 2015-09-29:
I am using a combination of Bleno + Noble to try and receive ANCS notification from an iOS using a Linux machine. Bleno is used for the initial advertisement of a Peripheral Type that will be visible in iOS without an app needed. 

Once the Bleno advertised device shows up in iOS I click on the device to begin a connection. Bleno responds to the read request from iOS. Meanwhile, Noble is used to discover the ANCS Characteristics. It will then try to subscribe to one that is secure, discover it needs to pair/bond and send the request. However, iOS responds with SMP_PAIRING_FAILED for Unspecified Reason.

iOS does not display a dialog box asking if the User wants to Pair with the device.

Do you have any examples of how to trigger the iOS Pairing Box or where Pairing & Bonding completed successfully? That code would be really helpful I think!

Thanks!

robotastic, 2015-10-05:
I may have figured some of it out. In order for iOS to display the pairing request alert, it needs to find a Secure Characteristic on the Peripheral. Making some of the characteristics secure solved some of this. 

The other weird thing I have found is that even after I kill/stop bluetoothd, there is something automatically handling the SMP Paring Request. If I comment out the Writes in SMP.js, I can successfully pair with an iOS device.

Do you know if the kernel portion of Bluez automatically handles the pairing? I am running on Unbuntu 14.04 and I think it is Bluez 4.101. I am booting into the console with no desktop, so the bluetooth manager isn't running.

sandeepmistry, 2015-10-07:
> Do you know if the kernel portion of Bluez automatically handles the pairing? 

@robotastic 

Yes, the kernel does have code to handle pairing. In raw HCI socket mode (which noble and bleno currently use), it is activated - but in user channel HCI socket it is not. In bleno, I had to use management sockets to workaround this: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/mgmt.js

I'm thinking of changing everything to use user channel HCI sockets, but there are some consequences to making this change.

robotastic, 2015-10-13:
Awesome! That is good to know. I think I have everything working. I still have to work on setting the advertised name, but other than that things are working good. I am able to see the system directly from the Bluetooth settings menu on iOS without needing any sort of app running.

I have my code up here. It is a big mess, but it works.
https://github.com/robotastic/ble-ancs

robotastic, 2015-10-19:
I am actually having a problem with getting Pairing to work with SMP. Things work fine if I depend completely on the kernel and commenting out all the instances where Bleno/Noble write to the ACL to respond regarding SMP, like here: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/smp.js#L119

For this to work, I need to have **bluetoothd** load and then quit it. If **bluetoothd** is never loaded, it doesn't work. I want to try to have it more predictable, so I want to handle it with the SMP code in Bleno. However, it doesn't seem to work. It looks like when the iOS sends a `LE Long Term Key Request`, the OS layer is automatically responding with a `LE Long Term Key Request Negative Reply`. Is the longterm key not being correctly passed to the HCI?

````
> ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0x0005
      type-uuid 0x2a00
< ACL data: handle 69 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Read By Type req (0x08) on handle 0x0001
> ACL data: handle 69 flags 0x02 dlen 11
    SMP: Pairing Request (0x01)
      capability 0x04 oob 0x00 auth req 0x0d
      max key size 0x10 init key dist 0x03 resp key dist 0x03
      Capability: KeyboardDisplay (OOB data not present)
      Authentication: Bonding (MITM Protection)
      Initiator Key Distribution:  LTK IRK 
      Responder Key Distribution:  LTK IRK 
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
< ACL data: handle 69 flags 0x00 dlen 11
    SMP: Pairing Response (0x02)
      capability 0x03 oob 0x00 auth req 0x01
      max key size 0x10 init key dist 0x00 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (No MITM Protection)
      Initiator Key Distribution:    
      Responder Key Distribution:  LTK  
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
> ACL data: handle 69 flags 0x02 dlen 21
    SMP: Pairing Confirm (0x03)
      key e2f20f1452c197adeaa89dfa2599bc80
< ACL data: handle 69 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key a73b2f4f42cd7afd2cad501f50d42a57
> ACL data: handle 69 flags 0x02 dlen 21
    SMP: Pairing Random (0x04)
      random 9967cb1ae09de4a5e6787021200f10e6
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
< ACL data: handle 69 flags 0x00 dlen 48
    SMP: Unknown (0x13)
< ACL data: handle 69 flags 0x00 dlen 21
    SMP: Pairing Random (0x04)
      random adcd82fa0039640d921a00c5b48bdf99
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
> HCI Event: LE Meta Event (0x3e) plen 13
    LE Long Term Key Request
< HCI Command: LE Long Term Key Request Negative Reply (0x08|0x001b) plen 2
> HCI Event: Command Complete (0x0e) plen 6
    LE Long Term Key Request Negative Reply (0x08|0x001b) ncmd 1
> ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0x0005
      type-uuid 0x2a00
< ACL data: handle 69 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Read By Type req (0x08) on handle 0x0001
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 69 reason 0x13
    Reason: Remote User Terminated Connection
````

robotastic, 2015-10-19:
For contrast, here is how the OS responds when it is handling everything:
````> ACL data: handle 68 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0x0008
      type-uuid 0x2a00
< ACL data: handle 68 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Read By Type req (0x08) on handle 0x0001
> ACL data: handle 68 flags 0x02 dlen 11
    SMP: Pairing Request (0x01)
      capability 0x04 oob 0x00 auth req 0x05
      max key size 0x10 init key dist 0x03 resp key dist 0x03
      Capability: KeyboardDisplay (OOB data not present)
      Authentication: Bonding (MITM Protection)
      Initiator Key Distribution:  LTK IRK 
      Responder Key Distribution:  LTK IRK 
< ACL data: handle 68 flags 0x00 dlen 11
    SMP: Pairing Response (0x02)
      capability 0x03 oob 0x00 auth req 0x05
      max key size 0x10 init key dist 0x03 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (MITM Protection)
      Initiator Key Distribution:  LTK IRK 
      Responder Key Distribution:  LTK  
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
> ACL data: handle 68 flags 0x02 dlen 21
    SMP: Pairing Confirm (0x03)
      key dc5886b16add33a0c8f714aaee499ead
< ACL data: handle 68 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key df3c076af455f863bc6bd347086abdc4
> ACL data: handle 68 flags 0x02 dlen 21
    SMP: Pairing Random (0x04)
      random 9ac9c6623e2b03d14640fceedfb7a505
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
< ACL data: handle 68 flags 0x00 dlen 21
    SMP: Pairing Random (0x04)
      random b2395fe2ba11cc490332ad8ae7759db2
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
> HCI Event: LE Meta Event (0x3e) plen 13
    LE Long Term Key Request
< HCI Command: LE Long Term Key Request Reply (0x08|0x001a) plen 18
> HCI Event: Command Complete (0x0e) plen 6
    LE Long Term Key Request Reply (0x08|0x001a) ncmd 1
> HCI Event: Encrypt Change (0x08) plen 4
    status 0x00 handle 68 encrypt 0x01
< ACL data: handle 68 flags 0x00 dlen 21
    SMP: Encryption Information (0x06)
      LTK fa9860b44ad6fdf4f86bfce7f95cf4de
< ACL data: handle 68 flags 0x00 dlen 15
    SMP: Master Identification (0x07)
      EDIV 0xa58d Rand 0x5f9f29a4a3bc47bd
< ACL data: handle 68 flags 0x00 dlen 27
    ATT: Find By Type req (0x06)
      start 0x0001, end 0xffff, uuid 0x2800
      value 0xd0 0x00 0x2d 0x12 0x1e 0x4b 0x0f 0xa4 0x99 0x4e 0xce 0xb5 0x31 0xf4 0x05 0x79
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
> ACL data: handle 68 flags 0x02 dlen 21
    SMP: Encryption Information (0x06)
      LTK a3e65d77220ac7874caf78b8b9e2a32d
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 68 packets 1
> ACL data: handle 68 flags 0x02 dlen 15
    SMP: Master Identification (0x07)
      EDIV 0x572c Rand 0x8ae4b4002c2eaff9
> ACL data: handle 68 flags 0x02 dlen 21
    SMP: Identity Information (0x08)
      IRK 07726c17f0cf8fc3eb01a038e5d647ac
> ACL data: handle 68 flags 0x02 dlen 12
    SMP: Identity Address Information (0x09)
      bdaddr 00:11:80:00:80:14 (Public)
> ACL data: handle 68 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0x0008
      type-uuid 0x2a00
> ACL data: handle 68 flags 0x02 dlen 9
    ATT: Find By Type resp (0x07)
      Found attr 0x0019, group end handle 0x0022
< ACL data: handle 68 flags 0x00 dlen 19
    ATT: Read By Type resp (0x09)
      length: 13
        handle 0x0003, value 0x72 0x61 0x73 0x70 0x62 0x65 0x72 0x72 0x79 0x70 0x69 
````

buildog, 2020-06-02:
Ok, it's look like the encryption process is being triggered when I try to subscribe to the notify characteristics but fail with :  

```
2020-06-02 10:57:45.657381 > ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Read By Type req (0x08) on handle 0x000f
2020-06-02 10:57:45.658692 < ACL data: handle 64 flags 0x00 dlen 11
    SMP: Pairing Request (0x01)
      capability 0x03 oob 0x00 auth req 0x01
      max key size 0x10 init key dist 0x00 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (No MITM Protection)
      Initiator Key Distribution:    
      Responder Key Distribution:  LTK  
2020-06-02 10:57:45.687326 > ACL data: handle 64 flags 0x02 dlen 6
    SMP: Pairing Failed (0x05)
      reason 0x05
      Reason Pairing Not Supported
``` 

I don't know how to solve this issue