noble/noble #244: warning: possible EventEmitter memory leak detected. 11 read listeners added.

lab101, 2015-10-01:
Hey,

I'm still kinda new to Node.Js development so there is a possibility that I'm doing something which is not done in node.

For testing I want to read characteristic with a timer.

```
 device.prototype.readBattery = function (){
      //console.log("+++" + this.id);

        setTimeout(function() {
            this.readBattery();
        }.bind(this), 2000);

   if(this.batteryChar == null) return;
//   this.batteryChar.read();

   this.batteryChar.read(function(error, data) {         
         if(data) console.log(_this.majorMinorId +  ' battery: ' + data.readUInt8(0) + '%');
    });
       
}
```

I'm getting an error on this.
(node) warning: possible EventEmitter memory leak detected. 11 read listeners added. Use emitter.setMaxListeners() to increase limit.
Does it mean that it adds a listener everytime I want to do a manual read?
Could not find any other example in de docs or the examples.

Thx
Kris

sandeepmistry, 2015-10-04:
@lab101 looks ok to me:

```javascript
var events = require('events');

var noble = require('./index');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning(['39e1fa0084a811e2afba0002a5d5c51b']);
  } else {
    noble.stopScanning();
  }
});

console.log('max event listeners = ' + events.EventEmitter.defaultMaxListeners);

noble.on('discover', function(peripheral) {
  console.log('discover')

  peripheral.once('disconnect', function() {
    process.exit(0);
  });

  console.log('connect')
  peripheral.connect(function(error) {
    console.log('connected')
    peripheral.discoverServices(['180f'], function(error, services) {
      var batteryService = services[0];
      console.log('discoveredBattery service');

      batteryService.discoverCharacteristics(['2a19'], function(error, characteristics) {
        var batteryLevelCharacteristic = characteristics[0];

        function readBattery() {
          batteryLevelCharacteristic.read(function(error, data) {
            console.log('batter level = ' + data[0]);
            console.log('read listener count = ' + batteryLevelCharacteristic.listeners('read').length);
            console.log('data listener count = ' + batteryLevelCharacteristic.listeners('data').length);
          });
        }

        setInterval(readBattery, 5000);
      });
    });
  });
});
```

```
pi@raspberrypi ~/git/noble $ sudo node issue-244.js 
max event listeners = 10
discover
connect
connected
discoveredBattery service
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
batter level = 52
read listener count = 0
data listener count = 0
```

PS: it would be great if you could share full code snippet on how to reproduce next time.

lab101, 2015-10-05:
Hey Sandeepmistry.

Thx for the feedback!
I was testing with a prototype from a hardware supplier.
Currently the prototype is broken so can't test it :(
Will test it tomorrow and give feedback.

sandeepmistry, 2015-10-28:
@lab101 any updates?

lab101, 2015-11-02:
Sorry for the huge delay.
I think I found the cause it goes wrong when the device goes offline and reconnects later.
If I use your code (state above)  on OSX and disactive and reactivate bluetooth I got error again.

```
batter level = 100
read listener count = 0
data listener count = 0
connected
discoveredBattery service
batter level = 100
read listener count = 0
data listener count = 0
(node) warning: possible EventEmitter memory leak detected. 11 read listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at Characteristic.addListener (events.js:239:17)
    at Characteristic.once (events.js:265:8)
    at Characteristic.read (/Users/zuppaman/Documents/pmi---cinder---iqos-connected/bluetoothService/node_modules/noble/lib/characteristic.js:39:10)
    at readBattery [as _repeat] (/Users/zuppaman/Documents/pmi---cinder---iqos-connected/bluetoothService/t.js:33:38)
    at wrapper [as _onTimeout] (timers.js:270:19)
    at Timer.listOnTimeout (timers.js:89:15)
(node) warning: possible EventEmitter memory leak detected. 11 read listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at Characteristic.addListener (events.js:239:17)
    at Characteristic.once (events.js:265:8)
    at Characteristic.read (/Users/zuppaman/Documents/pmi---cinder---iqos-connected/bluetoothService/node_modules/noble/lib/characteristic.js:39:10)
    at readBattery [as _repeat] (/Users/zuppaman/Documents/pmi---cinder---iqos-connected/bluetoothService/t.js:33:38)
    at wrapper [as _onTimeout] (timers.js:270:19)
    at Timer.listOnTimeout (timers.js:89:15)
(node) warning: possible Even
```

sandeepmistry, 2015-11-07:
@lab101 could you please provide more detailed steps for how to reproduce?

For example, how many times did you reactivate bluetooth? How many reconnects were performed before the error?

lab101, 2015-11-12:
Hey I just disable and enable bluetooth once. From that point on the event number keeps on adding up.
Just as a side note it's not blocking my project so feel free to close the issue if you can't reproduce it.

omeryagmurlu, 2015-11-21:
I can confirm this issue, to reproduce this make your peripheral a little far from device (make it disconnect and connect [make it lose packets]). Every time noble connects it again, the count increases

jrobeson, 2016-01-05:
I've seen this same issue in https://github.com/adafruit/adafruit-bluefruit-le-desktop/ which bundles noble.

felipecocco, 2016-03-01:
+ 1

jeberhardt, 2016-04-19:
I think to fix this error you should use **peripheral.once('connect', function(){});** instead of **peripheral.connect()**.

I previously had a similar problem when I had **peripheral.on('connect', function(){});**.  When my peripheral connected and disconnect, the listener for 'connect' was being added to the peripheral object, causing this warning.

tybro0103, 2017-10-09:
I'm experiencing this still. To reproduce, just connect and disconnect until you hit the limit (default is 10). Upon the 11th connect the error is thrown. This line is is the culprit in the stack trace:
https://github.com/sandeepmistry/noble/blob/master/lib/peripheral.js#L37

kuzyn, 2018-04-20:
Just to raise the profile for other searching for this issue: there was an issue (https://github.com/noble/noble/issues/193) started years ago about this

jasaw, 2019-08-05:
I'm getting this issue. The problem for me is noble registers servicesDiscover listener internally when I do `peripheral.discoverServices(serviceUUID, cb)` using once, but does not remove the listener if I get disconnected before discover services completes. It's problematic on Linux where bluez stack always gets itself into an "immediate disconnect" state where I get an immediate disconnect event after a connected event (but before I managed to call discoverServices). Adding the below 3 lines got rid of the problem.

```
peripheral.removeAllListeners('servicesDiscover');
peripheral.removeAllListeners('connect');
peripheral.removeAllListeners('disconnect');
```