noble/noble #245: Uncaught Error: LIBUSB_TRANSFER_STALL on many reads

PimButton, 2015-10-01:
I'm using noble to loop trough an array of around 10 BLE devices. The CSR A10 dongle i'm using only supports 5 concurrent connections so I wrote a function that
1. connects to the BLE device
2. Reads a specific characteristic
3. Disconnects the connection and calls the same function with the next peripheral in the array. 

I call this function twice, once for the even devices in the array and once for the odd devices in the array. However, I very quickly receive the following errors:
```Uncaught Error: LIBUSB_TRANSFER_STALL``` and ```Uncaught Error: LIBUSB_TRANSFER_ERROR```. Then when the program tries to connect to the next device the following error is shown: ```Uncaught TypeError: Cannot read property 'call' of undefined``` and the execution stops. 

Adding a delay of around 800ms after the disconnection, before connecting to the next device holds off the error for a little longer, but after around 1 minute the same errors appear. 

Is there anything I can do about this? Or is there a better way to loop trough devices after each other or if possible two at a time to speed it up a little? 

I'm using Electron on Windows 7 to run this as a separate program so share the code is a little complicated, but I will try and make a node only program if that helps to debug.

sandeepmistry, 2015-10-01:
@PimButton simplified code snippet to reproduce would be great (without electron etc).

Also, could be please share the stack trace for the error.

PimButton, 2015-10-02:
@sandeepmistry thanks for your quick reply again!

I created a simplified code snippet, you can find it here: http://pastebin.com/9HSi0yZg

The snippet looks for devices with a specific name. When found it puts the device in the ```imoConnectionList``` array. If 10 devices are found, the ```readAcc()``` function is called twice, once for the even devices and once for the odd devices. This function calls itself after a successful read of the Acc characteristic. If the read is not successful within 2.5 seconds after being called the function also calls itself, and skips the current read. 

The stack trace is as follows:

``` 
hci           data = 1106010007000018080008000118 +0ms
  att f7:75:57:11:4c:92: write: 100900ffff0028 +0ms
  hci write acl data pkt - writing: 0234000b0007000400100900ffff0028 +0ms
  hci-usb write: 0234000b0007000400100900ffff0028 +0ms
  hci-usb HCI event: 13050134000100 +0ms
  hci onSocketData: 0413050134000100 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 19 +0ms
  hci-usb HCI event: 13050134000100 +16ms
  hci onSocketData: 0413050134000100 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 19 +0ms
  hci-usb ACL Data In: 34201a00160004001114090012003b1f465aadc949c69c2b13712315b38b +15ms
  hci onSocketData: 0234201a00160004001114090012003b1f465aadc949c69c2b13712315b38b +0ms
  hci   event type = 2 +0ms
  hci           cid = 4 +0ms
  hci           handle = 52 +0ms
  hci           data = 1114090012003b1f465aadc949c69c2b13712315b38b +0ms
  att f7:75:57:11:4c:92: write: 101300ffff0028 +0ms
  hci write acl data pkt - writing: 0234000b0007000400101300ffff0028 +0ms
  hci-usb write: 0234000b0007000400101300ffff0028 +0ms
  hci-usb HCI event: 3e13010037000001ad5a4b49b1dc0c00 +16ms
  hci-usb HCI event: 0000c80000 +0ms
  hci onSocketData: 043e13010037000001ad5a4b49b1dc0c000000c80000 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 1 +0ms
  hci           LE meta event status = 0 +15ms
  hci           LE meta event data = 37000001ad5a4b49b1dc0c000000c80000 +0ms
  hci                   handle = 55 +0ms
  hci                   role = 0 +0ms
  hci                   address type = random +0ms
  hci                   address = dc:b1:49:4b:5a:ad +0ms
  hci                   interval = 15 +0ms
  hci                   latency = 0 +0ms
  hci                   supervision timeout = 2000 +0ms
  hci                   master clock accuracy = 0 +0ms
  att dc:b1:49:4b:5a:ad: write: 020001 +0ms
  hci write acl data pkt - writing: 023700070003000400020001 +0ms
  hci-usb write: 023700070003000400020001 +0ms
  hci-usb HCI event: 13050134000100 +0ms
  hci onSocketData: 0413050134000100 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 19 +0ms
  hci-usb ACL Data In: 34201a001600040011141300ffff9ecadc240ee5a9e093f3a3b50100406e +16ms
  hci onSocketData: 0234201a001600040011141300ffff9ecadc240ee5a9e093f3a3b50100406e +0ms
  hci   event type = 2 +0ms
  hci           cid = 4 +0ms
  hci           handle = 52 +0ms
  hci           data = 11141300ffff9ecadc240ee5a9e093f3a3b50100406e +0ms
  att f7:75:57:11:4c:92: write: 08090012000328 +0ms
  hci write acl data pkt - writing: 0234000b000700040008090012000328 +0ms
  hci-usb write: 0234000b000700040008090012000328 +0ms
  hci-usb HCI event: 13050134000100 +16ms
  hci onSocketData: 0413050134000100 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 19 +15ms
  hci-usb HCI event: 13050137000100 +16ms
  hci onSocketData: 0413050137000100 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 19 +0ms
[... 218 lines cut ...]
  hci                   master clock accuracy = 0 +0ms
  att f8:27:ed:af:2d:fd: write: 020001 +16ms
  hci write acl data pkt - writing: 023300070003000400020001 +0ms
  hci-usb write: 023300070003000400020001 +0ms
  hci create le conn - writing: 010d20196000300000019fca0e3b12c60006000c000000c80004000600 +0ms
  hci-usb write: 010d20196000300000019fca0e3b12c60006000c000000c80004000600 +0ms
  hci-usb HCI event: 100158 +0ms
  hci onSocketData: 04100158 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 16 +0ms
  hci set scan enabled - writing: 010c20020001 +15ms
  hci-usb write: 010c20020001 +0ms
  hci disconnect - writing: 01060403330013 +0ms
  hci-usb write: 01060403330013 +0ms
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: LIBUSB_TRANSFER_STALL
    at Error (native)

C:\Users\Kokkerelletje\Documents\VanErum\PC Applicatie\testNoble>
```

Due to the small buffer size of the windows command prompt I am not able to create a larger stack trace. If you need one I will have to figure out a way around that...
With debug enabled it takes significantly longer before the error appears. Ater a few tests it seems to always happen after disconnecting from a device.

sandeepmistry, 2015-10-03:
@PimButton thanks for capturing this, the code to reproduce is fairly complex, in terms of environment and number of devices, so I won't be able to reproduce locally for sure.

To move forward, it would be great if you could find a simpler scenario to reproduce with one or two devices, then I can try it out here.

Some other options, try to capture the errors on the various endpoints: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L48
 similar to how the data event is done here: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L68 (There is more info on the node-usb read me: https://github.com/nonolith/node-usb#event-errorerror-1) Then we can at least determine which endpoint is causing the error.

sandeepmistry, 2015-10-28:
@PimButton any updates?

sandeepmistry, 2015-11-07:
Closing due to inactivity, re-open with requested info if you are still interested.

closed 2015-11-07 by sandeepmistry

scallacs, 2018-06-29:
I have the same error when i'm trying to connect/diconnect from a peripheral multiple times. 

With this test it always happen after 21 tries:

    describe("connect -> disconnect until error ", function(){
        let iteration = 1;

        it("will eventually crash but when ?", async function(){
                await (async function loop() {
                    for (; ; iteration++){
                        console.log(`\t\t\t - Iteration ${iteration}`);
                        await TestHelper.callbackMethodAsPromise(peripheral.connect.bind(peripheral));
                        expect(peripheral.state).to.be.eq("connected");
                        await TestHelper.callbackMethodAsPromise(peripheral.disconnect.bind(peripheral));
                        expect(peripheral.state).to.be.eq("disconnected");
                    }
                })();
        });
    });

    function callbackMethodAsPromise(mtd) {
        return new Promise((resolve, reject) => {
            mtd(() => {
                resolve();
            })
        });
    }

This produce error: 
     Uncaught Error: LIBUSB_TRANSFER_STALL

Using this BLE dongle: https://www.amazon.fr/gp/product/B01MR3BH4G
Noble version: 1.9.1

titaniumwhite, 2021-04-15:
@scallacs have you solved that? I have the same problem.