noble/noble #248: Error while installing noble

saurabh162, 2015-10-13:
Dear Developers

I have followed  following link in order to install "noble" (a node.js BLE).

http://github.com/sandeepmistry/noble. 

But after giving command: npm install noble. I received some error messages which I have saved in attached file for your kind consideration. 

Kindly inform me what can be the reasons for these errors and how can I eliminate them.

Following are the details of the environment I am using:

=> Ubuntu  with Kernel version 3.8.0-19-generic

Kindly inform me if you need any other information from me.

Thanks and Regards 

Saurabh Jain

sandeepmistry, 2015-10-26:
@saurabh162 

From your log file:
```
node-pre-gyp ERR! node -v v4.1.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok 
npm WARN optional dep failed, continuing usb@1.0.6
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../node_modules/nan/nan.h:41:3: error: #error This version of node/NAN/v8 requires a C++11 compiler
In file included from /home/ift/.node-gyp/4.1.2/include/node/node.h:42:0,
                 from /home/ift/.node-gyp/4.1.2/include/node/node_buffer.h:4,
                 from ../src/BluetoothHciSocket.cpp:7:
```
Since you are using Node.js 4.1.2, I think you need to update your compiler to GCC 4.8. What version of gcc and g++ do you have installed?

PS: I replied to your email, but it was "rejected by the server"

sandeepmistry, 2015-10-27:
@saurabh162 should be solved by http://stackoverflow.com/a/32578761/2020087

Please re-open if that doesn't work for you.

closed 2015-10-27 by sandeepmistry

saurabh162, 2015-10-28:
Thank you very much sandeep for your answer. I will update my complier to GCC 4.8 and will update you with result ..!!