noble/noble #253: Error when installing noble and bluetooth-hci-socket on Raspberry Pi 1 B

lijundr, 2015-10-19:
I download the lastest nodejs binaries for ARMv6 from the official website, and add the binary directory to PATH. And I also compile and install the lastest bluez as explained in [this post](http://www.jaredwolff.com/blog/get-started-with-bluetooth-low-energy/). The noble installation failed when installing bluetooth-hci-socket. So I track down the error by installing bluetooth-hci-socket first

```
npm install bluetooth-hci-socket
```

but get error as follows: 

```
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/app/node-v4.2.1-linux-armv6l/lib/node_modules/npm/node_modules/node-
gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.18.11+
gyp ERR! command "/home/pi/app/node-v4.2.1-linux-armv6l/bin/node" "/home/pi/app/node-v4.2.1-linux-armv6l/lib/node_module
s/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/app/node-v4.2.1-linux-armv6l/lib/node_modules/bluetooth-hci-socket
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
npm ERR! Linux 3.18.11+
npm ERR! argv "/home/pi/app/node-v4.2.1-linux-armv6l/bin/node" "/home/pi/app/node-v4.2.1-linux-armv6l/bin/npm" "install"
 "bluetooth-hci-socket" "-g"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! bluetooth-hci-socket@0.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-hci-socket@0.4.0 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the bluetooth-hci-socket package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-hci-socket
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
```

The log file "npm-debug.log" is pasted in [here](http://pastebin.com/JuqhA3VR). Please advise. Thanks!

sandeepmistry, 2015-10-27:
@lijundr I think you need to upgrade your g++/gcc version to 4.8+ to be compatible with Node.js 4.x

See: http://stackoverflow.com/a/32578761/2020087

lijundr, 2015-10-27:
@sandeepmistry upgrading to Debian Jessie solved this problem. Thanks.

closed 2015-10-27 by lijundr

sandeepmistry, 2015-10-27:
@lijundr awesome, thanks for the follow up!

joseLourenco, 2016-03-09:
lijundr 
did you installed, with success, the bleno on jessie?