noble/noble #255: noble won't connect after a while

ayhoung, 2015-10-24:
After continuously connecting to peripheral for multiple times, the method "peripheral.connect" seems to fail without any warning given. As you can see from the logs, it never gets to the "peripheral connected!" log. This happens consistently after connecting multiple times. I've already printed out the HCI logs but not sure what's wrong?

```javascript
noble.on('discover', function(peripheral) {
        console.log("Trying to connect to " + peripheral.advertisement.localName);
        explore(peripheral);
});
```

```javascript
function explore(peripheral) {
    console.log('inside explore function');

    peripheral.connect(function(error) {
        console.log('peripheral connected!');
        if (!error) {
            console.log('connected to peripheral ' + peripheral.advertisement.localName);
            noble.stopScanning();
        } else {

        }
    });
}
```

```
Trying to connect to 3F635B0C
inside explore function
  hci onSocketData: 043e2b02010000c304106a73201f110746f878ba1b17068397459ef4904b69fb0909303030303030304602ffffbf +26ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 0000c304106a73201f110746f878ba1b17068397459ef4904b69fb0909303030303030304602ffffbf +0ms
  hci                   type = 0 +1ms
  hci                   address = 20:73:6a:10:04:c3 +0ms
  hci                   address type = public +1ms
  hci                   eir = 110746f878ba1b17068397459ef4904b69fb0909303030303030304602ffff +0ms
  hci                   rssi = -65 +1ms
  hci onSocketData: 043e0f02010400c304106a732003020a04bf +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +1ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 0400c304106a732003020a04bf +1ms
  hci                   type = 4 +0ms
  hci                   address = 20:73:6a:10:04:c3 +1ms
  hci                   address type = public +0ms
  hci                   eir = 020a04 +1ms
  hci                   rssi = -65 +0ms
Local name = 0000000F RSSI : -65 Manu: 255 No. Connections: 0
  hci onSocketData: 043e2a0201000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5bb +2ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +293ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5bb +0ms
  hci                   type = 0 +1ms
  hci                   address = b4:99:4c:4c:09:95 +0ms
  hci                   address type = public +1ms
  hci                   eir = 02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5 +0ms
  hci                   rssi = -69 +1ms
  hci onSocketData: 043e0c0201040095094c4c99b400bb +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +1ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 040095094c4c99b400bb +0ms
  hci                   type = 4 +0ms
  hci                   address = b4:99:4c:4c:09:95 +1ms
  hci                   address type = public +0ms
  hci                   eir =  +1ms
  hci                   rssi = -69 +0ms
  hci onSocketData: 043e23020100010235b473e35f1702010613ff4c000c0e0019c4b4dfc44fd4148ae5fdb9e9bc +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 00010235b473e35f1702010613ff4c000c0e0019c4b4dfc44fd4148ae5fdb9e9bc +0ms
  hci                   type = 0 +1ms
  hci                   address = 5f:e3:73:b4:35:02 +0ms
  hci                   address type = random +1ms
  hci                   eir = 02010613ff4c000c0e0019c4b4dfc44fd4148ae5fdb9e9 +0ms
  hci                   rssi = -68 +1ms
  hci onSocketData: 043e2a0201000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5bb +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +1ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5bb +1ms
  hci                   type = 0 +0ms
  hci                   address = b4:99:4c:4c:09:95 +1ms
  hci                   address type = public +0ms
  hci                   eir = 02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5 +1ms
  hci                   rssi = -69 +0ms
  hci onSocketData: 043e0c0201040095094c4c99b400bb +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 040095094c4c99b400bb +0ms
  hci                   type = 4 +1ms
  hci                   address = b4:99:4c:4c:09:95 +0ms
  hci                   address type = public +1ms
  hci                   eir =  +0ms
  hci                   rssi = -69 +1ms
  hci onSocketData: 043e2a0201000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5b8 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +1ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 000095094c4c99b41e02011a1aff4c000215e2c56db5dffb1702010613ff4c000c0e0019c4b4dfc44fd4148ae5fdb9e9bc +0ms
  hci                   type = 0 +1ms
  hci                   address = 5f:e3:73:b4:35:02 +0ms
  hci                   address type = random +1ms
  hci                   eir = 02010613ff4c000c0e0019c4b4dfc44fd4148ae5fdb9e9 +0ms
  hci                   rssi = -68 +1ms
  hci onSocketData: 043e2a0201000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5bb +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +1ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5bb +1ms
  hci                   type = 0 +0ms
  hci                   address = b4:99:4c:4c:09:95 +1ms
  hci                   address type = public +0ms
  hci                   eir = 02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5 +1ms
  hci                   rssi = -69 +0ms
  hci onSocketData: 043e0c0201040095094c4c99b400bb +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 040095094c4c99b400bb +0ms
  hci                   type = 4 +1ms
  hci                   address = b4:99:4c:4c:09:95 +0ms
  hci                   address type = public +1ms
  hci                   eir =  +0ms
  hci                   rssi = -69 +1ms
  hci onSocketData: 043e2a0201000095094c4c99b41e02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5b8 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +1ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 000095094c4c99b41e02011a1aff4c000215e2c56db5dffb^Clic +0ms
  hci                   eir = 02011a1aff4c000215e2c56db5dffb48d2b060d0f5a71096e000000000c5 +222ms
  hci                   rssi = -72 +1ms
  hci onSocketData: 043e0c0201040095094c4c99b400b9 +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +1ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 040095094c4c99b400b9 +1ms
  hci                   type = 4 +0ms
  hci                   address = b4:99:4c:4c:09:95 +1ms
  hci                   address type = public +0ms
  hci                   eir =  +1ms
  hci                   rssi = -71 +0ms
```

ayhoung, 2015-10-24:
I disabled the bindings queue (as it would just keep on incrementing without end), but I ended up with this:

```
[32301.669739] Bluetooth: hci0 link tx timeout
[32301.669819] Bluetooth: hci0 killing stalled connection 20:73:6a:10:03:97
```

```javascript
NobleBindings.prototype.connect = function(peripheralUuid) {
  console.log('\x1b[33m', 'Bindings -- bindings connecting!', '\x1b[0m');
  var address = this._addresses[peripheralUuid];
  var addressType = this._addresseTypes[peripheralUuid];

  // if (!this._pendingConnection) {
    console.log('\x1b[33m', 'Bindings -- pending connect!', '\x1b[0m');
    // this._pendingConnection = true;
    this._hci.createLeConn(address, addressType);
  // } else {
  //   this._connectionQueue.push(peripheralUuid);
  //   console.log('\x1b[33m', this._connectionQueue.length, '\x1b[0m');
  // }
};
```

ayhoung, 2015-10-24:
it seems the problem was occuring only when I stopped scanning AFTER the peripheral was disconnected, which resulted in some kind of timeout error...

sandeepmistry, 2015-10-28:
@ayhoung can you please let me know what kernel version you are using (```uname -a```) and well as Bluetooth 4.0 adapter?

A HCI dump during the tests would also be great: ```sudo hcidump -t -x```

ayhoung, 2015-10-29:
@sandeepmistry thanks for the response! let me look into it

ayhoung, 2015-11-03:
@sandeepmistry 

Here's the HCI dump:
```
2015-11-03 10:41:31.825858 > HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 20:73:6A:10:07:9F (Public)
      Unknown type 0x07 with 16 bytes data
      Complete local name: 'FFB8000F'
      Unknown type 0xff with 1 bytes data
      RSSI: -53
2015-11-03 10:41:31.829390 > HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 20:73:6A:10:01:83 (Public)
      Unknown type 0x07 with 16 bytes data
      Complete local name: 'DD4FD74D'
      Unknown type 0xff with 1 bytes data
      RSSI: -40
2015-11-03 10:41:31.834171 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 20:73:6A:10:01:83 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-03 10:41:31.835311 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed
2015-11-03 10:41:31.835505 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-03 10:41:31.835499 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 6D:3F:E5:66:26:A0 (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -88
2015-11-03 10:41:31.836504 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-11-03 10:41:31.968122 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
```

The script gets caught indefinitely in this mode, when "peripheral.connect(function(error) {...}" is called, the callback function never gets called. The interesting thing is when I stop my nodejs script with a sigint, the hcidump shows this:

```
...continued from before
2015-11-03 10:44:08.415214 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-03 10:44:08.421133 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
```

uname -a produces the following:
Linux loopd-edison-2 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015 i686 GNU/Linux

ayhoung, 2015-11-03:
here's another one that I think will shed more light:
```
2015-11-03 11:52:20.178951 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2015-11-03 11:52:20.194056 > ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Group req (0x10) on handle 0xff08
2015-11-03 11:52:20.198483 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0xff08, end 0xffff
      type-uuid 0x2800
2015-11-03 11:52:20.208046 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-03 11:52:20.209153 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2015-11-03 11:52:20.209290 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2015-11-03 11:52:20.210437 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2015-11-03 11:52:20.223479 > ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Group req (0x10) on handle 0xff08
2015-11-03 11:52:20.224555 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2015-11-03 11:52:20.224578 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

sandeepmistry, 2015-11-07:
@ayhoung thanks are you running a stock noble or have you removed the connection queue?

For some reason the LE Create Connection command is failing.

```
2015-11-03 10:41:31.835311 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed
```

Could you please provide a full capture?
1. run ```sudo hciconfig hci0 reset```
2. run ```sudo hcidump -t -x```
3. In a new terminal window or tab, run your node script 

Also, what kernel version and hardware are you using? 
```uname -a```
``lsusb``

ayhoung, 2015-11-11:
@sandeepmistry thanks the reply

I'm using stock noble in this example, kernel version is:
Linux loopd-edison-2 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015 i686 GNU/Linux

My hardware is Intel Edison.

here's a longer dump
```
2015-11-11 15:18:57.814140 > HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 20:73:6A:10:04:D4 (Public)
      Unknown type 0x07 with 16 bytes data
      Complete local name: 'EABFB2BF'
      Unknown type 0xff with 1 bytes data
      RSSI: -84
2015-11-11 15:18:57.825166 > ACL data: handle 64 flags 0x02 dlen 5
    ATT: Write resp (0x13)
2015-11-11 15:18:57.827434 < ACL data: handle 64 flags 0x00 dlen 8
    ATT: Write cmd (0x52)
      handle 0x002a value  0x07
2015-11-11 15:18:57.835971 > HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 20:73:6A:10:06:72 (Public)
      Unknown type 0x07 with 16 bytes data
      Complete local name: '83AEA5FA'
      Unknown type 0xff with 1 bytes data
      RSSI: -92
2015-11-11 15:18:57.836319 > HCI Event: LE Meta Event (0x3e) plen 15
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 20:73:6A:10:06:72 (Public)
      TX power level: 4
      RSSI: -96
2015-11-11 15:18:57.839452 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-11 15:18:57.840724 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2015-11-11 15:18:57.840744 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-11-11 15:18:57.842772 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x00 (disabled)
2015-11-11 15:18:57.848488 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-11-11 15:18:57.855501 > ACL data: handle 64 flags 0x02 dlen 16
    ATT: Handle notify (0x1b)
      handle 0x002a
      value 0x07 0x45 0x41 0x42 0x46 0x42 0x32 0x42 0x46
2015-11-11 15:18:57.867444 < ACL data: handle 64 flags 0x00 dlen 8
    ATT: Write cmd (0x52)
      handle 0x002a value  0x11
2015-11-11 15:18:57.884038 > HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 20:73:6A:10:01:81 (Public)
      Unknown type 0x07 with 16 bytes data
      Complete local name: 'FD95D077'
      Unknown type 0xff with 1 bytes data
      RSSI: -87
2015-11-11 15:18:57.884993 > HCI Event: LE Meta Event (0x3e) plen 15
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 20:73:6A:10:01:81 (Public)
      TX power level: 4
[... 953 lines cut ...]
      bdaddr 20:73:6A:10:03:07 (Public)
      Unknown type 0x07 with 16 bytes data
      Complete local name: '5F2F4B8B'
      Unknown type 0xff with 1 bytes data
      RSSI: -80
2015-11-11 15:19:00.859404 > HCI Event: LE Meta Event (0x3e) plen 15
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 20:73:6A:10:03:07 (Public)
      TX power level: 4
      RSSI: -80
2015-11-11 15:19:00.861980 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-11 15:19:00.864411 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-11-11 15:19:00.864648 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-11 15:19:00.865765 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
```

sandeepmistry, 2015-11-14:
@ayhoung thanks for the longer HCI dump capture. Here are the areas of interest in the logs:

```
2015-11-11 15:18:58.644703 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 20:73:6A:10:01:81 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-11 15:18:58.647439 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1

...

2015-11-11 15:18:58.758300 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 20:73:6A:10:01:81 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-11 15:18:58.762412 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0b ncmd 1
    Error: ACL Connection Already Exists
```

For whatever reason, your app is trying to connect to the same peripheral twice, and noble doesn't handle the non-successful command status (0x0f) properly, so the connection continue is stuck.

sandeepmistry, 2015-11-28:
@ayhoung @ank-khandelwal

I've pushed https://github.com/sandeepmistry/noble/commit/9a746e03ec006a117202909182fc84b6eddff8b4 to resolve this. Now the status of the LE Create Conn HCI command is checked, and reported as a connection error if it is non-zero.

Please try it out via ```npm install sandeepmistry/noble``` and let me know how things go.

closed 2015-11-28 by sandeepmistry

ayhoung, 2015-12-25:
Awesome! Can't wait to try it :)