noble/noble #256: Cannot connect to more than 6 devices on El Capitan

haughank, 2015-10-26:
To reproduce:

- connect to 6 devices
- on the 7th device, it can be discovered, but connection attempt fails

other:

- it was consistently observed on 3 different machines running El Capitan
- the exact same code base can connect to at least 12 devices on Yosemite

oroce, 2015-10-26:
see this: https://github.com/sandeepmistry/noble/issues/214#issuecomment-122686710

sandeepmistry, 2015-10-27:
@haughank this is probably due to a change in OS X 10.11 nothing that noble to change to fix it unfortunately. They have either updated the firmware running on the Bluetooth adapter or something in CoreBluetooth or blued.

Have you tried with a native CoreBluetooth app?

taktran, 2015-10-27:
I've noticed before that noble has separate files for handling the different mac operating systems in https://github.com/sandeepmistry/noble/tree/master/lib/mac, would we need something similar for El Capitan?

Also, do you have an example of a CoreBluetooth app?

sandeepmistry, 2015-10-27:
@taktran as far as I know, none of the XPC messages have changed in El Captitan, so no changes needed.

I don't have a CoreBluetooth app off hand, but there are lots of examples out there.

Btw, noble will eventually move to using this lib on OS X: https://github.com/sandeepmistry/node-core-bluetooth

taktran, 2015-10-28:
Ok. Thanks for the heads up on `node-core-bluetooth`.

Great work!

haughank, 2015-11-09:
@sandeepmistry Confirmed by writing a simple CoreBluetooth app that it seems to be an OS issue. Following message consistently received after 6 connections:

"failed to connect Optional(Error Domain=CBErrorDomain Code=10 "Reached max allowed connections." UserInfo={NSLocalizedDescription=Reached max allowed connections.})"

Why do you think they have done that?

sandeepmistry, 2015-11-14:
@haughank probably for 4.2 support or something.

I've added a note to the read me about the new maximum being 6 for El Capitan: https://github.com/sandeepmistry/noble#maximum-simulanteous-connections

There's not much noble can do about this so closing the issue.

closed 2015-11-14 by sandeepmistry