noble/noble #257: Example read and write data?

simar88, 2015-10-28:
Hi everyone,
i would like to use this module to send and receive data between a raspberry pi and an arduino using hm-11 ble module.
I can find peripheral without problem and connect to it, but i've some problem with data transfer.
I've to use readHandle and writeHandle, it is right?
Thanks in advance!

ManuelOverdijk, 2015-10-28:
Here is an example how I handled reading and writing characteristics:

``` javascript
characteristic.read(function (error, data) {});
                    characteristic.on('read', function (data, isNotification) {
                        \\ data is read
                    }.bind(this));
```

``` javascript
characteristic.write(payload, false, function (error) {
                            if (!error) {
                            \\ write succesfull
                            } else {
                              \\ write unsuccessfull
                            }
                        }.bind(this));
```

Hope this helps to give you a general direction. Key is to discover your characteristics before reading/writing operations

sandeepmistry, 2015-10-28:
@ManuelOverdijk thanks for helping out!

@simar88 here are some resources:
 * wiki:
  * https://github.com/sandeepmistry/noble/wiki/Getting-started#reading-a-characteristic
  * https://github.com/sandeepmistry/noble/wiki/Getting-started#getting-notified-on-characteristic-value-change
 * code for other modules which maybe similar to the HM-11
  * https://github.com/shokai/blendmicro-node
  * https://github.com/xorbit/node-hm-ble
  * https://github.com/bbx10/node-cc2540
  * https://github.com/bbx10/node-nRF8001

P.S.: Gittter is perfect for questions like this - https://gitter.im/sandeepmistry/noble

closed 2015-10-28 by sandeepmistry

simar88, 2015-10-30:
Hi @sandeepmistry @ManuelOverdijk thanks for yours reply!
I'll try some things suggested in these day becuase i'm migrating from the NRF8001 library to some other because i'm changing BLE module.
NRF8001 node.js module works very well but not with hm-11, so i need to try other solutions.
Since it's very simple, event driven and based on noble i thought that also it has the same simplicity of implementation but i've seen some difference.
Thanks again for your help!

sandeepmistry, 2015-11-01:
@simar88 I wouldn't expect the modules I linked to, to work as is with the hm-11. You'll need to modify to match the advertisement and UUID's that are in the firmware you are running.

> So i need an explanation about charateristics. To use them i have to discorver peripheral and then loop through services and then loop through charateristics. It's right?

Yes. See this for an explanation: https://epxx.co/artigos/bluetooth_gatt.html

bkht, 2018-01-29:
Hi Sandeep, thank you very much for noble!
Now I scan from a Health Temperature Sensor with this script:
noble/examples/peripheral-explorer.js
It shows:
1809 (Health Thermometer)
  2a1c (Temperature Measurement)
    properties  read, notify
    value       00720900fe | 'r    ~'
How can I display readable temperature data from temperature sensor (float)?
Something like:
    value      36.25

bkht, 2018-01-30:
I figured it worked this way for me:
`var val = data.readUInt32LE(0) / 25600.0;`
And I figured the way this data was transmitted:
`        uint32_t quick_ieee11073_from_float(float temperature) {
            uint8_t  exponent = 0xFE; //Exponent is -2
            uint32_t mantissa = (uint32_t)(temperature * 100);

            return (((uint32_t)exponent) << 24) | mantissa;
        }`
Hope this helps somebody.