noble/noble #258: Error: Module version mismatch. Expected 14, got 47.

peralles, 2015-11-04:
I'm trying to use noble in my Mac but I got the error below:

node noble-device.js 
Starting...
module.js:355
  Module._extensions[extension](this, filename);
                               ^
Error: Module version mismatch. Expected 14, got 47.
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/Users/danielcp/node_modules/xpc-connection/index.js:3:15)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)

I have the following versions:

OS X El Capitan 10.11.1

npm -v
2.11.3

node -v
v0.12.7

npm WARN unmet dependency /Users/danielcp/node_modules/mkdirp requires minimist@'0.0.8' but will load
npm WARN unmet dependency /Users/danielcp/node_modules/minimist,
npm WARN unmet dependency which is version 1.2.0
/Users/danielcp
├── bplist-parser@0.1.0
├── colors@1.1.2
├── debug@2.2.0
├── easyimage@2.1.0
├── json@9.0.3
├── minimist@1.2.0 invalid
├── mkdirp@0.5.1
├── ms@0.7.1
├── nan@2.1.0
├── noble@1.2.0
├── noble-device@1.1.0
├── q@1.4.1
└── xpc-connection@0.1.4

npm ERR! invalid: minimist@1.2.0 /Users/danielcp/node_modules/minimist

sandeepmistry, 2015-11-07:
@peralles were you using an older version of node before and then upgraded?

You'll need to do a fresh install of noble or rebuild using [node-gyp](https://github.com/nodejs/node-gyp#installation):
```sh
sudo npm install -g node-gyp
cd /Users/danielcp/node_modules/xpc-connection/
node-gyp rebuild
```

closed 2015-11-07 by sandeepmistry

oroce, 2015-11-07:
@sandeepmistry @peralles or just run `npm rebuild` (`npm rebuild noble` is fine also) which will recompile noble.