noble/noble #259: Scan for particular UUID's

ank-khandelwal, 2015-11-05:
I am trying to scan for particular UUID's
So I declare a variable, as you have shown in the documentation

var serviceUUIDs = ['80eaca000003','cb554fe68bab',....] ;\\ Not working
var serviceUUIDs = '80eaca000003','cb554fe68bab'; \\Not working
var serviceUUIDs = ["80eaca000003","cb554fe68bab"]; \\Not working
Then I use the command to scan
noble.startScanning(serviceUUIDs, true);

But it does not return any output.

But when I search only for a single UUID 
var serviceUUIDs = '80eaca000003'
It scans properly.

And sometimes when I am using different UUID, it will scan for one particular UUID only which I have not mentioned.

Am I doing something wrong. Please let me know. 
Thank you.

ank-khandelwal, 2015-11-05:
I even tried in the advertise-discovery.js, 
noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
var deviceuuid = '004d06016483';
    noble.startScanning(deviceuuid,false);
  } else {
    noble.stopScanning();
  }
});
But the output I get is 
peripheral discovered (80eaca000003 with address <80:ea:ca:00:00:03, public>, connectable true, RSSI -63:
	hello my local name is:
		GML-M6DEV
	can I interest you in any of the following advertised services:
		["0"]
It is the same beacon every time.

sandeepmistry, 2015-11-07:
@ank-khandelwal please see the read me, the ```startScanning``` API only supports service UUID filtering: https://github.com/sandeepmistry/noble#start-scanning

Service UUID's also need to be 16-bit or 128-bit.

You'll have to filter on ```peripheral.id``` in your ```discover``` event handler.

closed 2015-11-07 by sandeepmistry