noble/noble #260: Does noble support multi connection on Raspbian simultaneously?

kaiser-ren, 2015-11-06:
Hello,
I want to use noble to connect with multi peripheral device on Raspberry Pi running raspbian, so I want to know that does noble support this kind of requirement? If yes, do you have any sample code or code  snippet? 
Thanks.

ank-khandelwal, 2015-11-06:
I am also trying to do the same thing, Here is my code :
var noble = require('./index');
var events = require('events').EventEmitter.prototype._maxListeners =0;
//var events = require('events');
//events.setMaxListeners(0);
//var noble = require('noble');
noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
var deviceuuid = [];
    noble.startScanning(deviceuuid,true);
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {

//noble.stopScanning();
peripheral.connect(function(error) {
console.log('Connected to the peripheral : ' + peripheral.advertisement.localName);

  if (peripheral.advertisement.manufacturerData) {
    var str =JSON.stringify(peripheral.advertisement.manufacturerData.toString('hex'));
    var UUID = str.slice(9,41);
    var Major = str.slice(41,45);
    var Minor = str.slice(45,49);
    var Power = str.slice(49,51);
   console.log('\r\nName :' + peripheral.advertisement.localName +' UUID: '+ UUID + ' Major: ' + Major + ' Minor: ' + Minor + ' Measured Power: '+ Power + ' RSSI: ' + peripheral.rssi+'\r\n');
  }

//peripheral.connect(function(error) {
//    console.log('Device Connected');
    peripheral.discoverServices(['180a'],function(error, characterstics){
    var Deviceinformation = characterstics[0];
    console.log('Device Information: '+ Deviceinformation);
});
});
});

ank-khandelwal, 2015-11-06:
Hi,
I am also trying to do the same thing using Raspberry Pi, here is my example code :

var noble = require('./index');
var events = require('events').EventEmitter.prototype._maxListeners =0;

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
var deviceuuid = [];
    noble.startScanning(deviceuuid,true);
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {

peripheral.connect(function(error) {
console.log('Connected to the peripheral : ' + peripheral.advertisement.localName);

   console.log('Name :' + peripheral.advertisement.localName  ' RSSI: ' + peripheral.rssi+'\r\n');
    peripheral.discoverServices(['180a'],function(error, characterstics){
    var Deviceinformation = characterstics[0];
    console.log('Device Information: '+ Deviceinformation);
});
});
});

Sometimes I am able to connect to 3 devices, sometimes 5. But it stops after a certain time and I have to reset the hci. 
I am currently new to this, but I hope this would be little helpful.

sandeepmistry, 2015-11-07:
@kaiser-ren please see @ank-khandelwal's code snippet above.

@ank-khandelwal each Bluetooth adapter has a limit to the max number of connections is can support.
Please provide the following:
1. Bluetooth adapter you are using
2. Output of ```lsusb``` and ```uname -a```
3. HCI dump when you have issues connection to multiple devices: ```sudo hcidump -t -x```

Note: you cannot scan for a ```deviceuuid``` only service UUID with start scanning.

PS: why are you setting the max listeners to zero?

ank-khandelwal, 2015-11-09:
@sandeepmistry  I am using  3 CSR 4.0 Bluetooth adapters. Here is the output of lsusb : 
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. 
Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
And here is the output of uname -a:
Linux pibeacon 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux

I am setting the max listeners to zero because I get the following error:
(node) warning: possible EventEmitter memory leak detected. 11 read listeners added. Use emitter.setMaxListeners() to increase limit. 
I tried to increase the the maximum Listeners but it started crashing many times.

sandeepmistry, 2015-11-09:
@ank-khandelwal the CSR dongle looks like it only supports up to 5 connections at a time: http://stackoverflow.com/a/25902689/2020087

Could you please provide an HCI dump, from when you try to connect more than 5 devices at a time?

ank-khandelwal, 2015-11-09:
@sandeepmistry  Yes I know CSR can support only 5 connections at a time. I trying to make use of 3 CSR dongles and connect to 15 devices. Here is when I try to connect I am able to connect to 3 peripherals (sometimes 5), and the output of hcidump is as follows:
HCI sniffer - Bluetooth packet analyzer ver 5.31
device: hci0 snap_len: 1500 filter: 0xffffffff
2015-11-09 01:52:20.941678 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2015-11-09 01:52:20.943527 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2015-11-09 01:52:20.943678 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2015-11-09 01:52:20.945426 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2015-11-09 01:52:20.945521 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2015-11-09 01:52:20.958460 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2015-11-09 01:52:20.958709 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2015-11-09 01:52:20.960486 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2015-11-09 01:52:20.960602 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2015-11-09 01:52:20.962419 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2015-11-09 01:52:20.962579 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2015-11-09 01:52:20.964425 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:15:83:E5:10:53
2015-11-09 01:52:20.973225 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2015-11-09 01:52:20.974444 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2015-11-09 01:52:20.974552 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2015-11-09 01:52:20.976432 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2015-11-09 01:52:20.984381 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x00 (disabled)
2015-11-09 01:52:20.986416 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-11-09 01:52:21.017416 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr EF:21:89:2F:AF:2D (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -80
2015-11-09 01:52:21.022430 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr F4:20:F5:D9:13:B0 (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -69
2015-11-09 01:52:21.025425 > HCI Event: LE Meta Event (0x3e) plen 36
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
[... 363 lines cut ...]
2015-11-09 01:52:27.927325 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 72 packets 1
2015-11-09 01:52:27.930312 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2015-11-09 01:52:28.967452 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 72
      interval 1000.00ms, latency 0.00ms, superv. timeout 4000.00ms
2015-11-09 01:52:31.448668 > ACL data: handle 70 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 02 08 00 A0 00 C8 00 00 00 F4 01 
2015-11-09 01:52:31.448830 < ACL data: handle 70 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 02 02 00 00 00 
2015-11-09 01:52:31.448871 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  46 00 A0 00 C8 00 00 00 F4 01 00 00 00 00 
2015-11-09 01:52:31.454772 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2015-11-09 01:52:31.465771 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x13 handle 70
      interval 0.00ms, latency 0.00ms, superv. timeout 0.00ms
2015-11-09 01:52:31.467774 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 70 reason 0x13
    Reason: Remote User Terminated Connection

sandeepmistry, 2015-11-14:
@ank-khandelwal thanks for the hcidump output here are the areas of interest:

```
2015-11-09 01:52:21.059533 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
bdaddr F4:20:F5:D9:13:B0 type 1
interval 96 window 48 initiator_filter 0
own_bdaddr_type 0 min_interval 6 max_interval 12
latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-09 01:52:21.063436 > HCI Event: Command Status (0x0f) plen 4
LE Create Connection (0x08|0x000d) status 0x00 ncmd 1

...

2015-11-09 01:52:21.535390 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
bdaddr CB:55:4F:E6:8B:AB type 1
interval 96 window 48 initiator_filter 0
own_bdaddr_type 0 min_interval 6 max_interval 12
latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-09 01:52:21.539514 > HCI Event: Command Status (0x0f) plen 4
LE Create Connection (0x08|0x000d) status 0x00 ncmd 1

...

2015-11-09 01:52:22.913806 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
bdaddr F4:20:F5:D9:13:B0 type 1
interval 96 window 48 initiator_filter 0
own_bdaddr_type 0 min_interval 6 max_interval 12
latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-09 01:52:22.916683 > HCI Event: Number of Completed Packets (0x13) plen 5
handle 72 packets 1
2015-11-09 01:52:22.917675 > HCI Event: Command Status (0x0f) plen 4
LE Create Connection (0x08|0x000d) status 0x0b ncmd 1
Error: ACL Connection Already Exists
```

For whatever reason your app is trying to connect the same peripheral twice, and noble does not handle the non-success command status correctly.

This is duplicate of #255 so closing, please follow #255 for updates.

closed 2015-11-14 by sandeepmistry

mcapraro, 2016-09-06:
@ank-khandelwal @sandeepmistry would either of you have any examples for how to read from multiple BTLE dongles and distinguish between the two? is that possible? thanks

sandeepmistry, 2016-09-06:
@mcapraro you can only do this via separate Node.js processes for now, see https://github.com/sandeepmistry/noble#multiple-adapters