noble/noble #261: Are messages limited to 20 char?

simar88, 2015-11-06:
Is possible that this module broke the message at the 20th char?
I need to send a longer string.
Thanks in advance!

Niksko, 2015-11-07:
I believe this is a limitation of Bluetooth LE characteristics. I think their maximum size is 20 bytes. If you need longer, either compress the information, use multiple characteristics, send it in bursts or try sending over a serial connection.

sandeepmistry, 2015-11-07:
@simar88 as @Niksko mentioned, due to some devices having an MTU of 23 bytes, the maximum size of characteristic values is 20 bytes. This is not related to noble, but caused by the device you are using.

closed 2015-11-07 by sandeepmistry

simar88, 2015-11-07:
Thanks for your reply, but i'm asking this becuase i'm using the hm11 and in its datasheet is written that there is no bytes limit.
Using some mobile test app in fact i'm able to receive string with more then 20 chars.

sandeepmistry, 2015-11-07:
@simar88 if you're using Linux, please provide an HCI dump: ```sudo hcidump -t -x```

Which mobile app test app were working?

simar88, 2015-11-07:
@sandeepmistry yes i'm using it with raspian on raspberry pi 2. I'll try the command next week where i've all the things.
The mobile app i'm using and that seems to do the job is jbwave