noble/noble #264: Run example enter-exit.js with 25 iBeacon - Not good

tinamore, 2015-11-13:
Hi

I have 25 iBeacon with Broadcast interval is: 1000ms (1 second), and Tx Power of each iBeacon is strong enough to send Bluetooth Adapter (I used USB bluetooth CSR 4.0 Dongle)

I set: EXIT_GRACE_PERIOD = 15000 (15 second), that is 15 times the opportunity to receive signals from Ibeacon.

Sometimes (1 - 3 minutes) I see a few ibeacon exit, This is undesirable.

I have a question:

I do not want to increment EXIT_GRACE_PERIOD, I don't want to wait too long to detect iBeacon exit (15 minutes is long enough). Also, I don't want  reduce Broadcast interval of ibeacon (reduce the battery's life time).

Node ble supports multiple USB adapter during a Scan (May be, same time run 2 USB adapter for receive signals from Ibeacon) ? This solution will increase the chance of receiving signals from Ibeacon.

Or anyone can help me slove my problem.

Thanks.

sandeepmistry, 2015-11-15:
@tinamore 

Are the beacons all close by? Have you tried adjusting the RSSI threshold (```RSSI_THRESHOLD```)?

> Can Node ble supports multiple USB adapter during a Scan ? 

No, it currently only supports one adapter per node instance, supporting more than one is fairly complicated.

tinamore, 2015-11-15:
@sandeepmistry  Thank you support
>Are the beacons all close by? Have you tried adjusting the RSSI threshold (RSSI_THRESHOLD)?
NO beacons all close by. A Few (1 or 2 or 3 or 4) ibeacon close by and ibeacon comback again after some the time loop in the setInterval in enter-exit.js

I keep RSSI_THRESHOLD = -90. Because, this value is minimum capture signal. If set value of RSSI_THRESHOLD > -90 then the adapter will be difficult to capture signals ?

I think the cause is because accidentally the transmissions of ibeacons is overlap. Some advertising packets of a few ibeacon is congestion.

If I test with 10 ibeacon then run pretty good, but with 25 ibeacon, few ibeacon exit and comback.Not stable, Although the each ibeacon has 15 times the opportunity to receive by USB adapter.

sandeepmistry, 2015-11-20:
@tinamore the issues you are seeing are probably not related to noble specifically.

It's more likely they are caused by the Bluetooth adapter you are using. I expect you'll see similar behaviour running the following command: ```sudo hcitool lescan --duplicates```

> I keep RSSI_THRESHOLD = -90. Because, this value is minimum capture signal. If set value of RSSI_THRESHOLD > -90 then the adapter will be difficult to capture signals ?

A higher value would trigger and exit event when the beacon is closer in range.

sandeepmistry, 2015-11-28:
Closing due to inactivity, please re-open with info request in previous comment.

closed 2015-11-28 by sandeepmistry