noble/noble #266: `NOBLE_REPORT_ALL_HCI_EVENTS` is only used on Linux/Win32 platform

swstack, 2015-11-17:
Hey Sandeep,

I'd like to be able to write an OSX application that monitors raw advertisement data without waiting for the scan response...is this possible currently? It seems that the environment variable `NOBLE_REPORT_ALL_HCI_EVENTS` is only ever read in the file `hci_socket/bindings.js` which is only ever required by Linux and Windows platform. If I am understanding the code correctly then it seems that `NOBLE_REPORT_ALL_HCI_EVENTS` is ignored on OSX.

Thanks! Let me know, I am willing to contribute back to the library if you could steer me in the right direction.

sandeepmistry, 2015-11-18:
@swstack correct, the env. var is not supported in OS X. It's currently in the Linux only section of the read me (https://github.com/sandeepmistry/noble#running-on-linux).

I don't think OS X will be able to support this, as we are limited to what CoreBluetooth/blued provides. You could see what events are reported here: https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js#L121

swstack, 2015-11-18:
Ah, i'm sorry, I didn't realize that the environment variable was nested under the Linux section of the README, makes sense now. Thanks for your feedback.

closed 2015-11-18 by swstack

sandeepmistry, 2015-11-20:
@swstack no worries.

Let me know if you have any suggestions on if this can be made clearer in the read me.