noble/noble #267: On OS X El Capitan Version 10.11.1 (15B42) no beacons are recognized

pide2000, 2015-11-17:
Since the update to El Capitan noble doesn't recognize Beacons at all. 
Android apps recognize the beacons, tried several manufacturers.

Here is the output of 
 DEBUG=* node index.js  

```
yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1447760133912",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +0ms
  express:application set "x-powered-by" to true +0ms
  express:application set "etag" to 'weak' +2ms
  express:application set "etag fn" to [Function: wetag] +2ms
  express:application set "env" to 'development' +0ms
  express:application set "query parser" to 'extended' +0ms
  express:application set "query parser fn" to [Function: parseExtendedQueryString] +0ms
  express:application set "subdomain offset" to 2 +0ms
  express:application set "trust proxy" to false +0ms
  express:application set "trust proxy fn" to [Function: trustNone] +1ms
  express:application booting in development mode +0ms
  express:application set "view" to [Function: View] +0ms
  express:application set "views" to '/Users/ry29/dev/thePhysicalWeb/views' +0ms
  express:application set "jsonp callback name" to 'callback' +0ms
  express:router use / query +1ms
  express:router:layer new / +0ms
  express:router use / expressInit +1ms
  express:router:layer new / +0ms
  express:router:route new /beaconList +3ms
  express:router:layer new /beaconList +0ms
  express:router:route get /beaconList +1ms
  express:router:layer new / +0ms
  express:router:route new /beaconInformation +0ms
  express:router:layer new /beaconInformation +0ms
  express:router:route get /beaconInformation +0ms
  express:router:layer new / +0ms
Example app listening at http://:::3000
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +164ms
  yosemite-bindings state change poweredOn +1ms
  noble stateChange poweredOn +0ms
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {
    "kCBScanOptionAllowDuplicates": 1
  },
  "kCBMsgArgUUIDs": []
} +0ms
  noble scanStart +0ms
Scan started...
```

sandeepmistry, 2015-11-18:
@pide2000 thanks for reporting!

What are you using for the beacon?

Things seem ok on my 2013 MacBook Pro Retina running the same version of El Capitan. I tested with the iOS Estimote beacon app and a Linux based beacon using node-bleacon.

Could you kill ``blued`` and try again? ```sudo killall blued```

pide2000, 2015-11-18:
@sandeepmistry 
I'm using radius network beacons and ibks. Tested with android radius app -> all beacons are visible

Runs on macbook pro retina early 2013

Tried node-bleacon also -> no success. 
Killed all blued -> no success

Yesterday I cleared the PRAM.

pide2000, 2015-11-19:
@sandeepmistry  
Solved the Problem so please close the issue.

Solution: 
First step, cleared the PRAM but I don't know if this is really necessary. 
Second step, removed all com.apple.bleutooth* files in /Library/Preferences

sandeepmistry, 2015-11-20:
@pide2000 good to know, they can get messed up sometimes.

I use this when blued gets confused: http://stackoverflow.com/a/23573314/2020087

pide2000, 2015-11-20:
Thank you.  :+1: 

Those Preferences stuff is quite fun sometimes and ........ always the last thing you think on

closed 2015-11-21 by sandeepmistry