noble/noble #268: Connection drops, LMP Response Timeout

PimButton, 2015-11-18:
Thanks to the support for the Asus BT-400 I'm rewriting my program to keep the connection to several peripherals open. 

Currently I am only using one peripheral but something strange happens. I connect to the peripheral and set a notify callback on one of the characteristics. Whenever the characteristic value is updated on the device the callback is called and I can read the new value. This works however only for about 30 seconds, then the connection with the peripheral is disconnected. 

Using the debugger on the peripheral I found the reason for the disconnect is "LMP Response Timeout" I have no clue what to do with this and Google won't help me much further.
Can you tell me if this has something to do with Noble? Do I need to adjust something to keep the connection alive? Or is this a issue with the peripheral? 

I pasted the relevant code (based on the Pizza example and heavily commented) for the notify and callback here: http://pastebin.com/YKx9nuHa

sandeepmistry, 2015-11-20:
@PimButton please provide an HCI dump capture: ```sudo hcidump -t -x```

Also, which kernel, linux distro, and BlueZ version do you have installed?

PimButton, 2015-11-20:
@sandeepmistry 
I'm on windows so ```hcidump``` is not available. I just ```set debug=true``` instead. The output is here: http://pastebin.com/mCrAWrYT Most of the output is from receiving a new value (orientation reading from an accelerometer) 

This seems to be the relevant piece:

```
  hci-usb ACL Data In: 40200b00070004001b0b004f523a34 +1s
  hci onSocketData: 0240200b00070004001b0b004f523a34 +2ms
  hci   event type = 2 +1ms
  hci           cid = 4 +1ms
  hci           handle = 64 +1ms
  hci           data = 1b0b004f523a34 +1ms
Nieuwe acc data: OR:4
  hci-usb ACL Data In: 40200b00070004001b0b004f523a36 +2s
  hci onSocketData: 0240200b00070004001b0b004f523a36 +3ms
  hci   event type = 2 +1ms
  hci           cid = 4 +1ms
  hci           handle = 64 +0ms
  hci           data = 1b0b004f523a36 +1ms
Nieuwe acc data: OR:6
  hci-usb HCI event: 050400400013 +654ms
  hci onSocketData: 04050400400013 +3ms
  hci   event type = 4 +1ms
  hci   sub event type = 5 +0ms
  hci           handle = 64 +1ms
  hci           reason = 19 +1ms
Connection disconnected
```

Just after the last 'New acc data: OR:6'  the connection is disconnected with handle 64 and reason 19. Does that provide some insight? I'm using Windows 7 64 bit and the Asus BT-400 dongle.

sandeepmistry, 2015-11-28:
@PimButton could you please provide full debug logs?

I'm wondering if your hitting the conditions discussed in https://github.com/sandeepmistry/noble/issues/269#issuecomment-159769073.

PimButton, 2015-12-23:
Ok, i'm an idiot. I screwed up using the NRF debugger and got the wrong disconnect reason. Instead of 0x22 (LMP Response Timeout) it's 0x16 which is LOCAL_HOST_TERMINATED_CONNECTION. So the issue is still there (and apparently my peripheral is the one that is disconnecting the connection) but it has nothing to do with LMP so I will close this issue and continue with the great debugging done by @george-hawkins :)

closed 2015-12-23 by PimButton