noble/noble #269: Noble stops getting characteristic updates after 30 seconds

george-hawkins, 2015-11-18:
Hi Sandeep --

I've run into an issue with Noble on Ubuntu 14.04. After about 30 seconds I can no longer read values or receive notification of value changes (without any obvious error being reported).

Initially I was working with a characteristic that used notify - after 30 seconds I would stop getting new notifications. No such issue occurred if, instead of using Noble, I monitored the characteristic with the Nordic [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) app on my Android phone.

I then switched to a setup where instead of notifications I just explicitly tried to read the characteristic every second (triggered by a standard node timer) and again the callback that I was passing to the characteristic's read(...) method simply stopped getting called back after things had been running for 30 seconds.

Initially I thought things were stopping working after 30 events (as my remote side was pushing out changes every second) - but it definitely seems to be 30 seconds - if I reduced the duration on my node timer from 1s to 500ms I could read the characteristic 60 times before silence kicked in.

Initially I was using an older version of noble that was still using `l2cap-ble.c` and I did some debugging with that - after 30 seconds the select call (with its 1s timeout) would just continuously timeout with a result of 0, i.e. there were no obvious errors it just looked like nothing more was coming in on `l2capSock`.

I then upgraded to Noble 1.2.1 which uses bluetooth-hci-socket - I see the underlying native code has gone through a dramatic update in the process. I haven't tried debugging this yet but I experience the same behavior.

Have you any ideas?

PimButton, 2015-11-18:
Well that's a coincidence, I just created a issue with exactly the same problem: #268 
Also using a Nordic NRF chip as peripheral.

george-hawkins, 2015-11-18:
@PimButton - that is a strange coincidence :smile: 

If I Google I do find a couple of references (but not the original specification) that point out that the LMP response timeout should be 30 seconds - so that in itself is interesting:

* These posts of the [TI forum](http://e2e.ti.com/support/wireless_connectivity/f/307/p/278353/974624) and the [FreeBSD list](http://lists.freebsd.org/pipermail/freebsd-bluetooth/2008-December/001275.html).
* The [LMP section](https://books.google.ch/books?id=mUyiREePWHwC&pg=PT156&lpg=PT156&dq=LMP+Response&source=bl&ots=0ArHWbwuOq&sig=TAbioMi0b-JkdVaiVabuAVA3cUI&hl=en&sa=X&ved=0CDUQ6AEwA2oVChMIkOyRjN6ayQIVK79yCh0wSAgw#v=onepage&q=LMP%20Response&f=false) of the book "Bluetooth 1.1: Connect Without Cables".

It's also covered e.g. here in the [Link Manager section](https://books.google.ch/books?id=-XpGAAAAQBAJ&pg=PA305&lpg=PA305&dq=LMP+Response&source=bl&ots=noalUGBSO5&sig=Mcs_0X6LftVciehF2_C4MWxNQ50&hl=en&sa=X&ved=0CFgQ6AEwCWoVChMIrcXZw9uayQIV5r1yCh0PIwCk#v=onepage&q=LMP%20Response&f=false) of the book "The Handbook of PAN Technologies and Protocols".

Where do you see the message "LMP Response Timeout"? I'm using `hcidump` to monitor things but don't see anything unusual:

    $ hcidump -X
    ...
    > HCI Event: Command Status (0x0f) plen 4
        LE Create Connection (0x08|0x000d) status 0x00 ncmd 1

I see the last event shown above, i.e. "Create Connection", as the connection is established and I don't see anything after that, i.e. I don't see anything bad being logged at the 30 second boundary.

I can actually monitor both ends, i.e. the Noble node code running on my Linux box and the nrf51822 (a RedBear BLE Nano) - as I have a serial link open to the BLE Nano.

The BLE Nano is running [arduino-BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral), another project from Sandeep, and I do not see it reporting any issue at the 30 second boundary, i.e. as far as it's concerned there's no issue. And it does notice the disconnect initiated from my Noble logic long after the 30 seconds are up.

I.e. I can run my Noble logic for e.g. 60 seconds - after 30 seconds it stops being able to read any characteristic - but this does not seem to mean that the connection has been lost at this point - only when I explicitly disconnect in my Noble logic at the end of the 60 seconds (or whatever) does the nrf51822 fire its `BLEDisconnected` event, i.e. the link must still exist at this point (and not simply have been disconnected at some lower level) for the disconnect on the Noble side and the event on the nrf51822 to line up.

As I said in my initial comment everything also _looks_ good in the Noble native logic, i.e. it's not any obvious error at this level that's stopping data being read - from the point of view of the `AF_BLUETOOTH` socket everything looks fine but it just stops getting further data after the 30 seconds are up.

So I am not seeing anything like "LMP Response Timeout", i.e. something that shouts that there's a problem (other than the obvious lack of received data).

I have the more detailed data output from `hcidump -w hcidump.out` but Github won't allow me attach anything other than text files, images, PDFs and Microsoft Office files :disappointed:

PimButton, 2015-11-18:
@george-hawkins I'm not using the RedBear BLE Nano, I have designed a custom board around the nrf51822 and am using Nordic's soft devices to program the chip directly. 

In my case the connection does get disconnected after the 30 seconds, Nordic's ```on_ble_evt``` function is called and a ```BLE_GAP_EVT_DISCONNECTED``` event is being passed. This event carries a reason for the disconnect which in my case is ```0x22```. This corresponds to the LMP Response Timeout according to Nordic's documentation: https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a01162.html#ga91c40ef8a7c61f6a3a036eac70956491

I haven't done a lot of logging in my Noble Logic but as far as I can see do Noble and the NRF51822 both agree on the disconnect. I.E. the status of the peripheral in Noble changes to disconnected. If I reconnect the cycle starts again, Noble and the NRF both respond well, but only between the first 30 seconds. 

Since @sandeepmistry has written the arduino-BLEPeripheral he must have some experience with the NRF so I hope he can shed some light on the issue. If you want to attach other files, I can recommend PasteBin to paste larger outputs :)

sandeepmistry, 2015-11-20:
@george-hawkins an HCI dump would be great (you can paste the output here, or in pastebin, Github gist etc.): ```sudo hcidump -t -x```

Also, what kernel version, BlueZ version and BLE adapter are you using?

george-hawkins, 2015-11-21:
@sandeepmistry - thanks as always for your prompt follow ups :smile: 

I've put together lots of details in [a gist](https://gist.github.com/george-hawkins/3cce07684f1033678818#file-noble-issue-269-md).

I've included output comparing the events my nNRF51822 receives when interacting with Noble and when doing much the same thing using the [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en).

I've also included version information for my Bluetooth related modules etc. and details about my Bluetooth device, along with hcidump and strace output.

The thing that surprises me most in what I captured is the lack of discovery events for the Noble setup - I've commented on this more in the gist. Maybe characteristics etc. are somehow being cached rather than discovered anew each time?

george-hawkins, 2015-11-25:
OK - I'm an idiot - a lot of my carefully compiled gist is gibber resulting from:

* Not looking hard enough at the data structures involved (and taking account of the fact they involved unions).
* Not reading the relevant [Nordic documentation](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v10.0.0%2Fble_serialization_s110_events_gattc.html).
* Most importantly, not understanding that the discovery events seen in my nRF51822 console weren't triggered from the Noble side of things, but were responses generated by discovery triggered by the code running on the nRF51822.

I've cleaned up the gist to reflect this - but the important thing is that the nrf51822, on every connection, tries to discover the primary services of the GATT server on the central. See [`nRF51822.cpp:449`](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/f1cd7bcb9df371b5369b99ca40f459f11f49cd7d/nRF51822.cpp#L449):
```
sd_ble_gattc_primary_services_discover(this->_connectionHandle, 1, NULL);
```

When the central is my phone (running nRF Master Control Panel) everything works fine and we get a `BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP` response to our request - this is handled and the code then tries to discover further services and so gets a further `BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP` telling us there are no more services to be discovered.

When the central is my Ubuntu box running a Noble application this phase fails - we get no response so the discovery times out after 30 seconds and as [per the spec](https://devzone.nordicsemi.com/question/31769/recover-from-ble_gattc_evt_timeout/?answer=31774#post-id-31774) "No more attribute protocol requests, commands, indications or notifications shall be sent to the target device on this ATT Bearer."

So apparently my Ubuntu box is expected to be running a GATT server but isn't.

If I comment out the above discovery call in the library logic then everything works fine. I.e. I get no timeout and reads and writes between the two sides proceed without problem past the 30 second  boundary.

This doesn't help @PimButton so much as I don't think he can change the behavior of his peripheral. So the real solution is to find out why my Ubuntu box (and in @PimButton's case his Windows box) isn't running a GATT server (assuming we have the same problem) when it should.

Note: even after I comment out the discovery logic there's still another difference between what's seen when the central is on Android and when it's on Ubuntu - I only see the `BLE_GATTS_EVT_SYS_ATTR_MISSING` event in the Ubuntu case. I haven't been able to find out much about what this really means. The message sequence is covered [here](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s110.api.v8.0.0/group___b_l_e___g_a_t_t_s___s_y_s___a_t_t_r_s___u_n_k___p_e_e_r___m_s_c.html) and I can see that @sandeepmistry's code is responding in the right way - but I can't find why the Ubuntu setup would be sending this message and the Android setup not.

george-hawkins, 2015-11-25:
Who should be supplying the GATT server? It you're using BlueZ 5.X then it does provide a GATT server - however I killed this in the past as it and Bleno can't work together - see [this old comment from Sandeep](https://github.com/sandeepmistry/bleno/issues/24#issuecomment-61910517) - has this situation changed?

Anyhow most Ubuntu users will not have the 5.X BlueZ user space stuff - for whatever reason the kernel side of things has been up-to-date since forever but the user space stuff was only changed to 5.X with Wily Werewolf, i.e. the October 2015 release (see this [search result](http://packages.ubuntu.com/search?keywords=bluez&searchon=names&exact=1&suite=all&section=main) for the user space BlueZ version number for each Ubuntu release). It won't be until April 2016 that the first LTS release has it.

So I'm pretty sure the 4.X user space stuff wouldn't include a GATT server?

Out of curiosity I did a little checking as to what actually was discovered by `sd_ble_gattc_primary_services_discovered(...)` when my Android phone is acting as the central device. Surprise surprise - it was the two services:

* Generic Access (0x1800)
* Generic Attribute (0x1801)

I've submitted an [arduino-BLEPeripheral pull request](https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/36) that fixes the situation in my case where I control both sides.

But from @PimButton's experience it sounds like many or at least some other peripheral devices may have similar issues. So is there an obvious fix if you've got the default, i.e. 4.X, BlueZ user space tools on Ubuntu? And on Windows (as @PimButton is) should the default stack be handling providing the GATT server?

Really this doesn't look like a Noble issue anymore - but I'll keep it open for others to comment.

george-hawkins, 2015-11-25:
I noticed in [`gatt.js:83`](https://github.com/sandeepmistry/bleno/blob/e7dbeb8c9fa4bf51523a8aa20e76c3a90725f67d/lib/hci-socket/gatt.js#L83) that any service you advertise with Bleno will include Generic Access and Generic Attribute automatically (as it should).

So I tried mixing Bleno and Noble in one application to see if I could in effect provide a dummy service that might respond to any `sd_ble_gattc_primary_services_discovered(...)` request while letting the real Noble logic get on with its work.

But Bleno and Noble don't seem to play well at the same time (and I wouldn't be surprised if I'm misunderstanding how a central side GATT server should be brought into the mix anyway).

In fact after running Bleno logic I found I had to reset my Bluetooth dongle (plug it and out) and reset the corresponding device with `hciconfig` before I could get my Noble logic working again :disappointed:

george-hawkins, 2015-11-26:
OK - I'll stop spamming this issue. Last update for today. So Bleno and Noble can work together fine - I had Bleno in advertising mode while I was trying to do things in Noble and this is what (probably fairly obviously) didn't play nice (and it was when I killed my app while Bleno was still advertising that things seemed to get into the hard-to-recover state described where I couldn't use Noble again until I'd zapped my Bluetooth device).

However even after realizing this I still wasn't able to get things into the state I wanted, i.e. get Bleno to somehow provide the missing GATT server that is causing the peripheral's remote service discovery step to timeout (and so halt communication with my real logic, i.e. the Noble logic).

sandeepmistry, 2015-11-26:
@george-hawkins awesome debugging so far! I would definitely say you are not spamming :)

Did you take a look at these lines?
 * https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gatt.js#L87
 * https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gatt.js#L89

The 2nd linked line should probably reply with an ATT error response instead of ignoring, something like: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js#L341

In regards to noble + bleno working together, it might be more straightforward if there was a shared singleton for bluetooth-hci-socket. Although it might be fine with the current setup on linux if the connection complete event treats noble requests as a new connection so it can respond to the central. The one gotcha is if noble stops ignoring the requests, and sends an error response, there is no opportunity for bleno to step in ...

george-hawkins, 2015-12-02:
OK - I've submitted pull request [`#277`](https://github.com/sandeepmistry/noble/pull/277) against Noble.

My previous pull request ([`#36`](https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/36)) against arduino-BLEPeripheral fixed things for me in the case where I was also responsible for the peripheral logic.

This pull request against Noble fixes the situation where you don't have any control over the peripheral and can't prevent it initiating a service discovery request.

This should hopefully work for situations like @PimButton's case.

My [writeup](https://github.com/sandeepmistry/noble/pull/277#issue-120003361) for the pull request is way longer than the actual fix which is really just commit [`0223525`](/george-hawkins/noble/commit/0223525). There's also a followup commit allowing you to turn off this behavior if really required - [`287ffff`](/george-hawkins/noble/commit/287ffff) - but I'd guess there isn't so much point in merging this addition.

PimButton, 2015-12-23:
As I explained in #268, I screwed up reading the disconnect reason. It turned out it wasn't 0x22 (LMP Response Timeout) but LOCAL_HOST_TERMINATED_CONNECTION (0x16). So it seems the NRF51822 I'm using is initiating the disconnect. So far I have not been able to find out **why** it drops the connection. 

However, according to the [Nordic Developer Zone](https://devzone.nordicsemi.com/question/5177/getting-reason-0x16-on-disconnect/):
```0x16 means Local Host Closed Connection (the stack terminated because of something, could be a L2Cap connection parameter update timeout or a pairing failure)```

Unfortunately the solution provided in #291 did not work for me. The connection still drops after 30-60 seconds. However, I still think the issues are related. Is it possible that my perhiperhal sends a L2CAp connection parameter update that Noble ignores? In that case a timeout is expected, and the scenario is identical to the problem @george-hawkins solved with his commit. I expect the request to be different however, so instead of denying the request it is still timing out. 

Does this make any sense to you? I have done some debugging of my own and captured two Wireshark streams: one between [BLE Tool](https://itunes.apple.com/nl/app/bluetooth-smart-tool/id525235616?mt=8) on my iPad & the NRF51822 peripheral and one between Noble & the peripheral. With the iPad application no disconnect happens and the connections stays alive. It seems it responds differently to the L2Cap requests but I'm a little out of my depth with this level of debugging and BLE knowledge. I attached the two captures, hopefully @sandeepmistry can take a look and see what is going on. I used the filter ```btle.data_header.length > 0``` to hide the empty packets. 

@george-hawkins I hope you don't mind me continuing here, since I think the issues are still related. If you feel like i'm hijacking your issue please let me know and I'll open a separate one.

george-hawkins, 2016-01-03:
@PimButton - if you're using my changes from #291 then it shouldn't just ignore a request, no matter what it is.

If you look at the `if` statement at [`gatt.js:86`](https://github.com/george-hawkins/noble/blob/6701fc61c5c0f44b948d90f12013ea464baffc80/lib/hci-socket/gatt.js#L86) you'll see there are various `debug` calls for the various paths.

You might want to see if these calls are outputting anything interesting - you can enable this as outlined [here](https://www.npmjs.com/package/debug) - you just need to set the environment variable `DEBUG` to `att`, i.e. the value seen when `debug` is pulled in at the top of `gatt.js`, and then run your node program as normal.

Beyond that I'm out of ideas :disappointed:

sandeepmistry, 2016-01-03:
I'm closing this now that PR #291 from @george-hawkins has been merged.

@PimButton please open a new issue for the issue you are seeing, and we can follow up on the discussion there. I tried to open the .pcapng files you provided in Wireshark 2.0.1 for OS X, but the decoder doesn't seem to work. I can post a screenshot in the new issue.

closed 2016-01-03 by sandeepmistry