noble/noble #272: Cannot install bluetooth-hci-socket

arusu0629, 2015-11-20:
I will execute example code but error occuer.
```
[pi@raspberry] $ node node_modules/noble/test.js                                                                    
module.js:339
    throw err;
    ^

Error: Cannot find module 'bluetooth-hci-socket'
    at Function.Module._resolveFilename (module.js:337:15)
    at Function.Module._load (module.js:287:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (/home/pi/testBluetooth/node_modules/noble/lib/hci-socket/hci.js:6:26)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
```

So, I will install 'bluetooth-hci-socket'
```
[pi@raspberry] $ npm install bluetooth-hci-socket
npm ERR! Linux 4.1.7-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "bluetooth-hci-socket"
npm ERR! node v4.2.2
npm ERR! npm  v2.14.7
npm ERR! code ENOSPC
npm ERR! errno -28
npm ERR! syscall write

npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc This is most likely not a problem with npm itself
npm ERR! nospc and is related to insufficient space on your system.
npm ERR! Linux 4.1.7-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "bluetooth-hci-socket"
npm ERR! node v4.2.2
npm ERR! npm  v2.14.7
npm ERR! code ENOSPC
npm ERR! errno -28
npm ERR! syscall write

npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc This is most likely not a problem with npm itself
npm ERR! nospc and is related to insufficient space on your system.
npm ERR! Linux 4.1.7-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "bluetooth-hci-socket"
npm ERR! node v4.2.2
npm ERR! npm  v2.14.7
npm ERR! path npm-debug.log.76d42d43ae7b7e32ec90f03716e083d3
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall rename

npm ERR! enoent ENOENT: no such file or directory, rename 'npm-debug.log.76d42d43ae7b7e32ec90f03716e083d3' -> 'npm-debug.log'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/testBluetooth/npm-debug.log
```

I see #253, my Raspberry Pi2 gcc version = 4.9.2
cannot install bluetooth-hci-socket with gcc 4.9.2?
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10)
```

Please Sir.

sandeepmistry, 2015-11-21:
@arusu0629 based on this:

```
npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc This is most likely not a problem with npm itself
npm ERR! nospc and is related to insufficient space on your system.
```

Looks you are out of disk space, so nothing related to bluetooth-hci-socket.

closed 2015-11-21 by sandeepmistry

arusu0629, 2015-11-21:
@sandeepmistry I'm sorry sir...
I didn't see error exactly. I do work well after Raspberry Pi disk increasing.
Thank you a lot.

CherylSha, 2018-07-19:
But in windows, it doesn't work. They are the same error.

Guley, 2019-07-30:
**Error while using node-eddystone-beacon-scanner**

[pi@raspberrypi:~/beacons $ node index.js
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'bluetooth-hci-socket'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/home/pi/beacons/node_modules/noble/lib/hci-socket/hci.js:6:26)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)](url)

**I did  install bluetooth-hci-socket using following command npm i bluetooth-hci-socket**

[](pi@raspberrypi:~/beacons $ npm i bluetooth-hci-socket
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

> usb@1.6.0 install /home/pi/beacons/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/2643fa-usb-v1.6.0-node-v64-linux-arm.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.0/usb-v1.6.0-node-v64-linux-arm.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.0/usb-v1.6.0-node-v64-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=10.15.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/home/pi/beacons/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:10: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
          ^~~~~~~~~~~
compilation terminated.
make: *** [libusb.target.mk:140: Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory '/home/pi/beacons/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.57-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/beacons/node_modules/usb
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 

> bluetooth-hci-socket@0.5.1 install /home/pi/beacons/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/beacons/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
[... 109 lines cut ...]
gyp ERR! cwd /home/pi/beacons/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/beacons/package.json'
npm WARN beacons No description
npm WARN beacons No repository field.
npm WARN beacons No README data
npm WARN beacons No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.6.0 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.6.0 install: `prebuild-install --verbose || node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-hci-socket@0.5.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.)

Could you please help me out
Thanks

snolangps, 2021-08-20:
I just recently purchased a Raspberry Pi Zero, and I'm also seeing this after trying to install `noble` and running `node app.js`.

Are there workarounds or ways to get around these errors, or fix them?